tianocore/edk2 #15: Failed to start command .\SecMain on Ubuntu

mingloo, 2015-06-18:
Hi all,

When I trying to use `build run` command to run the UEFI emulator, I've got the following failed information:
[image]

However, if I run the `./EmulatorPkg/build.sh run`, the GOP Window will be started as follows:
[image]

Any difference between these two command? And why the `build run` command will failed?

jljusten, 2015-11-25:
@mingloo: 'build run' was developed in a very hacky way just to support the NT32 platform. Maybe it could be made more general.

For both EmulatorPkg, OvmfPkg and DuetPkg, we use a build.sh script that helps to build and run the platform, and we don't bother with 'build run'.

lersek, 2016-02-02:
Closing as per previous commit.

closed 2016-02-02 by lersek