tianocore/edk2 #18: Windows 8.1 Installation crashes with latest OVMF

Sneaker15, 2015-07-17:
With the most recent build of OVMF (https://www.kraxel.org/repos/jenkins/edk2/edk2.git-ovmf-x64-0-20150716.b1120.g387e7c1.noarch.rpm) it is not possible to install Windows 8.1. Right after the disk selection, the installation crashes.

see also the comments here: http://vfio.blogspot.de/2014/08/primary-graphics-assignment-without-vga.html?showComment=1437118970658#c4684864484547064854

BugMaze, 2015-07-25:
Hi .

I'm having this exact issue trying to install Windows 10 RTM . The BSOD throws this error code : 
"kmode_exception_not_handled"

It happens right after selecting the disk to install Windows to .

This happens with a fresh code pulled from git today .

Thank you .

ariccio, 2015-07-30:
For reference: [Bug Check 0x1E: KMODE_EXCEPTION_NOT_HANDLED](https://msdn.microsoft.com/en-us/library/windows/hardware/ff557408.aspx)

What are the displayed values of parameters one through four?

BugMaze, 2015-07-30:
Hi @ariccio 
Unfortunately , Windows 8.1 and 10 do not show any useful info regarding this error except for "KMODE_EXCEPTION_NOT_HANDLED" .

However , this issue seems to be solved in recent code . I no longer have any issue installing Windows 10 .

Sneaker15, 2015-08-04:
yeah, can confirm the bug is fixed with latest release.

also I'm currently installing Windows 10, no issues so far (let's keep fingers crossed...)

BugMaze, 2015-08-21:
Hi .
dmesg still gets flooded with these messages (When trying to install Windows 10) : 

[Fri Aug 21 11:30:40 2015] kvm_get_msr_common: 92 callbacks suppressed
[Fri Aug 21 11:30:40 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:40 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:40 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:40 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:40 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:40 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:40 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:40 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:40 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:40 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:45 2015] kvm_get_msr_common: 92 callbacks suppressed
[Fri Aug 21 11:30:45 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:45 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:45 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:45 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:45 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:45 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:45 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:45 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:45 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:45 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:50 2015] kvm_get_msr_common: 92 callbacks suppressed
[Fri Aug 21 11:30:50 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:50 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:50 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:50 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:50 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:50 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:50 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:50 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:50 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:50 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:55 2015] kvm_get_msr_common: 92 callbacks suppressed
[Fri Aug 21 11:30:55 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:55 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:55 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:55 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:55 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:55 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:55 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:55 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:55 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639
[Fri Aug 21 11:30:55 2015] kvm [2926]: vcpu0 ignored rdmsr: 0x639

lersek, 2015-11-25:
With regard to the rdmsr's and KMODE_EXCEPTION_NOT_HANDLED with pre-release builds of Windows 10, you might want to consult

http://answers.microsoft.com/en-us/insider/forum/insider_wintp-insider_perf/bsod-kmodeexceptionnothandled-on-10041-in-kvm-with/7330c7af-abfa-4755-b851-b7720e34d80c

These should not pop up any longer, with released versions of Windows 10.

This issue should be closed. (But I can't.)

tomty89, 2015-11-26:
@BugMaze I think those kvm error are irrelevant to this issue? I bet you can get them without ovmf as well. I saw them myself too when I have Windows 10 on kvm but they don't seem to actually cause any problem. Anyway they should be reported to the kernel bugzilla.

BugMaze, 2015-12-01:
Hi . This issue is no longer happening , neither with Windows 8.1 nor with released 10 .
@tomty89 : I can see them happening with Arch Linux VMs , but not with Windows VMs anymore . Anyway , as you pointed out , they should be harmless (at least from the user's perspective) . 

I believe it is safe to request this issue closed .

stamerlan, 2016-01-20:
Hi @BugMaze, I have such issue now. I built OVMF from the github repo. Which version of OVMF shell I use to avoid this issue? 
P.S. Qemu 2.5.0 NVMe virtual drive, windows 10 x86_64

BugMaze, 2016-01-20:
Hi @stamerlan 
It has been months since I installed Windows so I can't really comment on this issue precisely . But I think your issue lies withing this NVMe drive . Have your tried installing to virtio-blk-pci drive ?

stamerlan, 2016-01-20:
@BugMaze, thank you for your reply.
I tried virtio-blk-pci:
qemu-system-x86_64 \
    -m 4096 \
    -smp 2 \
    -boot d \
    \
    -drive file=nvme.img,if=none,id=disk0,format=raw \
    -device virtio-blk-pci,drive=disk0 \
    \
    -cdrom win10_x86_64.iso \
    \
    -bios OVMF.fd

But I had KMODE_EXCEPTION_NOT_HANDLED error. 
And as far as I understand virtio-blk-pci isn't NVMe device. My goal is to install windows to qemu NVMe device.
May be you can give a piece of advice.

BugMaze, 2016-01-20:
@stamerlan 

I tried to do an installation of Windows 8.1 to a nvme device , but I couldn't even start the VM : 

qemu-system-x86_64: -device nvme,drive=drive0,bootindex=1: Device initialization failed

However , I have no issue with an OVMF split package built using (yaourt -S ovmf-git) on Arch .

This is my full CMD : 

qemu-system-x86_64 -name gaming -M q35 -nographic \
-enable-kvm -m 12G -cpu host,kvm=off,hv_relaxed,hv_spinlocks=0x1fff,hv_vapic,hv_time,hv_vendor_id=BeefyVMs -smp sockets=1,cores=4,threads=1 \
-object iothread,id=gaming0 \
-object iothread,id=gaming1 \
-vga none -nodefconfig \
-drive if=pflash,format=raw,readonly,file=/usr/share/ovmf/x64/ovmf_code_x64.bin \
-drive if=pflash,format=raw,file=/SSD/ovmf_gaming.bin \
-device ioh3420,bus=pcie.0,addr=1c.0,multifunction=on,port=1,chassis=1,id=root.1 \
-device vfio-pci,host=05:00.0,bus=root.1,addr=00.0,multifunction=on,romfile=/SSD/GTX770.rom \
-device vfio-pci,host=05:00.1,bus=root.1,addr=00.1 \
-device vfio-pci,host=0f:00.0 \
-device vfio-pci,host=10:00.0 \
-drive file=/SSD/Win_Gaming.img,cache=none,format=raw,if=none,id=drive0,aio=threads \
-device virtio-blk-pci,drive=drive0,ioeventfd=on,bootindex=1,iothread=gaming0 \
-drive file=/SSD/Win_Gaming_E.img,cache=none,format=raw,if=none,id=drive1,aio=threads \
-device virtio-blk-pci,drive=drive1,ioeventfd=on,bootindex=2,iothread=gaming1 \
-device virtio-scsi-pci,id=scsi \
-drive file=/SSD/Windows8.iso,id=iso_install,if=none,format=raw \
-device scsi-cd,drive=iso_install \
-cdrom /SSD/virtio.iso \
-localtime \
-netdev bridge,id=br0 -device virtio-net-pci,ioeventfd=on,netdev=br0,id=nic0,mac=B9:A7:2A:6A:09:B4 \
-monitor unix:/tmp/vm_gaming,server,nowait &

stamerlan, 2016-01-20:
@BugMaze 

You need to specify serial for nvme:
-device nvme,drive=drive0,serial=1234

BugMaze, 2016-01-20:
@stamerlan 

Using the following CMD : 

qemu-system-x86_64 -name test -M q35 -nographic \
-enable-kvm -m 3G -cpu host,kvm=off,hv_relaxed,hv_spinlocks=0x1fff,hv_vapic,hv_time,hv_vendor_id=NetworkBreak -smp sockets=1,cores=4,threads=1 \
-vga std -nodefconfig \
-object iothread,id=test0 \
-drive if=pflash,format=raw,readonly,file=/usr/share/ovmf/x64/ovmf_code_x64.bin \
-drive if=pflash,format=raw,file=/SSD/ovmf_test.bin \
-drive file=/SSD/Test.img,cache=none,format=raw,if=none,id=drive0,aio=threads \
-device nvme,drive=drive0,serial=1234,bootindex=1 \
-device virtio-scsi-pci,id=scsi \
-drive file=/SSD/Windows8.iso,id=iso_install,if=none,format=raw \
-device scsi-cd,drive=iso_install \
-cdrom /SSD/virtio.iso \
-localtime \
-net none \
-vnc :9 \
-monitor unix:/tmp/vm_test,server,nowait &

I was able to install Windws 8.1 with no issue at all : 

[image]

However , after the installation was completed and the VM rebooted for the first time , Windows Boot Manager refused to boot and kept on dropping me to the UEFI shell .

lersek, 2016-01-20:
This probably happens for two reasons:

* At the moment OVMF does not include the NVMe driver from edk2, `MdeModulePkg/Bus/Pci/NvmExpressDxe`. So the installation might work (because you boot the Windows installer off a CD-ROM, and Windows has a builtin driver for NVMe), but when you try to boot off the NVMe device, OVMF doesn't recognize it at all. I vaguely recall that one of the GNU/Linux distros include the NVMe driver as a downstream-only patch. Maybe SuSE, I'm not sure.
* Even if the NVMe driver were included in OVMF, OVMF's QEMU Boot Order Library would have to be updated to recognize and translate the relevant OpenFirmware device path pattern from QEMU -- so your `...,bootindex=N` option on the command line would work, when used in connection with `-device nvme`.

Supporting NVMe boot can be an OVMF feature request, of course (no promises on any schedule from my part though), but in any case, it should be a separate issue tracker item. This is about an installation crash (that I think is caused by performance MSR accesses performed by early Windows 10 pre-releases). If that problem has ceased, then this tracker item should be probably closed, and the NVMe RFE (request for enhancement) should be tracked separately. Thanks.

stamerlan, 2016-01-20:
Hi @lersek,
Is there any opportunity to configure OVMF to use NVMe driver from edk2?

lersek, 2016-01-20:
@stamerlan Configuration at the user's site cannot solve this; it is an enhancement that will take patches. Please see the comment right above yours.

stamerlan, 2016-01-20:
@lersek Sorry, I misunderstood you. May be you know any bios/uefi to run on qemu to install windows on virtual nvme drive?

BugMaze, 2016-01-20:
@stamerlan : 

You can either use OVMF or Seabios with QEMU . OVMF still needs some coding for proper NVMe support . You can add a feature request for that as @lersek suggested . Best of luck !

lersek, 2016-01-22:
@Sneaker15 @BugMaze Can you guys confirm the original issue reported in this item is gone? Thanks.

BugMaze, 2016-01-22:
@lersek  : 

I can confirm that the original issue is gone . Windows 8.1 and Windows 10-released install just fine for me . Thanks .

lersek, 2016-01-22:
@BugMaze: thank you.
@Sneaker15 or @jljusten: could you please close this? Thanks.

closed 2016-01-24 by jljusten

lersek, 2016-07-21:
This (closed) item has been manually migrated to:
https://tianocore.acgmultimedia.com/show_bug.cgi?id=72