tianocore/edk2 #21: OVMF can't handly large VMs with qemu/kvm and linux >= 4.2

blu3bird, 2015-09-01:
I'm using qemu 2.4.0 to start a Windows-VM with OVMF (Commit 8b4c856c9113f5ab4ca838075fc7d95e105c8417). After updating my kernel from 4.1.6 to 4.2.0 I noticed that it wouldn't start anymore.

Here's the output I get:
```
KVM internal error. Suberror: 1
emulation failure
EAX=00000800 EBX=00000002 ECX=c0000080 EDX=00000000
ESI=00000000 EDI=00003ed8 EBP=00000000 ESP=00000000
EIP=000a0000 EFL=00010006 [-----P-] CPL=0 II=0 A20=1 SMM=0 HLT=0
KVM internal error. Suberror: 1
emulation failure
KVM internal error. Suberror: 1
emulation failure
KVM internal error. Suberror: 1
emulation failure
KVM internal error. Suberror: 1
emulation failure
ES =0008 00000000 ffffffff 00c09300 DPL=0 DS   [-WA]
CS =0010 00000000 ffffffff 00c09b00 DPL=0 CS32 [-RA]
SS =0008 00000000 ffffffff 00c09300 DPL=0 DS   [-WA]
DS =0008 00000000 ffffffff 00c09300 DPL=0 DS   [-WA]
FS =0008 00000000 ffffffff 00c09300 DPL=0 DS   [-WA]
GS =0008 00000000 ffffffff 00c09300 DPL=0 DS   [-WA]
LDT=0000 00000000 0000ffff 00008200 DPL=0 LDT
TR =0000 00000000 0000ffff 00008b00 DPL=0 TSS32-busy
GDT=     bff11f18 0000003f
IDT=     00000000 0000ffff
CR0=00000033 CR2=00000000 CR3=00000000 CR4=00000640
DR0=0000000000000000 DR1=0000000000000000 DR2=0000000000000000 DR3=0000000000000000 
DR6=00000000ffff0ff0 DR7=0000000000000400
EFER=0000000000000800
Code=af af af af af af af af af af af af af af af af af af af af <00> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
EAX=bff28000 EBX=00000002 ECX=c0000080 EDX=00000000
ESI=00000000 EDI=00003eb8 EBP=00000000 ESP=00000000
EIP=000a0000 EFL=00010086 [--S--P-] CPL=0 II=0 A20=1 SMM=0 HLT=0
ES =0008 00000000 ffffffff 00c09300 DPL=0 DS   [-WA]
CS =0010 00000000 ffffffff 00c09b00 DPL=0 CS32 [-RA]
SS =0008 00000000 ffffffff 00c09300 DPL=0 DS   [-WA]
DS =0008 00000000 ffffffff 00c09300 DPL=0 DS   [-WA]
FS =0008 00000000 ffffffff 00c09300 DPL=0 DS   [-WA]
GS =0008 00000000 ffffffff 00c09300 DPL=0 DS   [-WA]
LDT=0000 00000000 0000ffff 00008200 DPL=0 LDT
TR =0000 00000000 0000ffff 00008b00 DPL=0 TSS32-busy
GDT=     bff11f18 0000003f
IDT=     00000000 0000ffff
CR0=00000033 CR2=00000000 CR3=bff28000 CR4=00000640
DR0=0000000000000000 DR1=0000000000000000 DR2=0000000000000000 DR3=0000000000000000 
DR6=00000000ffff0ff0 DR7=0000000000000400
EFER=0000000000000800
Code=af af af af af af af af af af af af af af af af af af af af <00> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
EAX=00000660 EBX=00000002 ECX=c0000080 EDX=00000000
ESI=00000000 EDI=00003e90 EBP=00000000 ESP=00000000
EIP=000a0000 EFL=00010006 [-----P-] CPL=0 II=0 A20=1 SMM=0 HLT=0
ES =0008 00000000 ffffffff 00c09300 DPL=0 DS   [-WA]
CS =0010 00000000 ffffffff 00c09b00 DPL=0 CS32 [-RA]
SS =0008 00000000 ffffffff 00c09300 DPL=0 DS   [-WA]
DS =0008 00000000 ffffffff 00c09300 DPL=0 DS   [-WA]
FS =0008 00000000 ffffffff 00c09300 DPL=0 DS   [-WA]
GS =0008 00000000 ffffffff 00c09300 DPL=0 DS   [-WA]
LDT=0000 00000000 0000ffff 00008200 DPL=0 LDT
TR =0000 00000000 0000ffff 00008b00 DPL=0 TSS32-busy
GDT=     bff11f18 0000003f
IDT=     00000000 0000ffff
CR0=00000033 CR2=00000000 CR3=bff28000 CR4=00000660
DR0=0000000000000000 DR1=0000000000000000 DR2=0000000000000000 DR3=0000000000000000 
DR6=00000000ffff0ff0 DR7=0000000000000400
EFER=0000000000000800
Code=af af af af af af af af af af af af af af af af af af af af <00> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
EAX=00000800 EBX=00000002 ECX=c0000080 EDX=00000000
ESI=00000000 EDI=00003ed8 EBP=00000000 ESP=00000000
[... 148 lines cut ...]
GDT=     bff11f18 0000003f
IDT=     00000000 0000ffff
CR0=00000033 CR2=00000000 CR3=00000000 CR4=00000000
DR0=0000000000000000 DR1=0000000000000000 DR2=0000000000000000 DR3=0000000000000000 
DR6=00000000ffff0ff0 DR7=0000000000000400
EFER=0000000000000000
Code=af af af af af af af af af af af af af af af af af af af af <00> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
The error only happens if I try to start a VM with lots of CPUs and/or RAM. This is the stripped down command to reproduce it:
```
qemu-system-x86_64 \
-enable-kvm \
-smp 8 \
-m 8192 \
-drive if=pflash,format=raw,readonly,file=/path/to/OVMF_CODE_x86_64.fd \
-drive if=pflash,format=raw,file=/path/to/OVMF_VARS_x86_64.fd
```
It is not always reproducable, sometimes a VM with 44 CPUs and 1G works, sometimes it doesn't. Here are the maximum SMP/MEM combinations I was able to get to work(sometimes, at least):
- 44/1G
- 33/2G
- 10/4G
- 5/8G
- 3/16G

If I boot back to linux 4.1.6, I am unable to reproduce the issue, therefore it must be caused by a recent kvm kernel change. However since it doesn't happen with seabios(i can create really insane SMP/MEM combinations there) I'm assuming it's an OVMF bug.

januszmk, 2015-09-15:
also have this problem + 100% cpu usage. I know there was a problem, but found that patch was already placed in kernel http://www.spinics.net/lists/kvm/msg119044.html
any workarround?

BugMaze, 2015-10-06:
I'm also having this issue since 4.2-rc1 . No VM activity other than 100% CPU usage . It happens on OVMF VMs only , SeaBIOS VMs are booting just fine . This issue is still present even on 4.3-rc4 , which is supposed to have a commit to revert the problematic code : https://lkml.org/lkml/2015/10/1/735 .

Thanks .

oatberry, 2015-10-09:
I have a similar problem with qemu 2.4.0 and ovmf 18419, Arch Linux amd64 host: https://ptpb.pw/VD1X.html

jljusten, 2015-10-22:
At one time it sounded like programming the MTRRs before decompression might help this. Over time it sounds less and less like that is the case.

Nevertheless I put together a patch that I think should set the MTRRs before decompression.

https://github.com/jljusten/edk2/tree/decompress-mtrr

Unfortunately, I can't reproduce the slow boot issue, so I can't really tell if my patch helps, or even if it correct.

infernix, 2015-10-28:
@jljusten I've tried your patch with 4.3-rc7 (qemu-kvm 2.4) and have about 50% success rate of booting with vfio passthrough for vga, usb and sata controllers, and 2GB ram with 1 core. When it boots, it still boots *very slow*. So doesn't seem to do much of anything. ACPI shutdown is equally slow..

When I increase RAM to 11G success rate drops to 0% or it's so slow that it's unusable (waited several minutes before giving up).

infernix, 2015-10-28:
Here's some snippets from an OVMF debug over serial port where the boot is terribly slow; full 5 minute log is here: https://bpaste.net/show/8734e2167373

```
======= Wed Oct 28 13:16:18 2015 (adjust=0.0us)
00.000: ecCoreStartupWithStack(0xFFFCC000, 0x818000)
22.979: Register PPI Notify: DCD0BE23-9586-40F4-B643-06522CED4EDE
...
24.519: InstallProtocolInterface: 3EBD9E82-2C78-4DE6-9786-8D4BFCB7C881 BF3CD028
27.177: ecCoreStartupWithStack(0xFFFCC000, 0x818000)
50.205: Register PPI Notify: DCD0BE23-9586-40F4-B643-06522CED4EDE
...
51.735: InstallProtocolInterface: 3EBD9E82-2C78-4DE6-9786-8D4BFCB7C881 BF3CD028
54.363: ecCoreStartupWithStack(0xFFFCC000, 0x818000)
77.401: Register PPI Notify: DCD0BE23-9586-40F4-B643-06522CED4EDE
...
78.924: InstallProtocolInterface: 3EBD9E82-2C78-4DE6-9786-8D4BFCB7C881 BF3CD028
81.594: SeCoreStartupWithStack(0xFFFCC000, 0x818000)
104.725: Register PPI Notify: DCD0BE23-9586-40F4-B643-06522CED4EDE
```

qemu commandline:
```
qemu-system-x86_64 -enable-kvm -machine pc-i440fx-2.3,accel=kvm,usb=off,mem-merge=off -cpu host,kvm=off -name Win81VFIO \
        -serial pipe:/tmp/qemudebugpipe -m 11000 -realtime mlock=on -mem-path /dev/hugepages -mem-prealloc \
        -rtc base=localtime,clock=host,driftfix=slew -no-hpet -global kvm-pit.lost_tick_policy=discard \
        -smp sockets=1,cores=3,threads=1 \
        -device vfio-pci,host=01:00.0 -device vfio-pci,host=01:00.1 \
        -vga none -device qxl  -no-quit \
        -drive if=pflash,format=raw,readonly,file=/home/infernix/Win81OVMF-CODE.fd \
        -drive if=pflash,format=raw,file=/home/infernix/Win81OVMF-VARS.fd \
        -device vfio-pci,host=00:1d.0 \
        -device vfio-pci,host=03:00.0 \
        -netdev type=tap,ifname=win81,script=/etc/qemu-ifup,id=net0,queues=3,vhost=on -device virtio-net-pci,netdev=net0,mac=52:54:00:12:34:56,vectors=7 \
        -monitor telnet:127.0.0.1:1234,server,nowait &
```

lersek, 2015-11-25:
Fixed in host kernel commit

commit 879ae1880449c88db11c1ebdaedc2da79b2fe73f
Author: Laszlo Ersek <lersek@redhat.com>
Date:   Wed Nov 4 12:54:41 2015 +0100

    KVM: x86: obey KVM_X86_QUIRK_CD_NW_CLEARED in kvm_set_cr0()

which is part of v4.4-rc1.

Item should be closed.

jljusten, 2015-11-25:
I know we've tried multiple things to help out with this in OVMF. Have we ruled out all possibilities?

lersek, 2015-11-25:
The kernel side fix is actually the extension of a preexistent KVM quirk. It doesn't make UefiCpuPkg and/or OvmfPkg immediately right. But, it does address the user-facing issue that the OVMF-user community primarily cares about.

It has also been mentioned that the code path in KVM that the extended quirk eliminates is not so nice in its own right -- it creates genuine lock contention on KVM's mmu_lock when the APs are started up all at once. So some rework in KVM might very well be justified even with the quirk disabled.

Regarding this isse. We can have two (or more) long-term tickets open *separately* for the MTRR setting problem, and for the UefiCpuPkg problem(s). I won't have bandwidth for the MTRR stuff in OvmfPkg any time soon.

Regarding the UefiCpuPkg problems, the MP services design should be documented from the ground up, and the implementation should be reworked similarly from the ground up. I don't suggest *replacing* the design; instead, the multiprocessing / locking aspects of the code should all be cleaned up, so that the code implement the design *safely*. Unlike the MTRR problem, I might have *some* bandwidth for this, simply because I'm interested in it and have some relevant multi-threading experience.

I agree that in the long term, OVMF should work well even with the KVM quirk disabled (that's why it is called a "quirk") -- which may possibly take (or not take) further KVM changes as well. But, I'd like to avoid that more GPU assignment users come to this issue tracker item, and say "sigh, it's still broken".

No, it's not broken any longer, as far as users should be concerned. The kernel patch is in, and it is likely going to be backported to both the 4.2.y and 4.3.y stable trees.

So please feel free to open two more separate issues. Thanks.

Manouchehri, 2015-11-28:
I'm encountering this bug with small VMs, sometimes even four cores and 2GB of RAM fails to start. I'm on 4.2.6.

I have not applied 879ae1880449c88db11c1ebdaedc2da79b2fe73f yet.

BugMaze, 2015-12-16:
@lersek:
4.4-rc5 here on an X99 platform , both Windows VMs (OVMF) still have this issue . It's not different than running a 4.2.0 kernel .

bonzini, 2016-01-07:
Please try 4.4-rc7 (relevant patches attached to https://bugzilla.kernel.org/show_bug.cgi?id=107561).

infernix, 2016-01-07:
4.4-rc8 here and things work fine with a 3-core 11GB VM

jljusten, 2016-01-07:
@BugMaze: You get a chance to try rc7+?

BugMaze, 2016-01-07:
Compiling 4.4-rc8 right now . Will report back .

lersek, 2016-01-22:
@jljusten Based on the most recent feedback, I think we should close this.
@bonzini Thanks a lot for your work on upstream kernel BZ 107561.

jljusten, 2016-01-22:
@lersek: Ok. I'll close it. I was hoping we could do something in OVMF to help out people that were running the affected kernels, but it doesn't seem like we were able to find a good fix for that.

closed 2016-01-22 by jljusten

Frogging101, 2016-01-25:
Found something that might be relevant, or at least interesting. When I heard that the problem was fixed in 4.4, I upgraded and found that the problem was still present. After some trial and error, I found that compiling/installing a new version of OVMF fixed it for me. 

I was curious about what OVMF could have to do with it, so I ran a bisection. Revisions prior to commit 68f0674 have the problem. It's fixed after that. I don't have a clue why; it's not obvious what the changes in that commit have to do with anything.

lersek, 2016-01-25:
@Frogging101 

The EFI MP Services Protocol allows the caller to run "background tasks" on APs (application processors). The code that the APs run like this cannot call into UEFI itself, but the APs can run other, independent code, especially if that code doesn't mess with hardware state (i.e., it doesn't try to interfere with UEFI in any way). Think number crunching or whatever.

Also, "background" is not a requirement; the caller can synchronously wait until all APs finish their jobs; so it can be considered a synchronous work distribution for better performance.

Anyway, the code that is being patched by the commit you mention, commit 68f0674, is the event loop of application processors. It's an infinite loop. When there is something to do, a "job", the AP will do it; otherwise the AP will just loop idly.

Before commit 68f0674, this loop was a simple busy loop, with quite bad consequences even for the device assignment-less case. (Think power consumption for example.) I reported this at:

http://thread.gmane.org/gmane.comp.bios.tianocore.devel/10916/focus=11446

IIRC commit 68f0674 got written as a response to that report. In the event loop, each AP would be then put to sleep for real (with the HLT instruction, see `CpuSleep()`). And, on any new job, the BP would wake APs by rebooting them from scratch, with an INIT-SIPI-SIPI interprocessor interrupt (=IPI) sequence.

I think the problem you see with pre-68f0674 builds is that device assignment must make some memory uncacheable, and the pre-68f0674 event loop (a busy loop) creates catastrophic contention on the system RAM. Namely, in each iteration of the event loop, each AP takes and releases a spinlock, to see if there's a new task to perform. Ultimately you have all APs continuously contending for the same spinlock.

Although each AP executes `PAUSE` once per iteration too:

    /**
      Requests CPU to pause for a short period of time.

      Requests CPU to pause for a short period of time. Typically used in MP
      systems to prevent memory starvation while waiting for a spin lock.

    **/
    VOID
    EFIAPI
    CpuPause (
      VOID
      )
    {
      __asm__ __volatile__ ("pause");
    }

This is apparently not enough (pre-68f0674) to prevent memory starvation, when device assignment makes some RAM uncacheable.

We formed a plan to document and clean up the EFI MP Services implementation in UefiCpuPkg, but thus far it has remained only a plan. :(

FWIW, fresh OVMF builds should work in practice on v4.4 hosts.

Frogging101, 2016-01-25:
@lersek

That is interesting. One thing I still don't understand however, is why this problem did not occur on kernel 4.1.x even when PCI device assignment was used. The symptoms are similar if not identical to [the bug I linked](https://bugzilla.kernel.org/show_bug.cgi?id=107561), which affects kernels >=4.2 and <4.4. And this UefiCpuPkg bug appears to rear its head on a similar range of kernel releases (it works fine on 4.1.12, breaks down on 4.4). It could be a coincidence, as a lot of other things have happened between 4.1 and 4.4. It's still interesting though that the symptoms are so similar as well.

bonzini, 2016-01-26:
Even with 4.4 there are some changes compared to 4.1 that may cause this bug. 4.4 only fixed the bugs, it didn't revert everything.

lersek, 2016-07-21:
This (closed) item has been manually migrated to
https://tianocore.acgmultimedia.com/show_bug.cgi?id=73