tianocore/edk2 #23: kvm emulation failure

szalwia, 2015-09-30:
using the ovmf-git AUR package on Arch Linux (tianocore/edk2-OvmfPkg@304e481ef86ffae4aa5742da66a20677126174ed), qemu spits this out on boot:
```
KVM internal error. Suberror: 1
emulation failure
KVM internal error. Suberror: 1
emulation failure
EAX=00000023 EBX=00000002 ECX=00000000 EDX=00000600
ESI=00000000 EDI=00001fc2 EBP=00000000 ESP=00000000
EIP=00001000 EFL=00010002 [-------] CPL=0 II=0 A20=1 SMM=0 HLT=0
ES =0000 00000000 0000ffff 00009300 DPL=0 DS16 [-WA]
CS =9f00 0009f000 0000ffff 00009b00 DPL=0 CS16 [-RA]
SS =0000 00000000 0000ffff 00009300 DPL=0 DS16 [-WA]
DS =9f00 0009f000 0000ffff 00009300 DPL=0 DS16 [-WA]
FS =0000 00000000 0000ffff 00009300 DPL=0 DS16 [-WA]
GS =0000 00000000 0000ffff 00009300 DPL=0 DS16 [-WA]
LDT=0000 00000000 0000ffff 00008200 DPL=0 LDT
TR =0000 00000000 0000ffff 00008b00 DPL=0 TSS32-busy
GDT=     bff13f18 0000003f
IDT=     00000000 0000ffff
CR0=00000033 CR2=00000000 CR3=00000000 CR4=00000000
DR0=0000000000000000 DR1=0000000000000000 DR2=0000000000000000 DR3=0000000000000000 
DR6=00000000ffff0ff0 DR7=0000000000000400
EFER=0000000000000000
Code=af af af af af af af af af af af af af af af af af af af af <00> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
EAX=00000023 EBX=00000002 ECX=00000000 EDX=00000600
ESI=00000000 EDI=00001fc2 EBP=00000000 ESP=00000000
EIP=00001000 EFL=00010002 [-------] CPL=0 II=0 A20=1 SMM=0 HLT=0
ES =0000 00000000 0000ffff 00009300 DPL=0 DS16 [-WA]
CS =9f00 0009f000 0000ffff 00009b00 DPL=0 CS16 [-RA]
SS =0000 00000000 0000ffff 00009300 DPL=0 DS16 [-WA]
DS =9f00 0009f000 0000ffff 00009300 DPL=0 DS16 [-WA]
FS =0000 00000000 0000ffff 00009300 DPL=0 DS16 [-WA]
GS =0000 00000000 0000ffff 00009300 DPL=0 DS16 [-WA]
LDT=0000 00000000 0000ffff 00008200 DPL=0 LDT
TR =0000 00000000 0000ffff 00008b00 DPL=0 TSS32-busy
GDT=     bff13f18 0000003f
IDT=     00000000 0000ffff
CR0=00000033 CR2=00000000 CR3=00000000 CR4=00000000
DR0=0000000000000000 DR1=0000000000000000 DR2=0000000000000000 DR3=0000000000000000 
DR6=00000000ffff0ff0 DR7=0000000000000400
EFER=0000000000000000
Code=af af af af af af af af af af af af af af af af af af af af <00> 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

Kernel 4.2.1, ovmf x64, `-M i440fx` on Intel Core i7 4790k.
When passing through a dedicated GPU (R9 280x) via vfio-pci, I just get a black screen without any error output from qemu.

mrschyte, 2015-11-07:
I think this is the same issue as https://github.com/tianocore/edk2/issues/21

lersek, 2015-11-25:
Duplicate of #21, should be closed as such.

lersek, 2016-01-22:
@jljusten do you agree this is a dupe of #21?

closed 2016-01-24 by jljusten

lersek, 2016-07-21:
This (closed) item has been manually moved to
https://tianocore.acgmultimedia.com/show_bug.cgi?id=74