tianocore/edk2 #24: EmulatorPkg not spawning window, triggering breakpoint instead, when built with gcc 4.9

rsalveti, 2015-10-09:
Host:
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu Wily Werewolf (development branch)
Release:	15.10
Codename:	wily

$ gcc-4.9 --version
> gcc-4.9 (Ubuntu 4.9.3-4ubuntu1) 4.9.3
> Copyright (C) 2015 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

EmulatorPkg$ ./build.sh 
> Initializing workspace
> /home/rsalveti/projects/tianocore/edk2/BaseTools
> Loading previous configuration from $WORKSPACE/Conf/BuildEnv.sh
> WORKSPACE: /home/rsalveti/projects/tianocore/edk2
> EDK_TOOLS_PATH: /home/rsalveti/projects/tianocore/edk2/BaseTools
> using prebuilt tools
> Build environment: Linux-4.2.0-14-generic-x86_64-with-Ubuntu-15.10-wily
> Build start time: 12:51:00, Oct.09 2015
> 
> WORKSPACE        = /home/rsalveti/projects/tianocore/edk2
> ECP_SOURCE       = /home/rsalveti/projects/tianocore/edk2/EdkCompatibilityPkg
> EDK_SOURCE       = /home/rsalveti/projects/tianocore/edk2/EdkCompatibilityPkg
> EFI_SOURCE       = /home/rsalveti/projects/tianocore/edk2/EdkCompatibilityPkg
> EDK_TOOLS_PATH   = /home/rsalveti/projects/tianocore/edk2/BaseTools
> 
> Architecture(s)  = X64
> Build target     = DEBUG
> Toolchain        = GCC49
> 
> Active Platform          = /home/rsalveti/projects/tianocore/edk2/EmulatorPkg/EmulatorPkg.dsc
> Flash Image Definition   = /home/rsalveti/projects/tianocore/edk2/EmulatorPkg/EmulatorPkg.fdf

EmulatorPkg$ ./build.sh run 
> Initializing workspace
> /home/rsalveti/projects/tianocore/edk2/BaseTools
> Loading previous configuration from $WORKSPACE/Conf/BuildEnv.sh
> WORKSPACE: /home/rsalveti/projects/tianocore/edk2
> EDK_TOOLS_PATH: /home/rsalveti/projects/tianocore/edk2/BaseTools
> using prebuilt tools
> Reading symbols from /home/rsalveti/projects/tianocore/edk2/Build/Emulator/DEBUG_GCC49/X64/Host...done.
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
> 
> EDK II UNIX Host Emulation Environment from http://www.tianocore.org/edk2/
>   BootMode 0x00
>   OS Emulator passing in 128 KB of temp RAM at 0x40000000 to SEC
>   FD loaded from ../FV/FV_RECOVERY.fd at 0x102000000 contains SEC Core
> 
> 0x1020003e0 Loading /home/rsalveti/projects/tianocore/edk2/Build/Emulator/DEBUG_GCC49/X64/EmulatorPkg/Sec/Sec/DEBUG/EmuSec.dll with entry point 0x1020003e0
> SEC Has Started
> 0x1020052e0 Loading /home/rsalveti/projects/tianocore/edk2/Build/Emulator/DEBUG_GCC49/X64/MdeModulePkg/Core/Pei/PeiMain/DEBUG/PeiCore.dll with entry point 0x1020052e0
> 
> Program received signal SIGTRAP, Trace/breakpoint trap.
> add symbol table from file "/home/rsalveti/projects/tianocore/edk2/Build/Emulator/DEBUG_GCC49/X64/EmulatorPkg/Sec/Sec/DEBUG/EmuSec.dll" at
> 	.text_addr = 0x1020003e0
> add symbol table from file "/home/rsalveti/projects/tianocore/edk2/Build/Emulator/DEBUG_GCC49/X64/MdeModulePkg/Core/Pei/PeiMain/DEBUG/PeiCore.dll" at
> 	.text_addr = 0x1020052e0
> CpuBreakpoint () at /home/rsalveti/projects/tianocore/edk2/MdePkg/Library/BaseLib/X64/GccInline.c:107
> 107	}
> (gdb) c
> Continuing.
> 0x10202b000 Loading /home/rsalveti/projects/tianocore/edk2/Build/Emulator/DEBUG_GCC49/X64/MdeModulePkg/Universal/StatusCodeHandler/Pei/StatusCodeHandlerPei/DEBUG/StatusCodeHandlerPei.dll with entry point 0x10202b000
> 
> Program received signal SIGTRAP, Trace/breakpoint trap.
> add symbol table from file "/home/rsalveti/projects/tianocore/edk2/Build/Emulator/DEBUG_GCC49/X64/MdeModulePkg/Universal/StatusCodeHandler/Pei/StatusCodeHandlerPei/DEBUG/StatusCodeHandlerPei.dll" at
> 	.text_addr = 0x10202b000
> CpuBreakpoint () at /home/rsalveti/projects/tianocore/edk2/MdePkg/Library/BaseLib/X64/GccInline.c:107
> 107	}
> (gdb) c
> Continuing.
> 
> Program received signal SIGSEGV, Segmentation fault.
> 0x000000010202b295 in StatusCodeHandlerPeiEntry (FileHandle=0x10202ad68, PeiServices=0x4000f878)
>     at /home/rsalveti/projects/tianocore/edk2/MdeModulePkg/Universal/StatusCodeHandler/Pei/StatusCodeHandlerPei.c:57
> 57	    Status = RscHandlerPpi->Register (SerialStatusCodeReportWorker);                     
> (gdb) quit

But it works just fine with GCC 4.8.

mytbk, 2015-10-12:
I ran into a SIGTRAP too. I'm using Arch Linux with GCC 5.2.0.

jljusten, 2015-11-26:
I posted a patch that should fix this:
"EmulatorPkg: Fix GCC49 build hang in PeiCore"
http://article.gmane.org/gmane.comp.bios.edk2.devel/4755

closed 2015-12-03 by jljusten