tianocore/edk2 #30: Bug in Math64.c - undefined behavior (UDK2014.SP1 branch)

bcran, 2015-11-25:
With a checkout of svn/branches/UDK2014.SP1, building a driver with Clang 3.8 the following error is generated:
```
edk2/MdePkg/Library/BaseLib/Math64.c:89:22: error: shifting a negative signed value is undefined [-Werror,-Wshift-negative-value]
  TestValue = (((-1) << (sizeof (-1) * 8 - 1)) >> (sizeof (-1) * 8 - 1));
                ~~~~ ^
```

lersek, 2015-11-25:
This is indeed undefined behavior in C99 and later. Not (necessarily) undefined behavior in C89, C90, and C95 (which edk2 uses, as far as I know). A somewhat related discussion can be seen here, in QEMU-land: http://thread.gmane.org/gmane.comp.emulators.qemu/376789

The problem is that this occurs in edk2 code that is supposed to test, *at runtime*, whether the compiler supports arithmetic right shift (i.e., when the new MSB matches the old MSB). I think that InternalMathARShiftU64() should be patched to perform its action purely in UINT64.

The fallback code near the end of the function seems to do something like that, except the cast to INTN for checking the sign. The sign check should instead be performed with an explicit test for the MSB.

My two cents anyway.

lersek, 2016-01-22:
@bcran: it looks like @ardbiesheuvel fixed this in commit b331b99f, on the master branch. Since you reported it for UDK2014.SP1, I guess Ard's patch will have to be cherry-picked for that stable release.

lersek, 2016-02-02:
Labelling this as UDK due to the previous comment.

ajfish, 2016-02-02:
It is compiler defined behavior in the older world, and that is why clang has added the warning. I also see the error in the base Tools as it has a case of shifting -1. 

https://github.com/tianocore/edk2/blob/master/BaseTools/Source/C/Common/FvLib.c
  //
  // Get next file, compensate for 8 byte alignment if necessary.
  //
  *NextFile = (EFI_FFS_FILE_HEADER *) ((((UINTN) CurrentFile - (UINTN) mFvHeader + GetFfsFileLength(CurrentFile) + 0x07) & (-1 << 3)) + (UINT8 *) mFvHeader);

lersek, 2016-02-03:
Okay, since the first instance of this bug is UDK-only, but the second instance (in BaseTools) is master as well, I think we'll have to split this report into two.

lersek, 2016-02-03:
Okay, so this bug report will stand for: MdePkg only, and for the UDK branches only. Setting tags accordingly.

closed 2022-11-29 by mdkinney