tianocore/edk2 #31: Potential use-after-free in ParsePath (StdLib - UDK2014.SP1 branch)

bcran, 2015-11-25:
On Path.c:360:
```
free(MPath); /* Caller doesn't want it so let MPath go free */
```

But since `MPath` isn't set to NULL, the else statement on line 379 can be executed which then uses `MPath`:
```
/* This is a mapped path. */
Status = __DevSearch( MPath, NULL, &Node);
```

lersek, 2016-02-02:
The filename is `StdLib/LibC/Uefi/Devices/Utility/Path.c`, function `ParsePath()`. The report applies to edk2 master as of git commit dd9044b310ae2be2ae989a522979feee86b39cc8. Removing the UDK reference from the subject.

lersek, 2016-02-03:
So this applies to UDK and current master as well. Setting the right tags for this. (Note, UdkOnly will not be set.)

closed 2022-11-29 by mdkinney