tianocore/edk2 #34: GCC toolchain doesn't enable optimizations for X64 RELEASE builds

bcran, 2015-12-07:
There are a few places `-Os` is present in tools_def.txt, but none for X64 RELEASE GCC. Since the MSFT toolchains _do_ have optimizations enabled in release builds, it seems this is a bug.

mytbk, 2015-12-11:
I used -Os in tools_def.txt before, but gcc fails to build because -Os has some conflicts with (ms_abi), so there's no -Os for gcc.

bcran, 2015-12-11:
I'm using -Os successfully with gcc 4.9 at least.

lersek, 2016-02-01:
gcc-4.8 accepts `-Os` if `-maccumulate-outgoing-args` is specified in addition, but unfortunately it also misbuilds edk2 that way. Sergey Isakov reported however that gcc-4.9 worked for him, so that seems consistent with @bcran's comment https://github.com/tianocore/edk2/issues/34#issuecomment-163924630. See this thread:

http://thread.gmane.org/gmane.comp.bios.tianocore.devel/10963

This is quite unfortunate, because `-Os` brings significant space savings.

lersek, 2016-02-02:
@bcran since this appears to apply to edk2 master as well, I propose to remove the UDK reference from the title. Thanks.

bcran, 2016-02-03:
@lersek I added the UDK reference to indicate that it wasn't a new bug - because it might be nice to have fixes committed to other branches such as UDK2014.SP1 and UDK2015 in addition to master.

lersek, 2016-02-03:
@bcran point taken, setting tags.

closed 2022-11-29 by mdkinney