tianocore/edk2 #37: Compilation error for MdeModulePkg

fjullien, 2015-12-14:
Architecture(s)  = IA32
Build target     = DEBUG
Toolchain        = GCC44

Active Platform          = /home/franck/travail/uefi/edk2/MdeModulePkg/MdeModulePkg.dsc
Processing meta-data ............... done!
.......
Building ... /home/franck/travail/uefi/edk2/MdeModulePkg/Library/DxeFileExplorerProtocol/DxeFileExplorerProtocol.inf [IA32]
"gcc" -g -fshort-wchar -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -ffunction-sections -fdata-sections -c -include AutoGen.h -fno-common -DSTRING_ARRAY_NAME=DxeFileExplorerProtocolStrings -m32 -malign-double -fno-stack-protector -D EFI32 -fno-asynchronous-unwind-tables -Os -o /home/franck/travail/uefi/edk2/Build/MdeModule/DEBUG_GCC44/IA32/MdeModulePkg/Library/DxeFileExplorerProtocol/DxeFileExplorerProtocol/OUTPUT/./DxeFileExplorerProtocol.obj -I/home/franck/travail/uefi/edk2/MdeModulePkg/Library/DxeFileExplorerProtocol -I/home/franck/travail/uefi/edk2/Build/MdeModule/DEBUG_GCC44/IA32/MdeModulePkg/Library/DxeFileExplorerProtocol/DxeFileExplorerProtocol/DEBUG -I/home/franck/travail/uefi/edk2/MdePkg -I/home/franck/travail/uefi/edk2/MdePkg/Include -I/home/franck/travail/uefi/edk2/MdePkg/Include/Ia32 -I/home/franck/travail/uefi/edk2/MdeModulePkg -I/home/franck/travail/uefi/edk2/MdeModulePkg/Include /home/franck/travail/uefi/edk2/MdeModulePkg/Library/DxeFileExplorerProtocol/DxeFileExplorerProtocol.c
In file included from /home/franck/travail/uefi/edk2/MdeModulePkg/Library/DxeFileExplorerProtocol/DxeFileExplorerProtocol.c:24:
/home/franck/travail/uefi/edk2/MdeModulePkg/Include/Library/FileExplorerLib.h:30: error: redefinition of typedef 'CHOOSE_HANDLER'
/home/franck/travail/uefi/edk2/MdeModulePkg/Include/Protocol/FileExplorer.h:39: note: previous declaration of 'CHOOSE_HANDLER' was here
make: *** [/home/franck/travail/uefi/edk2/Build/MdeModule/DEBUG_GCC44/IA32/MdeModulePkg/Library/DxeFileExplorerProtocol/DxeFileExplorerProtocol/OUTPUT/DxeFileExplorerProtocol.obj] Error 1

lersek, 2016-01-22:
@fjullien are you still seeing this?

fjullien, 2016-01-24:
Hi, sorry but I don't have a IA32 machine anymore. Plus I don't work with edk2 at the moment.
So feel free to close the issue. Thanks anyway.

jljusten, 2016-01-24:
I can reproduce the failure with IA32 on GCC 4.4 (GCC44 toolchain), but not GCC 5 (GCC49 toolchain).

lersek, 2016-01-25:
I think the code is valid, both `MdeModulePkg/Include/Protocol/FileExplorer.h` and `MdeModulePkg/Include/Library/FileExplorerLib.h` have the exact same typedef:

    typedef
    BOOLEAN
    (EFIAPI *CHOOSE_HANDLER)(
      IN EFI_DEVICE_PATH_PROTOCOL  *FilePath
      );

It's probably a gcc-4.4 bug.

But, it's not hard to fix in edk2 I think, just replace the typedef in "MdeModulePkg/Include/Library/FileExplorerLib.h" with the inclusion of "MdeModulePkg/Include/Protocol/FileExplorer.h".

lersek, 2016-01-25:
... I guess the next person who runs into this will be asked to submit the patch. :)

closed 2022-11-29 by mdkinney