tianocore/edk2 #42: ASSERT failure on QEMU

Vogtinator, 2016-01-08:
Booting GRUB2 from HDD, running the command "exit" restarts the VM, but it enters an infinite crash loop with the following error:

ASSERT /home/abuild/rpmbuild/BUILD/ovmf-0.1+svn18393/OvmfPkg/PlatformPei/MemDetect.c(294): (MtrrSettings.MtrrDefType & 0x00000800) == 0

Apparently QEMU does not reset the Mtrrs properly. I found https://es.osdn.jp/projects/tinyvisor/scm/hg/tinyvisor-uefi/commits/c85a06e7b0de5a77d68a60a5372999de02eac31a and it seems to fix it.

lersek, 2016-01-22:
When I run the `exit` command in grub2, it does not restart the VM. It exits to the Boot Manager built into OVMF, which continues boot option processing.

At that point, I even re-set the VM using the `reset -c` UEFI shell command. Not seeing any issues.

Maybe this is related to the recent KVM-side MTRR handling changes. I suggest to retest this on your end with a 4.4-rc7+ host kernel. Thanks.

Vogtinator, 2016-01-22:
> When I run the exit command in grub2, it does not restart the VM.

Yeah, I mixed that up with a different bug somewhere else. Sorry for confusion.
It's easily reproducable by "reboot" in linux or simply using virsh reset.

> Maybe this is related to the recent KVM-side MTRR handling changes. I suggest to retest this on your end with a 4.4-rc7+ host kernel. Thanks.

Confirmed, with kernel 4.4.0 it works.

lersek, 2016-01-22:
@Vogtinator thanks much for the quick feedback!
@jljusten let's close this please. I think it's another dupe of #21 after all. Thanks!

Vogtinator, 2016-01-22:
> @jljusten let's close this please. I think it's another dupe of #21 after all. Thanks!

I can do that as well :)

closed 2016-01-22 by Vogtinator

lersek, 2016-07-21:
This (closed) item has been manually migrated to
https://tianocore.acgmultimedia.com/show_bug.cgi?id=77