tianocore/edk2 #44: Can't build EDK II BaseTools

alex3kov, 2016-01-19:
I'm on fresh Arch Linux, with dependencies `base-devel glibc iasl python2 subversion` installed.  
Here's what I'm getting when running `make -C edk2/BaseTools`:  
```
make[1]: Entering directory '/home/username/code/edk2/BaseTools/Source/Python'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/username/code/edk2/BaseTools/Source/Python'
make -C Tests
make[1]: Entering directory '/home/username/code/edk2/BaseTools/Tests'
Traceback (most recent call last):
  File "RunTests.py", line 22, in <module>
    import TestTools
  File "/home/username/code/edk2/BaseTools/Tests/TestTools.py", line 94
    print description, '(base64 encoded):'
                    ^
SyntaxError: invalid syntax
GNUmakefile:17: recipe for target 'test' failed
make[1]: *** [test] Error 1
make[1]: Leaving directory '/home/username/code/edk2/BaseTools/Tests'
GNUmakefile:25: recipe for target 'Tests' failed
make: *** [Tests] Error 2
make: Leaving directory '/home/username/code/edk2/BaseTools'
```
Python 3 is also installed (I need it for other things). `python` executable points to /usr/bin/python3. I tried setting `alias python='/usr/bin/python2'`. If python3 is the problem - is there a way to pass a path to python2 to `make`?

lersek, 2016-01-19:
Although Python 2 is not spelled out anywhere I can see as a specific requirement, there are several files that imply it (`BaseTools/BuildNotes.txt`, `BaseTools/gcc/README.txt`, etc.)

The `alias` utility does not affect the environment of shell scripts that you do not `source`; please refer to http://pubs.opengroup.org/onlinepubs/9699919799/utilities/alias.html .

I recommend that you create a separate subdirectory somewhere in your home directory. There, create a symlink called `python` to the python2 interpreter. Do not add this subdir to your `PATH` env var generally, but do *prepend* it to `PATH` when working with edk2.

I'm not a BaseTools maintainer, so this is just best effort advice.

alex3kov, 2016-01-19:
Ok, that worked, thanks.  
Not the most graceful way of doing things though :) In Qemu build it's just `./configure --python=/usr/bin/python2`

jljusten, 2016-01-20:
Yes, the tools only work with python2 today. (2.7 is the 'supported' version)

I think we should see if we can clean them up to work with both python 2.7 and python 3, but that would be a pretty significant effort.

closed 2016-01-20 by jljusten

1822871719, 2023-01-03:
Traceback (most recent call last):
  File "/home/zqx/downloads/edk2/BaseTools/Tests/CheckPythonSyntax.py", line 49, in <lambda>
    newmethod = lambda self: self.SingleFileTest(filename)
  File "/home/zqx/downloads/edk2/BaseTools/Tests/CheckPythonSyntax.py", line 27, in SingleFileTest
    self.fail('syntax error: %s, Error is %s' % (filename, str(e)))
AssertionError: syntax error: /home/zqx/downloads/edk2/BaseTools/Source/Python/FMMT/core/GuidTools.py, Error is   File "/home/zqx/downloads/edk2/BaseTools/Source/Python/FMMT/core/GuidTools.py", line 23
    self.guid: str = guid
             ^
SyntaxError: invalid syntax