tianocore/edk2 #46: Error when changing boot order after deleting boot option [UDK2014.SP1 and newer]

bcran, 2016-01-25:
The following error is displayed when the `Change Boot Order` menu is entered after deleting a boot option:
> Question value mismatch with Option value!
>  Press ENTER to continue

Steps to replicate:

Enter Setup menu
`Boot Maintenance Manager` => `Boot Options` => `Delete Boot Option`
Select any boot option to delete, and then choose `Commit Changes and Exit`
Now, choose `Change Boot Order`

_Reported on the edk2 mailing list in March 2015. Re-reporting since I suspect old bug reports were forgotten._

lersek, 2016-02-02:
@bcran can you please try to reproduce with current master? For now I'll tag this as UDK.

bcran, 2016-02-03:
@lersek the problem exists in UDK2014.SP1 and newer including current master.

lersek, 2016-02-03:
@bcran thanks for following up, I shall drop the UDK label I think.

(This is indeed quite messy, sorry. In a sensible bug tracker, the stable and the development products would be separate. If both products contained the bug, then both would have separate entries in the issue tracker, and those entries would be linked by *cloning*. The separate entries also allow for the occasionally necessary different implementations of the same fix, across the stable branches.

Here however I'm hard pressed to capture the fact that the bug is present on several branches. Should we introduce a separate label for each possible branch? `UDK2015`, `UDK2014.SP1`, and `master`? But, `master` would be overkill, since most bugs would be reported for `master`. Then again, some bugs could be reported for one of the released products *only*, so the *absence* of master should be indicated. Sigh. Let me try to come up with some more labels...)

bcran, 2016-02-03:
What we do at $work is have just a single bug ticket that has labels for each codename affected. But that's using JIRA, which can display information in a far more flexible format than Github, where adding 5 labels might cause excessive clutter.

closed 2022-11-29 by mdkinney