tianocore/edk2 #47: OVMF release build with GCC 5.2 runs out of space [master]

bcran, 2016-01-25:
When OVMF from github master (as of 2016-01-24) is built using `./OvmfPkg/build.sh -a X64 -t GCC49 -b RELEASE`, the build fails because the image is too big to fit in the 1MB rom. 

Passing in `-DFD_SIZE_2MB` to use a 2MB rom works around the problem.

lersek, 2016-01-25:
Please refer to the discussion at http://thread.gmane.org/gmane.comp.bios.edk2.devel/6885

One idea is to make the 2MB FD size the default, regardless of building a DEBUG vs. RELEASE target. Users of older toolchains where 1MB remains usable at all will have to specify `-D FD_SIZE_1MB` manually.

Alternatively, I guess we could set the defaults based on toolchain tag, but that would be horrible to maintain, and also very non-intuitive for users. Moving between gcc versions would unexpectedly change their flash image size. The current FD size selection, which depends on DEBUG vs. RELEASE, is already a bit unintuitive.

"One size fits all" would be better I think.

bcran, 2016-01-26:
Oh, I did see that. My point of creating this bug report was your answer
> You are using incorrect build options.

I'd argue that people should expect `build.sh` to work without knowing to use `-DFD_SIZE_2MB`.

lersek, 2016-01-27:
@bcran: pinged @jljusten in http://thread.gmane.org/gmane.comp.bios.edk2.devel/6889/focus=6952

closed 2016-01-29 by jljusten

lersek, 2016-07-21:
This (closed) item has been manually migrated to
https://tianocore.acgmultimedia.com/show_bug.cgi?id=78