tianocore/edk2 #49: shifting a negative signed value is undefined

lersek, 2016-02-03:
@ajfish 

cloned from https://github.com/tianocore/edk2/issues/30#issuecomment-178893905

> It is compiler defined behavior in the older world, and that is why clang has added the warning. I also see the error in the base Tools as it has a case of shifting -1.
> 
> https://github.com/tianocore/edk2/blob/master/BaseTools/Source/C/Common/FvLib.c
> 
>     //
>     // Get next file, compensate for 8 byte alignment if necessary.
>     //
>     *NextFile = (EFI_FFS_FILE_HEADER *) ((((UINTN) CurrentFile - (UINTN) mFvHeader + GetFfsFileLength(CurrentFile) + 0x07) & (-1 << 3)) + (UINT8 *) mFvHeader);

lersek, 2016-05-17:
Patch on list:
http://thread.gmane.org/gmane.comp.bios.edk2.devel/12270

closed 2016-05-18 by lgao4

ajfish, 2016-05-18:
> Closed #49 <https://github.com/tianocore/edk2/issues/49> via d3bb711 <https://github.com/tianocore/edk2/commit/d3bb711834acd3eda35a07d0be7911bc3dbb9e6f>.

Thanks,

Andrew Fish