tianocore/edk2 #50: Incorrect MBR primary partition numbering in PartitionInstallMbrChildHandles()

iliyap, 2016-02-08:
UEFI spec says that MBR primary partitions should be numbered by their MBR partition table entry index [1..4]. Current implementation gives primary partitions consecutive numbers, skipping type 0x00, type 0xEE, and zero sized partitions. Implementation should give partitions numbers based on entry index. I.e. `Index + 1` instead of `PartitionNumber++`.

lersek, 2016-02-08:
MdeModulePkg/Universal/Disk/PartitionDxe

niruiyu, 2016-02-14:
I didn’t find clear statement in UEFI spec to require the partition number equal to table entry index. Can you kindly point it for me?

Regards,
Ray

iliyap, 2016-02-14:
UEFI Specification 2.3.1 Errata D, Section 9.3.6.1, first paragraph:

Partitions are numbered according to their entry in their respective partition table, starting with 1.

Current edk2 code for GPT partitions do number partitions according to their index in partition table. Code for MBR primary partitions doesn’t but I think should.

От: Ruiyu Ni
Отправлено: ‎воскресенье‎, ‎14‎ ‎февраля‎ ‎2016‎ г. ‎12‎:‎49
Кому: tianocore/edk2
Копия: Iliya Peregoudov

I didn’t find clear statement in UEFI spec to require the partition number equal to table entry index. Can you kindly point it for me? 

Regards, 
Ray

niruiyu, 2016-02-18:
How about the patch below?
UEFI Spec 2.3.1.D Section 9.3.5.1 says Partitions are numbered
according to their entry in their respective partition table,
starting with 1.
Update the code for MBR partition to follow this rule.

Contributed-under: TianoCore Contribution Agreement 1.0

Cc: Feng Tian <feng.tian@intel.com>
---
 MdeModulePkg/Universal/Disk/PartitionDxe/Mbr.c | 10 ++++------
 1 file changed, 4 insertions(+), 6 deletions(-)

diff --git a/MdeModulePkg/Universal/Disk/PartitionDxe/Mbr.c b/MdeModulePkg/Universal/Disk/PartitionDxe/Mbr.c
index 4b0159c..1946334 100644
--- a/MdeModulePkg/Universal/Disk/PartitionDxe/Mbr.c
+++ b/MdeModulePkg/Universal/Disk/PartitionDxe/Mbr.c
@@ -12,7 +12,7 @@
         the legacy boot strap code.
 
 Copyright (c) 2014, Hewlett-Packard Development Company, L.P.<BR>
-Copyright (c) 2006 - 2013, Intel Corporation. All rights reserved.<BR>
+Copyright (c) 2006 - 2016, Intel Corporation. All rights reserved.<BR>
 This program and the accompanying materials
 are licensed and made available under the terms and conditions of the BSD License
 which accompanies this distribution.  The full text of the license may be found at
@@ -136,7 +136,6 @@ PartitionInstallMbrChildHandles (
   HARDDRIVE_DEVICE_PATH     HdDev;
   HARDDRIVE_DEVICE_PATH     ParentHdDev;
   EFI_STATUS                Found;
-  UINT32                    PartitionNumber;
   EFI_DEVICE_PATH_PROTOCOL  *DevicePathNode;
   EFI_DEVICE_PATH_PROTOCOL  *LastDevicePathNode;
   UINT32                    BlockSize;
@@ -192,8 +191,6 @@ PartitionInstallMbrChildHandles (
     }
   }
 
-  PartitionNumber = 1;
-
   ZeroMem (&HdDev, sizeof (HdDev));
   HdDev.Header.Type     = MEDIA_DEVICE_PATH;
   HdDev.Header.SubType  = MEDIA_HARDDRIVE_DP;
@@ -223,7 +220,7 @@ PartitionInstallMbrChildHandles (
         continue;
       }
 
-      HdDev.PartitionNumber = PartitionNumber ++;
+      HdDev.PartitionNumber = Index + 1;
       HdDev.PartitionStart  = UNPACK_UINT32 (Mbr->Partition[Index].StartingLBA);
       HdDev.PartitionSize   = UNPACK_UINT32 (Mbr->Partition[Index].SizeInLBA);
       CopyMem (HdDev.Signature, &(Mbr->UniqueMbrSignature[0]), sizeof (Mbr->UniqueMbrSignature));
@@ -252,6 +249,7 @@ PartitionInstallMbrChildHandles (
     // It's an extended partition. Follow the extended partition
     // chain to get all the logical drives
     //
+    Index             = 0;
     ExtMbrStartingLba = 0;
 
     do {
@@ -277,7 +275,7 @@ PartitionInstallMbrChildHandles (
         ExtMbrStartingLba = UNPACK_UINT32 (Mbr->Partition[0].StartingLBA);
         continue;
       }
-      HdDev.PartitionNumber = PartitionNumber ++;
+      HdDev.PartitionNumber = ++Index;
       HdDev.PartitionStart  = UNPACK_UINT32 (Mbr->Partition[0].StartingLBA) + ExtMbrStartingLba + ParentHdDev.PartitionStart;
       HdDev.PartitionSize   = UNPACK_UINT32 (Mbr->Partition[0].SizeInLBA);
       if ((HdDev.PartitionStart + HdDev.PartitionSize - 1 >= ParentHdDev.PartitionStart + ParentHdDev.PartitionSize) ||
-- 
2.7.0.windows.1

lersek, 2016-02-18:
Patch was posted at http://thread.gmane.org/gmane.comp.bios.edk2.devel/7741

closed 2022-11-29 by mdkinney