tianocore/edk2 #52: Possible bad data length check in EslSocketTransmit

ColinIanKing, 2016-02-09:
The following check in  EslSocketTransmit(), StdLib/EfiSocketLib/Socket.c looks a bit bogus to me:

         if (( NULL == pDataLength )
            && ( 0 > pDataLength )
            && ( NULL == pBuffer )) {
            if ( NULL == pDataLength ) {

I think it should be:

         if (( NULL == pDataLength )
            && ( 0 > *pDataLength )
            && ( NULL == pBuffer )) {
            if ( NULL == pDataLength ) {

This was flagged up as a potential error using static analysis:

[StdLib/EfiSocketLib/Socket.c:5369]: (warning) Logical conjunction always evaluates to false: pDataLength == 0 && pDataLength < 0.

AllenWynn, 2016-06-30:
Also, seems like the check should be OR rather than AND, like:
     if (( NULL == pDataLength )
        || ( 0 > *pDataLength )
        || ( NULL == pBuffer )) {
        if ( NULL == pDataLength ) {

nbuwe, 2016-07-11:
Since *pDataLength is unsigned (size_t), the check doesn't make sense anyway.

Also note that debug messages just below use (copy/paste, obviously) DEBUG_RX instead of DEBUG_TX.

jcarsey, 2016-07-11:
maybe the data length should be non-zero, but not evaluated < or >?

closed 2022-11-29 by mdkinney