tianocore/edk2 #56: Conf/tools_def.txt gets parsed every time a GUIDed tool is searched for

ajfish, 2016-02-16:
I updated BaseTools/BinWrappers/PosixLike/RunToolFromSource to dump performance data by turning on cProfile in Python:
`python` -m cProfile -s time "`dirname $0`/../../Source/Python"/`basename $0`/`basename $0`.py `$*`

I found that ToolDefClassObject.LoadToolDefFile() was getting called 47 times and it looks like the issue is a new instance of ToolDefClassObject gets created every time a search is made for an external tool for a GUIDed encapsulation section. This seems to have the side effect of re-parsing the Conf/tools_def.txt and this is a slow operation. 

Note: A workaround is to make the Conf/tools_def.txt only contain the toolchains you are actively using. This means you only have to parse ~150 lines vs. over 7,000 lines of text.

ajfish, 2016-02-18:
On my system parsing tools_def.txt takes about 0.18 seconds, so 8 1/2 seconds total. tools_def.txt is 463K, so that is why it is so slow to parse.

STLVNUB, 2016-04-12:
Another good reason for tools_def.txt to more manageable by way of making it modular

macplox, 2016-04-24:
Conf/tools_def.txt is already sectioned off into toolchain categories, but theres no benefit to this unless it's split up into Conf/tools_def_ARCH.txt or similar and parse each file only if ARCH was mentioned in the list defining TARGET_ARCH inside the Conf/target.txt. i don't know how many of us regularly build for all of the architectures every time we build, but those who don't will save time by not looking for things in places implicitly irrelevant to their target.

ajfish, 2016-06-29:
Fixed in 213ae07750fc7532e8ba62a234e958388d1f0359

closed 2016-06-29 by ajfish