tianocore/edk2 #59: OVMF 64MMIO / "above 4G decoding" for Tesla pass through

GamerSource, 2016-02-22:
Hi,

what is the state when I'd try to "PCI - pass-trough" a powerful graphic card (like the nvidia tesla K80)
to qemu with OVMF as UEFI/BIOS?

On a physical machine I may have to enable 64-bit MMIO (or like it's often called "above 4G decoding"), how is that with OVMF, as far as I have seen there is no such option in OVMF also I'm currently not sure what the state in qemu itself is I only found an old mailing post [1] regarding this issue and posted a week ago on there user/discuss list but received sadly no response yet.

So I'm hoping someone could point me in the right direction.
If this is the wrong place to ask I'm glad to move the question to an more appropriate one, but I did not found any mailing/discuss list for OVMF.

[1] https://lists.nongnu.org/archive/html/qemu-devel/2010-02/msg01106.html

lersek, 2016-02-22:
Hello @GamerSource, this is the right place -- or *one* right place anyway -- to ask.

QEMU definitely supports 64-bit MMIO for the x86_64 target.

The PCI host bridge / root bridge driver that OVMF uses at the moment lacks support for 64-bit PCI MMIO however. This is a known limitation, and it should be resolved *after* OVMF is ported to the new, central PCI host bridge / root bridge driver in edk2. (It won't happen automatically with the port, but when the port is complete, it shouldn't take much work on top.)

I started to work on that port recently, but ran into some issues with the central driver that OVMF should use. We've been discussing those topics recently on the mailing list, with @marcel-apf and @niruiyu .

Please refer to these (sub)threads in the list archive:
* http://thread.gmane.org/gmane.comp.bios.edk2.devel/6526/focus=6910
* http://thread.gmane.org/gmane.comp.bios.edk2.devel/7382
* http://thread.gmane.org/gmane.comp.bios.edk2.devel/7967
* http://thread.gmane.org/gmane.comp.bios.edk2.devel/7969

GamerSource, 2016-02-23:
Hello @lersek thanks for the response, really appreciate it!

Ok cool, good to know, I guessed it had the capabilities but did not found that much info on it and I'm not (yet) that sound on skimming through qemu code to see what's there.

Thanks for the write up and the thread links, I guess I found a new interesting mailing list to subscribe.

As this is then really an "issue" with OVMF I feel that its probably correct to not close this here on github yet, for sake of documenting and helping maybe someone other to find out the cause why it's not working yet, feel free to do otherwise.

Thanks again!

lersek, 2016-02-23:
This is a genuine, although likely not too urgent, shortcoming in OVMF, at the moment, so we should certainly keep this item open.

lersek, 2016-03-10:
Posted series

http://thread.gmane.org/gmane.comp.bios.edk2.devel/9030

lersek, 2016-03-10:
@GamerSource: I CC'd you on the patches and the cover letter (googling a combination of your name and your employer's name, taken from your github profile page, for your email). In case the emails don't reach you for any reason, I'm asking you here too to please test the patches with your hardware. Thanks!

lersek, 2016-03-14:
The patch series for this issue is now "wave 2" in a larger work:

http://thread.gmane.org/gmane.comp.bios.edk2.devel/9150

GamerSource, 2016-03-14:
Thanks! I'm following your posts on the edk2 mailing list closely. I'll rebuild and send it out to test, we have the K80 currently not available so I'm dependent of an other tester...

And sorry for not providing my email address, nice detective work, though! :)
Thanks for the work!

lersek, 2016-03-23:
Fixed by commits 8f35eb92c419..4f5eff819309.

closed 2016-03-23 by lersek

lersek, 2016-07-21:
This (closed) item has been manually migrated to
https://tianocore.acgmultimedia.com/show_bug.cgi?id=80