tianocore/edk2 #62: Use MdeModulePkg BDS in OVMF

jljusten, 2016-03-03:
OVMF should migrate to use MdeModulePkg/Universal/BdsDxe rather than IntelFrameworkModulePkg/Universal/BdsDxe.

lersek, 2016-03-03:
See the previous discussion at:

http://thread.gmane.org/gmane.comp.bios.edk2.devel/759

lersek, 2016-04-25:
v3 series from @niruiyu : http://thread.gmane.org/gmane.comp.bios.edk2.devel/10859/focus=11032

lersek, 2016-05-02:
v4 series from Ray: http://news.gmane.org/find-root.php?message_id=1461913264-137700-1-git-send-email-ruiyu.ni@intel.com

lersek, 2016-05-03:
v5 series from Ray: http://thread.gmane.org/gmane.comp.bios.edk2.devel/11566

closed 2016-05-04 by niruiyu

lersek, 2016-05-04:
To be clear, the commit range for the full series is d0a23f9f86dc..1a4232b741da.

lersek, 2016-07-21:
This (closed) issue has been manually migrated to
https://tianocore.acgmultimedia.com/show_bug.cgi?id=81