tianocore/edk2 #75: [ArmVirtPkg] aarch64 assert failure in ArmPlatformPrePeiCore PrePeiCore.c(83)

bcran, 2016-03-28:
Using edk2 git master as of 2016-03-27 (877c0a93be4317f5715347359cc78d41f4654748) a debug AARCH64 build of ArmVirtPkg (using gcc  5.3.1) fails to run, with the assert failure:
```
ASSERT [ArmPlatformPrePeiCore] /home/bcran/workspace/edk2/ArmPlatformPkg/PrePeiCore/PrePeiCore.c(83): ((UINTN)PeiVectorTable & ((1 << 11)-1)) == 0
```

The command to launch it (with qemu 2.5.50) is:
```
qemu-system-aarch64 -m 1G -cpu cortex-a57 -M virt -bios Build/ArmVirtQemu-AARCH64/DEBUG_GCC49/FV/QEMU_EFI.fd -serial stdio
```

lersek, 2016-03-29:
@bcran: This is weird -- could you perhaps bisect this issue? I assume it might have something to do with the recent ArmPkg changes.

@ardbiesheuvel: any ideas?

Thanks.

ardbiesheuvel, 2016-03-29:
This has to do with the alignment of the vector table. It needs to appear at a 2 KB aligned boundary

Is your tools_def.txt up to date? The DEBUG_GCC49 toolchain definition was recently updated to switch to the small model, since the UEFI shell could no longer be built in DEBUG mode using GCC49's default tiny model.

Other than that, it makes sense to manually inspect ArmPlatformPkg/PrePeiCore/PrePeiCoreUniCore/DEBUG/ArmPlatformPrePeiCore.dll using readelf, to check whether the 2 KB alignment is correct for the .text section, and please check the debug output whether this module appears at a suitably aligned offset in the final image.

bcran, 2016-03-29:
Sorry, it apparently was due to stale build files: removing and regenerating files in the `Conf` directory resolved the problem.

closed 2016-03-29 by bcran