tianocore/edk2 #77: tools_def.txt Enhancement

STLVNUB, 2016-04-09:
Hi. Not really an issue, but it is, IMHO

The tools_def file is growing rather large, so I propose to split it into sub TOOL_CHAIN files.
i.e 
Make all Tool chains seperate files.
e.g
VS_TOOLS.txt: Contains ALL VS data
GCC_TOOLS.txt: Contains ALL GCC data
ICC_TOOLS.txt: etc
UNIXGCC.txt: etc

Then the tools_def.txt will be quite small and the other files will be easier to manage.

Also 
Defining  a NASM and MAKE variables for custom building.
e.g  building on OS X using GCC49, nasm and make will fail.

*_*_*_NASM_PATH                = nasm
change to
*_*_*_NASM_PATH                = ENV(NASM_PREFIX)/nasm

*_GCC49_*_MAKE_PATH      = DEF(GCC44_IA32_PREFIX)make
change to
*_GCC44_*_MAKE_PATH      = make

nms42, 2016-04-11:
Some kind of 'include' (for the benefit of common code) would be good ;-)

Download-Fritz, 2016-05-14:
Michael Zimmermann just submited a patch that allows !include in tools_def. :)

closed 2022-11-29 by mdkinney