tianocore/edk2 #80: OVMF fails to boot with SMP enabled

jboeing, 2016-04-16:
If I enable SMP in my VM, QEMU either hangs at boot or dies with the error message:
`
KVM: entry failed, hardware error 0x80000021`

I git-bisected the failure to commit 94941c8: UefiCpuPkg: CpuDxe: broadcast MTRR changes to APs

The host is running Linux kernel 4.4.6 and QEMU 2.5.0.  The host is using GCC 5.3.0; I get the same failure whether I build OVMF with GCC 4.9.3 or 5.3.0.  A minimal QEMU command line to repro is:

qemu-system-x86_64 -enable-kvm -smp cpus=2 -drive if=pflash,format=raw,file=OVMF.fd

Here's the output from the debug build of OVMF:

```
SecCoreStartupWithStack(0xFFFCC000, 0x818000)
Register PPI Notify: DCD0BE23-9586-40F4-B643-06522CED4EDE
Install PPI: 8C8CE578-8A3D-4F1C-9935-896185C32DD3
Install PPI: 5473C07A-3DCB-4DCA-BD6F-1E9689E7349A
The 0th FV start address is 0x00000820000, size is 0x000E0000, handle is 0x820000
Register PPI Notify: 49EDB1C1-BF21-4761-BB12-EB0031AABB39
Register PPI Notify: EA7CA24B-DED5-4DAD-A389-BF827E8F9B38
Install PPI: B9E0ABFE-5979-4914-977F-6DEE78C278A6
Install PPI: DBE23AA9-A345-4B97-85B6-B226F1617389
Loading PEIM at 0x00000839BC0 EntryPoint=0x00000839E00 PcdPeim.efi
Install PPI: 06E81C58-4AD7-44BC-8390-F10265F72480
Install PPI: 01F34D25-4DE2-23AD-3FF3-36353FF323F1
Install PPI: 4D8B155B-C059-4C8F-8926-06FD4331DB8A
Install PPI: A60C6B59-E459-425D-9C69-0BCC9CB27D81
Loading PEIM at 0x000008435C0 EntryPoint=0x00000843800 StatusCodePei.efi
Install PPI: 229832D3-7A30-4B36-B827-F40CB7D45436
Loading PEIM at 0x000008491C0 EntryPoint=0x00000849400 PlatformPei.efi
Select Item: 0x0
FW CFG Signature: 0x554D4551
Select Item: 0x1
FW CFG Revision: 0x3
QemuFwCfg interface is supported.
Platform PEIM Loaded
CMOS:
00: 25 00 17 00 21 00 07 16 04 16 26 02 00 80 00 00
10: 40 00 00 00 07 80 02 FF FF 00 00 00 00 00 00 00
20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
30: FF FF 20 00 00 07 00 20 30 00 00 00 00 12 00 00
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Select Item: 0x19
Select Item: 0x25
S3 support was detected on QEMU
Install PPI: 7408D748-FC8C-4EE6-9288-C4BEC092A410
PublishPeiMemory: mPhysMemAddressWidth=36 PeiMemoryCap=65800 KB
PeiInstallPeiMemory MemoryBegin 0x3FBE000, MemoryLength 0x4042000
QemuInitializeRam called
Allocated Memory unaligned: Address = 0x7FD0000, Pages = 0x30, Type = 6 
After aligning to 0x10000 bytes: Address = 0x7FD0000, Pages = 0x20 
Updated aligned-mem HOB with BaseAddress = 7FD0000, Length = 20000, MemoryType = 6 
Created after-mem HOB with BaseAddress = 7FF0000, Length = 10000, MemoryType = 7 
Reserved variable store memory: 0x7FD0000; size: 128kb
Platform PEI Firmware Volume Initialization
Install PPI: 49EDB1C1-BF21-4761-BB12-EB0031AABB39
Notify: PPI Guid: 49EDB1C1-BF21-4761-BB12-EB0031AABB39, Peim notify entry point: 82491E
The 1th FV start address is 0x00000900000, size is 0x00900000, handle is 0x900000
Select Item: 0x19
Select Item: 0x19
Temp Stack : BaseAddress=0x814000 Length=0x4000
Temp Heap  : BaseAddress=0x810000 Length=0x19E0
Total temporary memory:    32768 bytes.
  temporary memory stack ever used: 16384 bytes.
  temporary memory heap used:       6624 bytes.
Old Stack size 16384, New stack size 131072
Stack Hob: BaseAddress=0x3FBE000 Length=0x20000
[... 121 lines cut ...]
If you're running a guest on an Intel machine without unrestricted mode
support, the failure can be most likely due to the guest entering an invalid
state for Intel VT. For example, the guest maybe running in big real mode
which is not supported on less recent Intel processors.

EAX=00000000 EBX=00000000 ECX=00000000 EDX=00000600
ESI=00000000 EDI=00000000 EBP=00000000 ESP=00000000
EIP=00000000 EFL=00000002 [-------] CPL=0 II=0 A20=1 SMM=0 HLT=0
ES =0000 00000000 0000ffff 00009300
CS =9f00 0009f000 0000ffff 00009b00
SS =0000 00000000 0000ffff 00009300
DS =0000 00000000 0000ffff 00009300
FS =0000 00000000 0000ffff 00009300
GS =0000 00000000 0000ffff 00009300
LDT=0000 00000000 0000ffff 00008200
TR =0000 00000000 0000ffff 00008b00
GDT=     00000000 0000ffff
IDT=     00000000 0000ffff
CR0=60000010 CR2=00000000 CR3=07efa000 CR4=00000000
DR0=0000000000000000 DR1=0000000000000000 DR2=0000000000000000 DR3=0000000000000000 
DR6=00000000ffff0ff0 DR7=0000000000000400
EFER=0000000000000000
Code=<eb> 06 3f 00 d8 87 ee 07 fa b8 00 9f 8e d8 bb 02 00 3e 66 0f 01 17 66 b8 23 00 00 00 0f 22 c0 66 ea 27 f0 09 00 10 00 b8 40 06 00 00 0f 22 e0 66 b8 08
```
If booting hangs rather than hits the KVM error, the debug output is the same as above except it cuts off after the "Does not find any HOB stored CPU BIST information!" message.

jboeing, 2016-04-16:
For the record, the same setup boots fine if the only change is to use SeaBIOS instead of OVMF.

jboeing, 2016-04-16:
I forgot to mention that the host in question has a quad-core Intel Penryn CPU (Q9550S).

lersek, 2016-04-18:
Thank you for the report, even though it makes me cry. :( :( :(

Can you please grab (or build) the most recent OVMF available to you, and bisect the host kernel under it, using the stable tree? (I realize this will be horribly annoying, so thank you for your cooperation.)

I'm requesting this because to my knowledge things have been working nicely on 4.4+ host kernels. Namely, the edk2 commit you identified is not new (it is dated Oct 19, 2015). On the other hand, the 4.4.6 stable / longterm kernel is new-ish (Mar 16,2016). I would recommend (more precisely, ask you for) bisecting 4.4.0 through 4.4.6.

Thank you
Laszlo

lersek, 2016-04-18:
... This may be wishful thinking on my part, but it looks like 4.4.6 regressed other virt-related stuff as well; see for example this thread: http://thread.gmane.org/gmane.comp.emulators.qemu/406416

lersek, 2016-04-18:
$ git log --oneline --reverse v4.4.5..v4.4.6 | grep -i kvm
    c9e1bbef7e77 kvm: cap halt polling at exactly halt_poll_ns
    0bbe5fa4f79a KVM: VMX: disable PEBS before a guest entry
    78939530542f KVM: s390: correct fprs on SIGP (STOP AND) STORE STATUS
    1c463a390a89 KVM: PPC: Book3S HV: Sanitize special-purpose register values on guest exit
    68ed2ca153c7 KVM: MMU: fix ept=0/pte.u=1/pte.w=0/CR0.WP=0/CR4.SMEP=1/EFER.NX=0 combo
    1ebd29d6b940 KVM: MMU: fix reserved bit check for ept=0/CR0.WP=0/CR4.SMEP=1/EFER.NX=0

lersek, 2016-04-18:
I've also asked @bonzini for his insight wrt. this report. Thanks!

jboeing, 2016-04-18:
Sure, I'll take a shot at bisecting the host kernel.

In the meantime, I tried using the 94941c8 OVMF build with older kernels I have on the host.  The VM booted with 4.1.6 on the host, but failed with 4.2.0.

lersek, 2016-04-18:
Please don't bother testing pre-4.4.0 hosts. Please see issue 21: hosts kernels >= 4.2 and < 4.4.0 are known to break OVMF.

... Well, actually, I find it useful that you tested 4.1.6. Your successful test suggests that your hardware is "appropriate", and that 4.4.6 has reintroduced similar issues that were fixed from 4.2.0 to 4.4.0.

jboeing, 2016-04-18:
I started bisecting 4.1 to 4.2 before I saw your last comment. It was pointing somewhere in the big KVM merge commit (4e24155) for 4.2-rc1 where I left off.

After looking through #21 and the linked kernel bug (https://bugzilla.kernel.org/show_bug.cgi?id=107561), I tested 4.4.0 (and 4.4.3). I still hit the same failure, so I think I'm seeing something that was missed in the 4.3/4.4 fixes rather than a regression in the 4.4 stable series.

lersek, 2016-04-18:
Thanks for the update.

@bonzini said in <https://github.com/tianocore/edk2/issues/21#issuecomment-174916085>,

> Even with 4.4 there are some changes compared to 4.1 that may cause this bug. 4.4 only fixed the bugs, it didn't revert everything.

That seems to be consistent with your results, yes? 4.1.6 is the last one that works for you, and none of 4.4.0, 4.4.3, and 4.4.6 do. Right?

jboeing, 2016-04-18:
That's right.

jboeing, 2016-04-18:
Finishing the kernel bisect points to commit d28bc9d: `KVM: x86: INIT and reset sequences are different` as where my failure starts.  Commit 5690891: `kvm: x86: zero EFER on INIT` undid a small part of it, but otherwise it seems to be intact on the latest kernel tip.

So...would you suggest reporting a bug against KVM, or wait for @bonzini to chime in?

lersek, 2016-04-19:
Thank you very much for completing the bisection.

I have emailed Paolo directly about this item, but I think he might need two weeks (possibly more) to get to it. Normally I would recommend to just wait until he responds here, but given the time frame (and your precise results with the bisection), I recommend opening a kernel bugzilla.

(In issue #21 too, a kernel BZ got referenced (107561), so this shouldn't count as unusual.)

Once you have a BZ filed in http://bugzilla.kernel.org/, could you please paste the link here? Thank you.

BTW, my colleague Radim has recently become co-maintainer for KVM, so I'm going to ask him to review this item as well. (I'm unsure if Radim has a github ID, which is why the direct email + a kernel BZ might be best.)

Thanks!

lersek, 2016-04-19:
@jboeing, Radim looked at your report, and said

> He has Penryn, which is one of first CPUs that shipped with VMX ... Have you tried running without kvm_intel.ept and/or other parameters?

Which immediately caused me to snap my forehead -- I have already determined that you *need* EPT for this stuff to work. Please refer to the following message:

http://thread.gmane.org/gmane.comp.bios.edk2.devel/9268/focus=9406

The original topic of said thread is different / independent, but the symptoms experienced there, and the cause (lack of EPT) should be identical. Can you please verify if you have EPT support in your host CPU?

... It does remain an interesting question why after KVM commit d28bc9d, EPT became a hard requirement for running OVMF. Anyway, I'll happily defer that question to our kernel developers. @jboeing, after you please file a kernel BZ, and confirm that your phys machine does lack EPT, I believe I'd like to close this OVMF report. Thanks!

jboeing, 2016-04-19:
Penryn doesn't have EPT; that was added for Nehalem.
```
$ cat /proc/cpuinfo
...
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush
dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx lm constant_tsc arch_perfmon pebs bts
rep_good nopl aperfmperf pni dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm
sse4_1 xsave lahf_lm dtherm tpr_shadow vnmi flexpriority
```
```
$ tail /sys/module/kvm_intel/parameters/*
==> /sys/module/kvm_intel/parameters/emulate_invalid_guest_state <==
Y

==> /sys/module/kvm_intel/parameters/enable_apicv <==
N

==> /sys/module/kvm_intel/parameters/enable_shadow_vmcs <==
N

==> /sys/module/kvm_intel/parameters/ept <==
N

==> /sys/module/kvm_intel/parameters/eptad <==
N

==> /sys/module/kvm_intel/parameters/fasteoi <==
Y

==> /sys/module/kvm_intel/parameters/flexpriority <==
Y

==> /sys/module/kvm_intel/parameters/nested <==
N

==> /sys/module/kvm_intel/parameters/ple_gap <==
0

==> /sys/module/kvm_intel/parameters/ple_window <==
4096

==> /sys/module/kvm_intel/parameters/ple_window_grow <==
2

==> /sys/module/kvm_intel/parameters/ple_window_max <==
1073741823

==> /sys/module/kvm_intel/parameters/ple_window_shrink <==
0

==> /sys/module/kvm_intel/parameters/pml <==
N

==> /sys/module/kvm_intel/parameters/unrestricted_guest <==
N

==> /sys/module/kvm_intel/parameters/vmm_exclusive <==
Y

==> /sys/module/kvm_intel/parameters/vpid <==
N
```
I'm working on the kernel BZ now.

jboeing, 2016-04-19:
https://bugzilla.kernel.org/show_bug.cgi?id=116731

lersek, 2016-04-19:
> https://bugzilla.kernel.org/show_bug.cgi?id=116731

Thank you. I've CC'd @rkrcmar (and myself) on that bug, and I'm closing this one.

closed 2016-04-19 by lersek

lersek, 2016-07-21:
This (closed) item has been manually migrated to
https://tianocore.acgmultimedia.com/show_bug.cgi?id=83