tianocore/edk2 #81: RAM Disk: removal action doesn't do anything when 9 disks exist

bcran, 2016-04-18:
If a user creates 9 raw RAM Disks using the default size of 0x1000, then at some point some of them become selected ("[X]") in the list automatically. 

Furthermore, the "Remove selected RAM disk(s)." option appears to have no effect anymore - none of the disks are deleted. 

Pressing ESC to leave the screen causes a "Changes have not been saved. Save Changes and exit?" dialog to appear. Pressing 'Y' has no effect.

bcran, 2016-04-18:
Label issue as OVMF.

lersek, 2016-04-18:
Hm, I don't think it is right to label this as OVMF. OvmfPkg does not provide any code related specifically to RAM Disks. I think it would be more appropirate to attribute this to RamDiskDxe, since the exact string that you quote ("Remove selected RAM disk(s)") can be found in `MdeModulePkg/Universal/Disk/RamDiskDxe/RamDiskHiiStrings.uni`.

hwu25, 2016-04-19:
It is a bug for the RamDiskDxe driver in the MdeModulePkg. I will send out  a patch to fix this issue later.

lersek, 2016-04-20:
Patch on list:
http://thread.gmane.org/gmane.comp.bios.edk2.devel/10968

hwu25, 2016-05-10:
The issue is fixed by commit 09abc636756b5d1f29224402181f7fd34f736c5f.
@lersek Could you help to close this issue?

lersek, 2016-05-10:
@hwu25: closing.

But, as a tianocore group member, I think you should also be able to close edk2 project issues. @jljusten, any thoughts? Thanks.

closed 2016-05-10 by lersek