tianocore/edk2 #82: Csm16.bin(seabios) failed to work in OVMF.

wqlxx, 2016-04-24:
I build the Csm16.bin from source code of seabios-1.9.2.tar.gz . And copy the Csm16.bin to OvmfPkg/Csm/Csm16/Csm16.bin , Build the OVMF_CODE.fd and OVMF_VARS.fd with command build -D CSM_ENABLE.
   Unfortunately, I test the OVMF_CODE.fd and OVMF_VARS.fd in qemu. It failed!
The log is:
        ASSERT_EFI_ERROR (Status = Not Found)
        ASSERT /root/tianocore-edk2/IntelFrameworkModulePkg/Csm/LegacyBiosDxe/LegacyBios.c(982): !EFI_ERROR (Status)

lersek, 2016-04-25:
Discussion on-list: http://thread.gmane.org/gmane.comp.bios.edk2.devel/11148

lersek, 2016-05-04:
Not an OVMF bug. Please lower the size of the SeaBIOS CSM binary to e.g. 192 KB (`CONFIG_ROM_SIZE=192`). Please see the discussion linked above for details.

closed 2016-05-04 by lersek

lersek, 2016-07-21:
This (closed) item has been manually migrated to
https://tianocore.acgmultimedia.com/show_bug.cgi?id=84