tianocore/edk2 #89: BaseTools fails to compile with Xcode 7.3

theracermaster, 2016-05-15:
Compilation of BaseTools (`make -C BaseTools`) fails with Xcode 7.3 (on OS X 10.11.4) due to negative shifts in FvLib.c.

`FvLib.c:197:128: error: shifting a negative signed value is undefined [-Werror,-Wshift-negative-value]
  *NextFile = (EFI_FFS_FILE_HEADER *) ((((UINTN) CurrentFile - (UINTN) mFvHeader + GetFfsFileLength(CurrentFile) + 0x07) & (-1 << 3)) + (UINT8 *) mFvHeader);
                                                                                                                            ~~ ^
FvLib.c:482:171: error: shifting a negative signed value is undefined [-Werror,-Wshift-negative-value]
    CurrentSection.CommonHeader = (EFI_COMMON_SECTION_HEADER *) ((((UINTN) CurrentSection.CommonHeader) + GetSectionFileLength(CurrentSection.CommonHeader) + 0x03) & (-1 << 2));`

[`-Wno-shift-negative-value` should be added to the Makefile](http://www.insanelymac.com/forum/topic/306156-clover-bugissue-report-and-patch/?p=2234676). The negative shifts could also [be replaced](http://www.insanelymac.com/forum/topic/306156-clover-bugissue-report-and-patch/?p=2234682).

ajfish, 2016-05-15:
From a pedantic point of view shifting negative numbers is undefined behavior in C so the correct fix is casting to an unsigned value. Thus -1 should be -1U (or -1ULL), this is a better fix that suppressing the warning.

lersek, 2016-05-17:
Duplicate of https://github.com/tianocore/edk2/issues/49, closing.

closed 2016-05-17 by lersek