tianocore/edk2 #91: unable to install windows10 with ovmf when smp is used

webczat, 2016-05-21:
Hello.

It seems that it is not possible to install windows10 on qemu virtual machine when the machine has more than one virtual core and uses ovmf as a firmware. Of course kv is enabled.
I know there was a similar issue reported here where ovmf itself failed to boot with error like "entry failed: 80000021). I currently have no logs and the number of the issue where that one was reported but the error is identical, of course except that the register dump below the error is different.
The difference between the previously reported issue and this one is that it does not happen during firmware boot. I am able to successfully boot the windows installation media, and I am able to install the system. The problem is that during the first startup/oob experience, it goes normally, then suddenly the machine halts, dumping the above error. I would guess that if you tried to boot an already installed and configured system, the same thing would happen.
I have verified that the system can be installed and booted when there is only one virtual core. It also works when using seabios instead of ovmf, that's the reasoning behind reporting it here.

My hardware and system information:
4 gigabytes of RAM memory
processor model name	: Intel(R) Celeron(R) CPU  N2820  @ 2.13GHz
linux kernel version: 4.5.4 (archlinux, so almost vanilla)
ovmf version: latest git, or almost latest, from the same week the issue was reported in
This is the machine bootup script:

#!/bin/sh
exec /usr/bin/qemu-system-x86_64 \
-nodefaults \
-machine q35,accel=kvm,smm=on -smp 2,maxcpus=4,sockets=1,cores=2,threads=1 -m 2048 \
-device qxl-vga -device ich9-intel-hda -device hda-micro \
-netdev vde,id=net -device virtio-net-pci,netdev=net,disable-modern=false \
-device virtio-rng-pci,disable-modern=false -device virtio-balloon-pci,disable-modern=false -chardev stdio,id=mux,mux=on -mon chardev=mux,mode=readline -device isa-serial,chardev=mux \
-display sdl -name winvm -device pvpanic -device ib700 \
-device nec-usb-xhci -drive id=cd,if=none,media=cdrom -device ide-cd,drive=cd,bus=ide.0 \
-drive id=cd2,if=none,media=cdrom -device ide-cd,bus=ide.1,drive=cd2 \
-drive id=disk,file=disk.img,if=none,media=disk -device virtio-blk-pci,drive=disk,disable-modern=false \
-cpu host \
-global driver=cfi.pflash01,property=secure,value=on \
-drive if=pflash,format=raw,file=firmware.bin -drive if=pflash,file=vars.bin,format=raw \
$@

webczat, 2016-05-21:
I was previously referring to issue #80

lersek, 2016-05-23:
I understand it doesn't reproduce with SeaBIOS. I still think this is more related to KVM than to OVMF (given that the firmware itself boots, and the Windows 10 installation succeeds too).

Am I right to think that this is yet another regression that was brought about by a host kernel update? (This is the first report I'm seeing against host kernel 4.5.*.) If so, can you please bisect the host kernel? Maybe even capture a (KVM trace)[http://www.linux-kvm.org/page/Tracing] please? Equipped with those, if this is a host kernel regression, then it should be reported to the kernel bugzilla. Thanks.

lersek, 2016-05-23:
Alternatively, if you are convinced this is due to an OVMF package upgrade, can you please bisect OVMF? Thanks.

lersek, 2016-05-23:
(Temporarily labeling this issue "OVMF" just because no issue should be unlabeled, and not because I'm confirming the problem is with OVMF.)

webczat, 2016-05-23:
I am sure this happens also with kernel 4.4.
Because it only happens with ovmf, and only with smp vm, ... then even
if it is not an ovmf bug, ovmf or windows uefi operation causes it. I
believe this bug is related to launching the AP processor, or something
like that.... I mean it *may* be possible you have more knowledge than
me for purposes of tracing this one.
As for bisecting, it may be quite hard for me as on this laptop building
a kernel may take ... really long, unless I can find binaries? Do you
have *any* suggestion as to what kernel version may be the good one, to
have something to test at first?

lersek, 2016-05-23:
My impression has been that the latest v4.4 stable release(s) of the host kernel made things work. Please see issue #21 and the associated [kernel BZ](https://bugzilla.kernel.org/show_bug.cgi?id=107561).

There has been one exception to that: issue #80 (which we closed in this tracker after relocating the problem to another [kernel BZ](https://bugzilla.kernel.org/show_bug.cgi?id=116731)).

Given your report, I immediately thought that it was a duplicate of either issue #21 or issue #80 (i.e., kernel BZs 107561 or 116731). However, since you mentioned issue #80 yourself, I thought you had read through it (including kernel BZ 116731 as well), and determined that your symptoms were due to a different root cause. Is that the case? Did you test the host kernel patches recommended by @rkrcmar in kernel BZ 116731?

> even if it is not an ovmf bug, ovmf or windows uefi operation causes it

I'm asking you to isolate the problematic component. Problems like this have been reported at least twice thus far, with both cases ending up in the kernel bugzilla. Personally I can't even reproduce such problems any longer (after the KVM fix for issue #21), let alone debug them.

Here's a list of steps:
* please see if your problem goes away when running the latest v4.4.* stable release (4.4.11 apparently) on the host
* if it doesn't, please see if the discussion in issue #80 applies to your hardware (i.e., no EPT). If it applies, please work with @rkrcmar in kernel BZ 116731.
* If none of the above apply, I don't know how I could help, without having access to your hardware. (Either way, I'd prefer if this problem was tracked in the kernel bugzilla, given the history above, perhaps as a separate KVM report.)

webczat, 2016-05-23:
Symptoms of this issue are the same (probably) as issue #80. but, I DO
have EPT in my cpu, and unrestricted guest mode, and almost everything,
so it is not the same issue.

webczat, 2016-05-23:
It is really not issue #21, the whole error is really different from mine.

lersek, 2016-05-23:
Thanks for checking.

Actually, now I can see that you are running QEMU with SMM enabled, and with SMM-only access to the pflash chip being enforced. (Also, you are apparently running a self-built OVMF binary -- can you please state your OVMF build options?)

SMM makes a huge difference, and I wish you had pointed it out explicitly in the report. :(

If I had realized your use of SMM, I wouldn't have mentioned those other reports at all. I have occasionally seen such issues with SMM enabled, and I even reported it once (as yet internally) to @bonzini, after some initial (superficial) investigation myself. I believe Paolo has it on his TODO list to check it out.

For me the circumstances that (rarely) trigger similar problems are *not* a single SMP guest. Instead, for me it shows up occasionally when I run an SMM guest in parallel with another guest (SMM or not doesn't matter in the second guest).

I recommend to report this issue to the kernel bugzilla (Product=Virtualization, Component=kvm), with as much information about your hardware, host kernel, guest configuration, and OVMF binary as possible. Please include an `ignore_loglevel` dmesg from the host kernel (from after the guest crash), the QEMU error report, and a `DEBUG_VERBOSE` enabled OVMF log too.

I obviously cannot claim for sure that it's a KVM (and not an OVMF) bug, but at this point KVM seems more likely, and the KVM maintainers are definitely better equipped to debug it at this stage. Thanks.

webczat, 2016-05-23:
Yes, I had smm enabled. I built ovmf with ipv6, http boot and secureboot
and smm.

webczat, 2016-05-23:
Also, my issue is always reproducible. it is not showing up sometimes,
but always at the same moment. I may try building smm and nonsmm ovmf,
then testing with both, both using and not using smp. I will probably
first fully install the os on a working configuration, that is
uniprocessor vm with smm enabled. and try if a fully installed os shows
the same issue when booting

bonzini, 2016-05-23:
Can you send me your OVMF binary (or the commit id + command line you built it from)? Also it can be useful to know whether you can reproduce it with e.g. Windows 7. Thanks!

lersek, 2016-05-23:
@webczat

> Also, my issue is always reproducible. it is not showing up sometimes, but always at the same moment

That sounds (comparatively) great, actually -- the issue I've occasionally encountered is very hard to tie to any guest actions. A deterministic reproducer is always welcome. If you can provide further environment details (including reproducibility with a Windows 7 guest, as Paolo suggests), it might help further.

@bonzini: thank you for chiming in here!

webczat, 2016-05-23:
I will now try to build few versions of ovmf, with and without smm. One
more detail: I was building from OvmfPkgIa32X64.dsc, not pure X64.
I do not have any windows7 thing at hand at the moment. I will now test
windows10 with different ovmf configurations to check if smm may be the
cause, or some interaction with smm on the second processor, or
whatever. I will tell you if any of those ovmf configurations will work.
if not, I may then try windows7. I hope it supports uefi mode.

webczat, 2016-05-23:
I am not sure what I am doing wrong, but now I am unable to run smm ovmf
at all. some bug introduced in a commit? I pulled the repository today
and it seems to me that it does not work, the version without smm works.
Yes, I ensured that I enabled smm in -machine, and secured my pflash.
Either I forgot about something or there is a new bug.

webczat, 2016-05-23:
the above affects build using OvmfPkgX64.dsc...

lersek, 2016-05-23:
@webczat 

The `OvmfPkgX64.dsc` build of OVMF does not run if both of the following conditions apply:
* S3 support is enabled on the QEMU command line (which is the default), and
* `-D SMM_REQUIRE` is passed to the build.

The reason is that the S3 Resume PEIM does not support SMM-based S3 resume when the PEI phase is 64-bit.

Please refer to the `S3Verification()` function in `OvmfPkg/PlatformPei/Platform.c`. When these circumstances are recognized, the function prints an error message to the QEMU debug port (the same place where all other OVMF messages go):

    S3Verification: S3Resume2Pei doesn't support X64 PEI + SMM yet.
    S3Verification: Please disable S3 on the QEMU command line (see the README),
    S3Verification: or build OVMF with "OvmfPkgIa32X64.dsc".

 In your report you mention "I currently have no logs", and also on your QEMU command line I see no `isa-debugcon` device / no `-debugcon` switch. A bit later I asked for an OVMF log specifically. `-debugcon` is documented in the `OvmfPkg/README` file as well.

I take it if you had not thrown away the OVMF log, you would have seen the above error message, with the clear instructions it gives. Pray tell, are we wasting each other's time?

webczat, 2016-05-23:
well, I was wondering if it is something like that, but did not remember
how to enable debugcon, and where the instructions were. thanks, then I
only have to test one smm and one non smm builds

webczat, 2016-05-24:
I have performed some tests. I tried to install windows10 professional
with two ovmf builds:
one is the x64 one with smm disabled, one is the IA32X64 one with smm
enabled.
Results: the one with smm enabled crashes between first and second
restart after installation, but the non smm ovmf does not crash.
Also, this issue seems not to affect all cases, as if I would have an
already installed and configured win10, it seems it would boot in both
cases.

And here is the actual error:
KVM: entry failed, hardware error 0x80000021

If you're running a guest on an Intel machine without unrestricted mode

support, the failure can be most likely due to the guest entering an
invalid
state for Intel VT. For example, the guest maybe running in big real
mode
 pionowy pasek przewijania 705 procent.
which is not supported on less recent Intel processors.

EAX=00000001 EBX=00491e8c ECX=966f0000 EDX=000c00e1

ESI=9448c6a0 EDI=c159a180 EBP=9448c410 ESP=9448c310

EIP=00008000 EFL=00000002 [-------] CPL=0 II=0 A20=1 SMM=1 HLT=0

ES =0000 00000000 ffffffff 00809300

CS =c000 7ffc0000 ffffffff 00809300

SS =0000 00000000 ffffffff 00809300

DS =0000 00000000 ffffffff 00809300

FS =0000 00000000 ffffffff 00809300

GS =0000 00000000 ffffffff 00809300

LDT=0000 00000000 000fffff 00000000

TR =0040 c3f2b070 00000067 00008b00

GDT=     c3f2a000 0000006f

IDT=     00000000 00000000

CR0=00050032 CR2=41b66000 CR3=5d948000 CR4=00000000

DR0=0000000000000000 DR1=0000000000000000 DR2=0000000000000000
DR3=0000000000000
000

DR6=00000000ffff0ff0 DR7=0000000000000400

EFER=0000000000000000
Code=ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff <ff> ff
ff ff f
f ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

bonzini, 2016-05-24:
I cannot reproduce this with Linux 4.7-rc1 and the following command line:

    qemu-system-x86_64 -nodefaults --enable-kvm -machine q35,smm=on -smp 2 -m 2048 \
         -debugcon mon:stdio -vnc :0 -cpu Haswell-noTSX,enforce \
         -global driver=cfi.pflash01,property=secure,value=on \
         -drive if=pflash,readonly=on,file=/usr/share/OVMF/OVMF_CODE.secboot.fd \
         -pflash win10-vars.img -hda win10-uefi.img -cdrom /vm/isos/en_windows_10_enterprise_n_version_1511_updated_apr_2016_x64_dvd_8709271.iso \
         -vga std -boot d -global isa-debugcon.iobase=0x402

Can you try again with Linux 4.6 and a command line as similar to the above as possible?

webczat, 2016-05-24:
Well, I believe archlinux will have 4.6 in a short while and I will then
be able to try.

webczat, 2016-05-24:
ahh one thing to note. what does -smp 2 do? windows does not support
multiple physical processors. try -smp cores=2 unless you are sure that
it has the same effect. I mean the windows system will probably detect,
but not enable, the second cpu.

bonzini, 2016-05-24:
Also, please capture from dmesg anything that starts with `*** Guest State ***`. I am particularly interested in the line starting with `VMEntry:` and the three after that. You can get that with 4.5 too, and please capture it two-three times to see whether it's always the same or it changes.  Finally, please include the output of `grep -r . /sys/module/kvm_intel/parameters/` here in the bug.

webczat, 2016-05-25:
Okay. I performed the necessary tests as you requested by running the vm
two times. Actually I ran it three times, here is the result:
It seems that this bug does not always occur, I ran the machine once
without any error! It may even be possible that this error does not
always happen at the same stage of "windows preparation".
I am now talking about the next two runs. I have run the machine two
more times and it crashed as I wanted. Here is the full dmesg log from
both runs: (it seems the lines you mentioned are the same in both cases
but I include them for completeness)...

[ 6026.560893] kvm: zapping shadow pages for mmio generation wraparound
[ 6044.472516] kvm [3412]: vcpu0, guest rIP: 0xfffff8019c8d6103
unhandled rdmsr: 0x611
[ 6046.201977] kvm [3412]: vcpu0, guest rIP: 0xfffff8019d7c9188
unhandled rdmsr: 0x606
[ 7058.291757] *** Guest State ***
[ 7058.291767] CR0: actual=0x0000000000050032,
shadow=0x0000000000050032, gh_mask=fffffffffffffff7
[ 7058.291770] CR4: actual=0x0000000000002050,
shadow=0x0000000000000000, gh_mask=fffffffffffff871
[ 7058.291771] CR3 = 0x00000000feffc000
[ 7058.291773] RSP = 0xffffd001be085420  RIP = 0x0000000000008000
[ 7058.291776] RFLAGS=0x00000002         DR7 = 0x0000000000000400
[ 7058.291778] Sysenter RSP=0000000000000000 CS:RIP=0000:0000000000000000
[ 7058.291781] CS:   sel=0xc000, attr=0x08093, limit=0xffffffff,
base=0x000000007ffc0000
[ 7058.291784] DS:   sel=0x0000, attr=0x08093, limit=0xffffffff,
base=0x0000000000000000
[ 7058.291786] SS:   sel=0x0000, attr=0x08093, limit=0xffffffff,
base=0x0000000000000000
[ 7058.291789] ES:   sel=0x0000, attr=0x08093, limit=0xffffffff,
base=0x0000000000000000
[ 7058.291791] FS:   sel=0x0000, attr=0x08093, limit=0xffffffff,
base=0x0000000000000000
[ 7058.291794] GS:   sel=0x0000, attr=0x08093, limit=0xffffffff,
base=0x0000000000000000
[ 7058.291796] GDTR:                           limit=0x0000006f,
base=0xfffff8013332a000
[ 7058.291799] LDTR: sel=0x0000, attr=0x10000, limit=0x000fffff,
base=0x0000000000000000
[ 7058.291801] IDTR:                           limit=0x00000000,
base=0x0000000000000000
[ 7058.291804] TR:   sel=0x0040, attr=0x0008b, limit=0x00000067,
base=0xfffff8013332b070
[ 7058.291806] EFER =     0x0000000000000000  PAT = 0x0007010600070106
[ 7058.291808] DebugCtl = 0x0000000000000000  DebugExceptions =
0x0000000000000000
[ 7058.291810] Interruptibility = 00000008  ActivityState = 00000000
[ 7058.291812] *** Host State ***
[ 7058.291814] RIP = 0xffffffffa0b25005  RSP = 0xffff880169be7cd8
[ 7058.291816] CS=0010 SS=0018 DS=0000 ES=0000 FS=0000 GS=0000 TR=0040
[ 7058.291819] FSBase=00007f658f535700 GSBase=ffff88017fc00000
TRBase=ffff88017fc12c40
[ 7058.291821] GDTBase=ffff88017fc09000 IDTBase=ffffffffff57c000
[ 7058.291823] CR0=0000000080050033 CR3=000000005ee7d000
CR4=00000000001026f0
[ 7058.291825] Sysenter RSP=0000000000000000 CS:RIP=0010:ffffffff815b7bd0
[ 7058.291828] EFER = 0x0000000000000d01  PAT = 0x0407010600070106
[ 7058.291829] *** Control State ***
[ 7058.291831] PinBased=0000003f CPUBased=b6a1edfa SecondaryExec=000000eb
[ 7058.291833] EntryControls=0000d1ff ExitControls=002fefff
[ 7058.291835] ExceptionBitmap=00060042 PFECmask=00000000 PFECmatch=00000000
[ 7058.291838] VMEntry: intr_info=800000d1 errcode=00000000 ilen=00000000
[ 7058.291840] VMExit: intr_info=800000ef errcode=00000000 ilen=00000002
[ 7058.291842]         reason=80000021 qualification=0000000000000000
[ 7058.291844] IDTVectoring: info=00000000 errcode=00000000
[ 7058.291846] TSC Offset = 0xfffff450f4a9d820
[ 7058.291847] TPR Threshold = 0x02
[ 7058.291849] EPT pointer = 0x000000011962301e
[ 7058.291851] Virtual processor ID = 0x0001
[... 56 lines cut ...]
[ 8937.143510] IDTVectoring: info=00000000 errcode=00000000
[ 8937.143512] TSC Offset = 0xfffff14d0542bc10
[ 8937.143514] TPR Threshold = 0x02
[ 8937.143516] EPT pointer = 0x000000010888701e
[ 8937.143517] Virtual processor ID = 0x0001

As requested, here are the kvm_intel module parameters:

parameters/ept:Y
parameters/pml:N
parameters/vpid:Y
parameters/enable_shadow_vmcs:N
parameters/eptad:N
parameters/flexpriority:Y
parameters/ple_window_grow:2
parameters/ple_window_shrink:0
parameters/ple_window_max:1073741823
parameters/fasteoi:Y
parameters/enable_apicv:N
parameters/unrestricted_guest:Y
parameters/emulate_invalid_guest_state:Y
parameters/nested:N
parameters/vmm_exclusive:Y
parameters/ple_window:4096
parameters/ple_gap:0

bonzini, 2016-05-25:
Good - I have a suspicion that the cause is the "VMEntry: intr_info=800000d1" line.

bonzini, 2016-05-26:
Indeed: "The IF flag (RFLAGS[bit 9]) must be 1 if the valid bit (bit 31) in the VM-entry interruption-information field is 1 and the interruption type (bits 10:8) is external interrupt." (Intel SDM, 26.3.1.4)

lersek, 2016-05-26:
@bonzini Trying to translate it for myself, also from the data in @webczat's comment...

We have:

    [ 7058.291757] *** Guest State ***
    ...
    [ 7058.291776] RFLAGS=0x00000002 DR7 = 0x0000000000000400

and

    [ 7058.291829] *** Control State ***
    ...
    [ 7058.291838] VMEntry: intr_info=800000d1 errcode=00000000 ilen=00000000

Bit 9 of `RFLAGS` is clear, which means that the guest has cleared the `IF` flag (= masking interrupts) before `VMEXIT`.

Furthermore, in the `intr_info` field, bit 31 is set ("Valid"), bits 10:8 are zero (that is, interruption type is "External interrupt"), and the least significant byte is 0xd1 (which is the vector for Windows' clock interrupt, as I've learned from the web).

This seems to mean that we're about to inject an external (clock) interrupt to the guest, with the next `VMRESUME` instruction. However, according to "CHAPTER 23
VM ENTRIES", the processor ensures (for the sake of the guest) that KVM cannot create an invalid (virtual) hardware state; in other words, the PCPU performs various state checks at vm-entry. One of those is that we can't inject an external interrupt (which would cause the guest to start executing an interrupt handler etc) when the guest has actually masked maskable interrupts. When we try, the `VMRESUME` instruction fails, and (given that this is a guest-state check), the error is returned to KVM as with a `VMEXIT` ("23.7
VM-ENTRY FAILURES DURING OR AFTER LOADING GUEST STATE").

We have

    VMExit: intr_info=800000ef errcode=00000000 ilen=00000002
    reason=80000021 qualification=0000000000000000

where the 0x21 value in the low uint16_t of the reason means "VM-entry failure due to invalid guest state."

So long my self-explanation is acceptable, I guess I understand this. Yay! But how do you fix this? Look at `RFLAGS.IF`, and if it is clear, simply throw away the external interrupt (leave the `Valid` bit in the interruption-information field clear, before `VMRESUME`)?

bonzini, 2016-05-27:
The interrupt shouldn't have been injected even. It's a bug in KVM.

bonzini, 2016-06-01:
Patch at http://thread.gmane.org/gmane.linux.kernel/2232266, this can be closed.

lersek, 2016-06-01:
I would like to subscribe to your newsletter! Bugfixing week FTW! :) Thans Paolo!

closed 2016-06-01 by lersek

lersek, 2016-07-21:
This (closed) item has been manually migrated to
https://tianocore.acgmultimedia.com/show_bug.cgi?id=85