tianocore/edk2 #93: UEFI Shell only sees EFI\BOOT\BOOTX64.EFI

4risu, 2016-05-30:
I tried several ISO images and an USB Stick with an EFI directory. UEFI Shell only sees BOOTX64.EFI in EFI\BOOT. There are other files in these directories.

[image]

RafaelRMachado, 2016-05-30:
Hi

Are you sure that FS0 is your correct device ?

Use the command map to see all detected file systems. Maybe FS0 points to a
FAT32 partition on a local HDD, and youe memory stick may be FS1 or any
other alias.

Thanks and Regards
Rafael

4risu, 2016-05-30:
Yes I'm sure. In this case it was the only filesystem available. I tested multiple filesystems and all just show \EFI\BOOT\BOOTX64.EFI. It's really strange.

lersek, 2016-05-30:
On CD-ROM images, the driver stack built into OVMF will not see the full iso9660 / Joliet / Rock Ridge filesystem that you see under Linux (for example) when you mount the disk (or disk image, with the loop device). Instead, the edk2 driver stack built into OVMF will only find the UEFI El Torito boot image on the CD-ROM (or CD-ROM image). What you store under `\EFI` in the iso9660 filesystem is irrelevant for UEFI firmware.

Instead, here's how you can expose a filesystem to UEFI firmware on a CD-ROM:

* First, create a FAT32 filesystem in a local file. The command commonly used for this is `mkdosfs -C "$FAT32_IMG" -n "$FAT32_LABEL" -- "$FAT32_IMG_SIZE_KB"`.
* Second, populate this filesystem with content. One possibility is the `mtools` utilities. For example, `mcopy -i "$FAT32_IMG" readme.txt ::` will copy the file `readme.txt` into the root directory of the FAT32 image.
* Third, once you are done with the FAT32 image file, you have to add it to the ISO image as a normal file, but also point some special options to it. For example, `genisoimage -efi-boot "$FAT32_IMG" -no-emul-boot -o "$ISO_IMG" -- "$FAT32_IMG"`.

This is a frequent misunderstanding with optical media images. Again, what you see in the EFI directory of the iso9660 / Joliet / Rock Ridge filesystem when mounting the disk (image) under Linux, and what you see in the EFI directory of the FAT32 filesystem image that is embedded into the CD-ROM image, are completely independent.

I'm closing this bug for now. If the above analysis is incorrect in your opinion, please comment on it, and we can reopen the item if it's justified. Thanks!

closed 2016-05-30 by lersek