tianocore/edk2 #94: random bsod of windows10

webczat, 2016-05-31:
Hello.

When I install windows10 on qemu with enabled kvm and the ovmf firmware on my dedicated server, the installed guest system experiences many quite random blue screens.
Depending on the current device configuration of the machine, the blue screen can be provoked by installing the network driver (only if the link is connected) or by connecting the network link after driver install, sometimes those blue screens happen randomly during windows usage, sometimes at regular intervals, sometimes it is enough to wait minute after oob to experience them, and it always happens nearly at the same time until machine reconfigured, at which point it may be that bsods will happen in a different way.
There is no good configuration. I tried using virtio-net driver and changing the virtio net card to e1000, tried to use virtio blk and virtio scsi for hard drives, add or remove pcie ports (yes, id made a difference for some reason), tried to change number of cpus or cpu type, without any success. The only thing I didn't try is changing machine type from q35 to something else.
So, windows10 guest just does not work in my environment when using ovmf. It seems that all those things that I have experienced do not happen at all when using seabios, and I have tested with few configurations.
This is every piece of information I can give about my setup and host/guest hardware/software:

host processor: Intel(R) Core(TM) i5 CPU         750  @ 2.67GHz
memory: 16 gigabytes of ram
linux kernel version: 4.5.4
qemu version: 2.5.2
edk2 commit: 3ab7066e8d8ae43d9cdee76600b90918f8bee5d9
ovmf build command: build -DSECURE_BOOT_ENABLE -DNETWORK_IP6_ENABLE -DHTTP_BOOT_ENABLE
guest windows version: 10 professional n, build 1511

This is one of the worst configurations of the vm that actually causes the bsod almost immediately after showing oob:

#!/bin/sh
umask 0077
cd ~/vms/vm
/bin/qemu-system-x86_64 \
-name vm -D vm.log \
-spice tls-port=25000,x509-cacert-file=../ca/cacert.pem,x509-cert-file=cert.pem,x509-key-file=key.pem,x509-dh-key-file=dh.pem,password=mypass \
-machine q35,accel=kvm,iommu=on -nodefaults -cpu host \
-chardev stdio,id=interface,mux=on,signal=off \
-chardev socket,id=control,server,nowait,path=control.sock \
-mon chardev=interface,mode=readline -mon chardev=control,mode=control \
-smp cores=4 -m size=2048M \
-device ib700 -device pvpanic -device isa-serial,chardev=interface \
-device ioh3420,port=1,chassis=1,id=pcieport1 \
-device ioh3420,id=pcieport2,port=2,chassis=2 \
-device virtio-balloon-pci,disable-modern=false \
-device virtio-rng-pci,disable-modern=false \
-device virtio-serial-pci,disable-modern=false \
-chardev socket,id=agent,server,nowait,path=agent.sock \
-device virtserialport,chardev=agent,name=org.qemu.guest_agent.0 \
-chardev spicevmc,name=vdagent,id=vdagent \
-device virtserialport,chardev=vdagent,name=com.redhat.spice.0 \
-device qxl-vga \
-device ich9-intel-hda -device hda-micro \
-netdev bridge,br=vmnet,id=net \
-device virtio-net-pci,disable-modern=false,netdev=net \
-device virtio-scsi-pci,disable-modern=false \
-device nec-usb-xhci -device nec-usb-xhci \
-chardev spicevmc,name=usbredir,id=usbredir1 \
-device usb-redir,chardev=usbredir1 \
-chardev spicevmc,name=usbredir,id=usbredir2 \
-device usb-redir,chardev=usbredir2 \
-device usb-tablet \
-drive id=cd1,media=cdrom,if=none \
-device ide-cd,drive=cd1,bus=ide.0 \
-drive id=cd2,if=none,media=cdrom \
-device ide-cd,drive=cd2,bus=ide.1 \
-drive id=disk,if=none,media=disk,format=qcow2,file=disk.img \
-device scsi-hd,drive=disk \
-drive if=pflash,media=disk,format=raw,file=firmware.bin \
-drive if=pflash,media=disk,format=raw,file=vars.bin \
-watchdog-action reset -rtc base=localtime \
-boot menu=on \
$@

lersek, 2016-06-01:
I also have a windows 10 guest (installed from `en_windows_10_enterprise_2015_ltsb_n_x64_dvd_6848316.iso`), and I have not experienced such problems.

Your build flags don't include `-D SMM_REQUIRE` which is both good and bad. Good because SMM is always much harder to debug, and bad because things should really just work.

Since it reproduces easily on your side, and I've never encountered it on my side, can you please (a) capture a screenshot of the BSOD, (b) perhaps even try a debug/checked build (for those the BSOD tends to give more info), (c) capture a [KVM trace](http://www.linux-kvm.org/page/Tracing)?

Also, what is "oob"? Thanks.

lersek, 2016-06-01:
BTW, have you tried the `ignore_msrs=1` parameter with the `kvm` module?

lersek, 2016-06-01:
(You can verify the current setting with `cat /sys/module/kvm/parameters/ignore_msrs`.)

webczat, 2016-06-01:
oob = out of box experience, this last config screen just before login.
Those bsods are... somehow random although it was easy to reproduce them
for me.
Not sure why they happen, really.

No, I am not running with ignore_msrs set. I am using a polish system
and not sure if you will be able to read bsod in this case although the
real body is still english hmmm

lersek, 2016-06-01:
Please try with `ignore_msrs` set; Windows 10 has been known (especially development / beta versions) to massage a huge variety of MSRs, some (most?) of which are not implemented by KVM.

IIRC, `ignore_msrs` will cause KVM to ignore writes to MSRs unknown to it, and return 0 when MSRs unkown to it are read, as opposed to injecting a fault at once. Of course, a blanket 0 return value is not guaranteed to be safe either, but it works surprisingly often.

 Please see the following references:
* http://answers.microsoft.com/en-us/insider/forum/insider_wintp-insider_perf/bsod-kmodeexceptionnothandled-on-10041-in-kvm-with/7330c7af-abfa-4755-b851-b7720e34d80c
* https://www.redhat.com/archives/vfio-users/2016-May/msg00099.html

webczat, 2016-06-01:
That msr usage is operating system, not firmware specific right?
Remember that I said that everything works very well on seabios, and so
windows probably does not have any reason to read some msrs only when
run in uefi mode. In seabios I did not experience any bluescreens, only
in when run with ovmf. I even tried rebuilding ovmf in case it was
somehow damaged, no luck.

lersek, 2016-06-02:
> That msr usage is operating system, not firmware specific right?

That's not guaranteed at all, to my knowledge. We can't say for sure that Windows 10 will not base various hardware decisions on the firmware's characteristics. For example, in the [SVVP test suite](https://blogs.technet.microsoft.com/keithcombs/2008/06/15/what-is-the-server-virtualization-validation-program-svvp/), a bunch of tests behave differently, and require different things from OS-level drivers, when they are executed in a VM with SeaBIOS vs. in a VM with OVMF. Here's an [example](https://bugzilla.redhat.com/show_bug.cgi?id=1314600).

> Remember that I said that everything works very well on seabios,

Yes, I remember. People are very quick to point out "it works in SeaBIOS", but they never list their reasons why they don't use SeaBIOS for their workload then.

You appear to run your guest on a dedicated server, and not do VFIO device assignment. So what is *your* use case for insisting on OVMF? (There are other valid uses for OVMF, I'm just curious if you actually have one.)

> and so windows probably does not have any reason to read some msrs only when run in uefi mode.

That's exactly what cannot be guaranteed. The item on answers.microsoft.com that I referenced above is about OVMF only too.

Here's another example. Windows 8 and Windows 10 installer ISOs exist for the following computer types:
* BIOS: Client and Server
* 64-bit UEFI: Client and Server
* 32-bit UEFI: Client only

Meaning, if you grab a 32-bit Windows 8 or 10 Client installer ISO, it is expected to boot the installer in both SeaBIOS and 32-bit OVMF guests. Except the latter case doesn't work *at all*, and months of on-and-off correspondence with a Microsoft developer (who has elected to discuss the isssue with me) has not produced results. Microsoft doesn't openly publish *everything* that they require from a UEFI platform, and SeaBIOS vs. OVMF may very well influence the behavior of the runtime OS.

Look, we can go back and forth on this. I've never encountered the issue you describe (I booted my Win10 VM yesterday to check again -- no issues). Also, I don't have the capacity to do exploratory testing, to see what "might" break, in the myriads of QEMU command line and Windows combinations. If you have one specific configuration that you would like to work, and it doesn't, you'll have to put in the effort to support our debugging and development.

In this item I have thus far recommended `ignore_msrs`, asked for a screenshot, and asked for a KVM trace. Instead of those, I'm seeing arguments why I'm supposedly wrong. Do you think that's productive? Or should I just ignore issue reports in this upstream tracker, like most other TianoCore maintainers do?

lersek, 2016-06-02:
Also, for any OVMF-on-KVM issue report, the OVMF debug log and the host dmesg should be included as well.

(BTW, please don't paste all these data into the issue tracker -- please host them somewhere external, and only reference them here. The GitHub issue tracker sucks extremely because it doesn't allow for attachments, unfortunately. Once we [move to Bugzilla](http://thread.gmane.org/gmane.comp.bios.edk2.devel/12658), attachments should become possible.)

webczat, 2016-06-02:
My server is not really a production thing, partially used for hosting
few things, partially being a... playground. I am using my server
because my laptop is too weak for really holding a vm especially if I
want to use other tools at the same time, so I want such a vm for things
I cannot use linux for like windows development/testing.
Why ovmf? well, I do not have any really good reason.
I was just trying to understand why things like ignore_msrs can matter
in case of different firmwares, nothing else. As for things like
screenshots and kvm trace, I currently had not enough time to try to
install windows again, but I should be able to do it shortly. The only
thing I am wondering about is if I will be able to get a screenshot in
time but we'll see.

webczat, 2016-06-02:
Ah, about logs: dmesg does not contain anything specific. if I am wrong
I will correct myself when getting the screenshot. About debug logs of
ovmf, I always don't know how to generate it, only that I use isa debugcon

lersek, 2016-06-02:
Not sure if it helps, but a screenshot can be taken from the QEMU monitor as well. The HMP command is called `screendump`.

Second, your QEMU command line seems to have a bit of cruft. What do you need:
* `-machine iommu=on` (emulated Intel IOMMU (VT-d) support).
* `-device ioh3420` (additional, unused PCIe root ports),
* two instances of `-device nec-usb-xhci` (two USB 3.0 host controllers),
for?

Can you reproduce the issue with a minimal command line?

webczat, 2016-06-02:
the problem is that I could. Or rather I could reproduce a different
form of this issue without root ports. As I said it is partially a
playground so I do not have the specific reason to add some things, for
example iommu is added just because I wanted to say if windows can use
it for anything, pcie root ports and xhci for increased hotplug
capability/windows hotplug testing maybe. I can try removing some things
further, like iommu etc, sure, but first I will test the configuration I
have pasted here for reference.
About screendump, the problem is the timing. I cannot disable automatic
restart of blue screen if the blue screen happens on the last setup
screens. Unless checked builds are known to have it defaulted, then it
will help me a lot.

lersek, 2016-06-02:
> Or rather I could reproduce a different form of this issue without root ports

This kind of experimentation can actually be valuable, but only if you isolate the test cases. What command line produces exactly what kind of behavior, what cmdline differences cause what differences in behavior, and so on.

> About screendump, the problem is the timing. I cannot disable automatic restart of blue screen if the blue screen happens on the last setup screens.

Can you perhaps write a shell script that connects to the monitor and dumps the screen into `dump_${COUNTER++}.ppm` every five seconds? If you convert the PPM files to JPG and upload them somewhere in a tarball, I could flip through them quickly I guess.

webczat, 2016-06-02:
well: if you remove, from the configuration I gave, all root ports, and
replace virtio scsi with virtio blk, you will get blue screens on
network driver installs. usually, that is the problem, sometimes it is
unpredictable.
As for screenshots, I will find a way to do that, for example using
gnome's screenshotting.
Could you please give me instructions or link on how to enable ovmf
debugging?

lersek, 2016-06-02:
> Could you please give me instructions or link on how to enable ovmf debugging?

Ah, certainly. Sorry for not providing those bits earlier.

There are two parts. The first part is setting the edk2 debug level (it is actually a debug mask). The second part is capturing the debug output.

The first part is optional, but I really like to get verbose messages, for which the `DEBUG_VERBOSE` bit has to be set in the debug mask. It can only be done at build time, so please apply the following patch, and then rebuild OVMF:

    diff --git a/OvmfPkg/OvmfPkgIa32.dsc b/OvmfPkg/OvmfPkgIa32.dsc
    index aa1a6193d19c..71567b8f0439 100644
    --- a/OvmfPkg/OvmfPkgIa32.dsc
    +++ b/OvmfPkg/OvmfPkgIa32.dsc
    @@ -397,7 +397,7 @@ [PcdsFixedAtBuild]
       # DEBUG_VERBOSE   0x00400000  // Detailed debug messages that may
       #                             // significantly impact boot performance
       # DEBUG_ERROR     0x80000000  // Error
    -  gEfiMdePkgTokenSpaceGuid.PcdDebugPrintErrorLevel|0x8000004F
    +  gEfiMdePkgTokenSpaceGuid.PcdDebugPrintErrorLevel|0x8040004F

     !ifdef $(SOURCE_DEBUG_ENABLE)
       gEfiMdePkgTokenSpaceGuid.PcdDebugPropertyMask|0x17
    diff --git a/OvmfPkg/OvmfPkgIa32X64.dsc b/OvmfPkg/OvmfPkgIa32X64.dsc
    index f4b55c56e012..13693af9d29b 100644
    --- a/OvmfPkg/OvmfPkgIa32X64.dsc
    +++ b/OvmfPkg/OvmfPkgIa32X64.dsc
    @@ -402,7 +402,7 @@ [PcdsFixedAtBuild]
       # DEBUG_VERBOSE   0x00400000  // Detailed debug messages that may
       #                             // significantly impact boot performance
       # DEBUG_ERROR     0x80000000  // Error
    -  gEfiMdePkgTokenSpaceGuid.PcdDebugPrintErrorLevel|0x8000004F
    +  gEfiMdePkgTokenSpaceGuid.PcdDebugPrintErrorLevel|0x8040004F

     !ifdef $(SOURCE_DEBUG_ENABLE)
       gEfiMdePkgTokenSpaceGuid.PcdDebugPropertyMask|0x17
    diff --git a/OvmfPkg/OvmfPkgX64.dsc b/OvmfPkg/OvmfPkgX64.dsc
    index 9c1118557012..81d28a1f51eb 100644
    --- a/OvmfPkg/OvmfPkgX64.dsc
    +++ b/OvmfPkg/OvmfPkgX64.dsc
    @@ -402,7 +402,7 @@ [PcdsFixedAtBuild]
       # DEBUG_VERBOSE   0x00400000  // Detailed debug messages that may
       #                             // significantly impact boot performance
       # DEBUG_ERROR     0x80000000  // Error
    -  gEfiMdePkgTokenSpaceGuid.PcdDebugPrintErrorLevel|0x8000004F
    +  gEfiMdePkgTokenSpaceGuid.PcdDebugPrintErrorLevel|0x8040004F

     !ifdef $(SOURCE_DEBUG_ENABLE)
       gEfiMdePkgTokenSpaceGuid.PcdDebugPropertyMask|0x17

For the second part (it's also documented in `OvmfPkg/README`), please just add the following two options to the QEMU command line:

    -debugcon file:ovmf.debug.log -global isa-debugcon.iobase=0x402

Thanks.

lersek, 2016-06-02:
BTW, after logging in to MSDN, and filtering for Windows 10 build 1511 / 64-bit / ISO, I don't see a Professional edition. Can you please give me the exact file name of your installer ISO image?

webczat, 2016-06-02:
hey, it is actually the multiple editions cd. my filename is:
en_windows_10_multiple_editions_debug-checked_version_1511_x64_dvd_7226317.iso
Explanation: I am downloading it. the previous one was the pl one
without debug things, but I just deleted it and will test on the above.

lersek, 2016-06-02:
I can't find this file in MSDN :(

lersek, 2016-06-02:
For the record, I didn't install guest agents (QEMU guest agent or Spice guest agent).

webczat, 2016-06-02:
Those cannot be drivers because except disk drivers I did not have any
other ones. and because changing network card to something supported by
windows itself also broke for me when booting. That means that as you
said I have to provide more data. I hope I will be able to do this, I
can do everything that does not involve rebooting (or often rebooting)
the server, but that should not be a problem unless the kernel must be
compiled with special options to use tracing?

lersek, 2016-06-02:
Right, rebooting the host should not be necessary, I expect.

webczat, 2016-06-02:
I know that bug does not appear on my laptop. with the similar
configuration. so this may not be the kernel, or may depend on hardware
somehow.

lersek, 2016-06-02:
Perhaps on your i5-750 processor (Nehalem microarchitecture, code name Lynnfield), the `-cpu host` option (documented as "KVM processor with all supported host features (only available in KVM mode)") tickles something in Windows 10 that Windows 10 doesn't like.

How about trying another `-cpu` option that exposes a well-defined set of CPU features to the guest, while those features remain a subset of your host's features? I'm thinking of `-cpu Penryn,enforce=true`.

If QEMU rejects it, you can try `-cpu Conroe,enforce=true`.

webczat, 2016-06-02:
It was one of the... known things to change the cpu model. and, I tried
that once. with absolutely no result. And I actually changed to
core2duo. It crashed like I did nothing.
I am now trying to install the system. one unknown bsod during disk
driver loading install, and ... the one on oob somehow does not happen
or rather probably happened once when i was not looking. it may be
somehow complicated, it seems, to reproduce.

webczat, 2016-06-02:
I currently have a big problem. On this build, it always goes to the first configuration screen, then I press use express settings, then it... not sure what, I do not see any blue screen. but it restarts. the problem is that I do not really know if this time my cd was damaged or rather it is the same bug. and kvm traces are extra large.

lersek, 2016-06-02:
Answering in reverse order:

> kvm traces are extra large.

Yes, they always are, but it can be mitigated if you can localize (or interactively trigger) a crash. In that case start `trace-cmd record -b 20000 -e kvm` and await the `Hit Ctrl^C to stop recording` message just before the crash, and interrupt tracing right after. (But, I guess you already know this.)

> if this time my cd was damaged

If you open the details for any given ISO in MSDN, it will tell you the SHA1 of the image, and you can verify it after download.

>On this build, it always goes to the first configuration screen, then I press use express settings, then it... not sure what, I do not see any blue screen. but it restarts.

I repeated the installation on my laptop (= same host as above), with special attention to reboots. There are two reboots in total, after the initial (normal) boot:
* the first is when the first phase of the installation completes, with the green progress bar giving you 10 or 15 seconds to click "restart". If you don't click it, it reboots automatically.
* Then there's a black screen with "getting devices ready" percentages and similar. Finally a "restarting" caption appears briefly (it was really just a flash for me), and indeed it restarts in a few seconds.

Then the "express settings" button appears (on the "get going fast" screen). After clicking it, there is *no* reboot. What follows are: "just a moment" animation, account creation, "hi we're happy you're here" animation, and that's it -- it proceeds to the normal desktop GUI.

So, two restarts in total, and none of those should be occurring after clicking "express settings". If you have an OVMF debug log, you should find three instances of the string `SEC: Normal boot` in it.

If the guest reboots for you after clicking "express settings", hopefully you can start and finish KVM tracing just around that button click.

... I've checked in our internal shared machine pool whether we have an i5-750 -- we don't appear to be. The closest I've found is an i5-650 (same microarchitecture, i.e., Nehalem, but a Clarkdale desktop processor rather than your Lynnfield). I queued a reservation for it. Perhaps I can reproduce a BSOD on it.

webczat, 2016-06-02:
honestly, after clicking this "use express settings" button, it
restarts, and then returns to the screen with this button. something
like a fatal error without blue screen? effect f a damaged cd? what it
was, I am not sure. and thanks for pointing for sha1 checksums, although
I am not sure if they are available on onthehub (I have my win10 from
dreamspark subscription). This thing is hard to localize to make a small
trace, unfortunately! I tried by removing few things that usually caused
only bsod after network driver install, I would even have a chance to
disable autorestarting.

zwei4, 2016-06-03:
I met the same “restart” issue after clicking this "use express settings" button while I verifying Windows 10 installation on my board.
In my case, it is caused by an invalid “century” value of RTC(real time clock).  I am not sure if this is the reason of your failure. But I suggest you to double-check “century” value of RTC.

Thanks,
David

Intel SSG BIOS

webczat, 2016-06-03:
If I only knew what is a century value...

lersek, 2016-06-03:
@zwei4 that's very helpful, thank you for chiming in!

@webczat I think @zwei4 means one of these commits: commit 41628cbc7cb or commit e38ab18a49ed. However, I don't think your OVMF build lacks these commits, because in the issue report you identified commit 3ab7066 as your build basis, and that one contains both of the former two.

webczat, 2016-06-03:
well, but actually that problem with pressing use express settings
appeared now, and I already have a newer commit. so it may be something
new, or specific to this windows build, or whatever.

lersek, 2016-06-03:
@webczat 

> honestly, after clicking this "use express settings" button, it restarts, and then returns to the screen with this button. something like a fatal error without blue screen

Again, can you start KVM tracing when the button first appears, just before clicking the button? And, stop tracing right after the VM reboots and the button appears for the second time?

> I have my win10 from dreamspark subscription

AHA! That's very interesting for a separate reason. As I mentioned above, I couldn't find a debug/checked ISO for Windows 10 in MSDN, only a separate debug symbols MSI (that one has to install inside the already running guest). I didn't understand this, so I googled it. The most interesting result is:

https://www.osronline.com/showthread.cfm?link=269544

It suggests that the debug/checked ISO is available *only* on the DreamSpark website. (Am I correct to think that DreamSpark serves student subscriptions only?) It's strange.

Anyway, if you give me the name of the ISO file you have, I can tell you the SHA1 for it (assuming I downloaded the same file already, or if it is at least available in MSDN too -- which is apparently not guaranteed, according to the above).

I can give you the following checksums right now:

    dbe728416545ea3e47fba05575e81ad0f595871f  en_windows_10_enterprise_2015_ltsb_n_x64_dvd_6848316.iso
    e9e214f128ed325cba8782ce1946727807340c8b  en_windows_10_enterprise_2015_ltsb_n_x86_dvd_6848317.iso
    edc8b56279baef8f9a9c97744426dbce3083cac6  en_windows_10_n_multiple_editions_version_1511_updated_apr_2016_x64_dvd_8710592.iso

webczat, 2016-06-03:
The name of the iso is the one I gave you previously:
en_windows_10_multiple_editions_debug-checked_version_1511_x64_dvd_7226317.iso
I have downloaded it then put on server. not sure if my downloaded copy
was valid, but I am sure it did not get damaged during upload to server.
Yes, I am going to try installation again and do what you suggest,
unless this will not happen again, of course. The thing is why this cd
behaves differently from the multiple editions cd without debug symbols?
Unless, of course, it was damaged.

webczat, 2016-06-03:
My cd seems to be older than the normal one. not sure if I already said
that?

webczat, 2016-06-03:
during last install I had two bsods in the first pass that is when it said it is getting ready probably, so I went impatient. I will not send a trace of the event for now, I will redownload the non debug version of windows and try with it. It should be more predictable and I can possibly get the original bsod's again. I can maybe even install pvpanic driver that will nicely pause the guest for me?

webczat, 2016-06-03:
and now it became weird because I cannot reproduce the issue any longer! I just redownloaded the cd directly on server and it works. but I am not so sure the first one was damaged. it worked on seabios, it worked on ovmf when run on my laptop, and the sha1 checksum was correct between the laptop's copy and the server's copy when I tried to upload it. I didn't check checksum the first time, but I did it the second time when uploading. I do not remember if I ever redownloaded this cd though. And now I installed 2 times, once with virtio blk and no pcie ports, once with virtio scsi and pcie ports, and for now it did not bsod. As you know, even if everything is because of a damaged iso image, I was unable to verify the image due to no checksums from msdn. What do you think?

lersek, 2016-06-03:
@webczat: I propose the following:
* keep the pvpanic device on the command line,
* install a fresh guest from known-good media,
* install the guest driver for pvpanic,
* use the guest as normal.

If you get a panic in a week, we can investigate it. Otherwise, we can close this item as insufficient data end of next week or so.

@stefanha: does upstream QEMU support the flight recorder with systemtap? I couldn't find any more upstream documentation than the comments in `scripts/kvm/kvm_flightrecorder`. Thanks.

(@webczat: If we figure out how to enable the KVM flight recorder, then you could also utilize it for a week. It saves trace events into a ring buffer in host memory, and when you get a crash, you can dump out the trace. The ring buffer size is configurable, so you can influence memory consumption and how far the log will reach back.)

webczat, 2016-06-03:
Oh, I can now tell you when this bug happens and when it does not. I cannot tell you why, however. I can probably confirm that it is a bug of ovmf. The reason is that the bug only happens when using a release build, even of a latest commit. When I build a debug build, even without the verbose flag, it works. Maybe the bug exists there too, but it may be at least harder to trigger. I am sorry for writing so many comments, I am just providing updates. This time, I just had an idea to rollback ovmf to the state it was before, like revert debug flag changes, checkout the commit i gave in the bug, then build a release, not debug build. And, all previous bsods appeared again. Then, I rebuilt ovmf but checking out the latest commit, and bsods were still there! I switched from release to debug build, then it stopped bsoding. It sounds weird, but seems to be true. What should I provide now? As said I cannot provide a ovmf debug log, but probably can do everything else. The above even explains one of the possible reasons why you could not reproduce this issue.

lersek, 2016-06-03:
Very interesting, indeed I never build RELEASE binaries. Let me see if it reproduces on my end.

webczat, 2016-06-03:
Even if it does not, then because the bug is pretty much always
triggered by connected network cable, I could probably install pvpanic
driver now, then trigger it manually. I believe I also had a release
build on my laptop and it worked, but laptop is far slower. Could it be
some kind of a timing issue, or something similar, maybe? If so, it is
not guaranteed that you will be able to reproduce it.

lersek, 2016-06-03:
Okay, since you apparently have a good, reliable reproducer, would you mind please testing the following two builds (no other changes to your environment, and please don't combine the following two points):
* please remove the `-D HTTP_BOOT_ENABLE -D NETWORK_IP6_ENABLE` options from the build command line
* please modify the virtio-net device like this, on the QEMU command line: `-device virtio-net-pci,...,rombar=0`.

lersek, 2016-06-03:
I would like to ask for the following info as well:
* the exact version of your NetKVM driver, and where you got it from
* what do you mean by "connecting the network link after driver install"

thanks

lersek, 2016-06-03:
It does not reproduce for me. I've built a RELEASE binary for OVMF, and when I installed the NetKVM driver in the guest, nothing bad happened.

lersek, 2016-06-03:
I've also done a disable/enable cycle on the virtio-net NIC in Windows, no problems.

webczat, 2016-06-03:
If you had my configuration with pcie ports, then actually the bug would
trigger without virtio net drivers, after second restart there would be
a tcpip.sys bsod. if you would remove pcie ports, then the bug would be
triggered by installing virtio network driver or rather by connecting
the link.
By connecting the link I mean the set_link command that you can use to
connect or disconnect the virtual network cable. I mean the hmp command ofc.
And now, other info you are asking me for: virtio drivers are at version
0.1.117 and I got them from ... redhat? fedora? not sure exactly.
I have done the tests you proposed.
I am not sure if I did it correctly, but setting rombar to 0 in
virtio-net-pci had no visible effect.
However, removing NETWORK_IP6_ENABLE and HTTP_BOOT_ENABLE did.
I did not combine two at once as suggested above.
Also, I did not try to login. the only working configuration from those,
that is release build without http boot and ipv6 support, did not cause
anything, but the one with v6, both when rombar was defaulted and set to
0, caused a bsod on boot, not sure if it even managed to show login screen.

lersek, 2016-06-03:
> If you had my configuration with pcie ports, then actually the bug would trigger without virtio net drivers, after second restart there would be a tcpip.sys bsod

My test script (see a bit higher up) does specify the two `ioh3420` devices, and I haven't seen the BSOD yet. BTW, since you mention `tcpip.sys`, can you upload a screenshot of the BSOD?

> set_link [...] hmp command

okay, I was missing this

Version 0.1.117 of NetKVM very likely came from https://fedoraproject.org/wiki/Windows_Virtio_Drivers , their latest version is 0.1.117 indeed.

So, if I understand your results correctly, with `-D HTTP_BOOT_ENABLE -D NETWORK_IP6_ENABLE`, you see the BSOD regardless of setting `rombar=0`. And, with those build options removed, you don't see the BSOD, again regardless of setting `rombar=0`.

Could you please determine which one of HTTP_BOOT_ENABLE and NETWORK_IP6_ENABLE makes the difference? Thanks.

lersek, 2016-06-03:
BTW, you mentioned QEMU 2.5.2 -- that version doesn't exist.

webczat, 2016-06-03:
maybe it was 2.5-2? not sure yet. that is version 2.5 build 2 (like
second packaging in arch), maybe I just made a mistake. assume 2.5 version.
I have tested what you wanted me to test. the result is: bsod persists
without http boot support, but vanishes with http boot support and
without ipv6. So, seems that the ipv6 support causes it all. If you want
me to send you a screenshot of the bsod, then... here it is:
http://webczatnet.pl/webczat/public/screenshot.png

I am not sure if I screenshotted the whole screen or just the vm/spice
window, but you should see the bsod. It is, however, the polish system,
so if you have any problems, please ask.

stefanha, 2016-06-03:
@lersek Flight recorder tracing is supported upstream.  The QEMU scripts/kvm/kvm_flightrecorder script uses kernel trace events instead of SystemTap.  It will trace all the kvm.ko events listed by "perf list | grep kvm:".

For QEMU tracing you can use either ./configure --enable-trace-backends=ftrace - this works together with the trace events flight recorder approach above.

Or if you like SystemTap you can use --enable-trace-backends=dtrace.  The qemu-*-simpletrace.stap file can generate a binary trace.  Use SystemTap's flightrecorder mode (https://www.sourceware.org/systemtap/SystemTap_Beginners_Guide/using-usage.html#flight-recorder) to run and dump the ring buffer.  You can pretty-print the trace using QEMU scripts/simpletrace.py --no-header ./trace-events path/to/ringbuffer.

lersek, 2016-06-03:
The screenshot is alright, I also used google translate to understand it. I googled the error message, and found nothing useful. :(

I retried with QEMU v2..5.0, no BSOD. Sigh.

Can you try grabbing a KVM trace after the second restat? When the tianocore logo is displayed, press ESC. That will enter the UEFI setup menu, and give you time for starting KVM tracing. Then you can resume the boot process with the Continue option. (I'm grasping at straws, sorry -- it's maddening that it only reproduces on your machine.)

webczat, 2016-06-03:
At this time I am not even reinstalling the system between tests! like
when you recompile firmware without ipv6 the bsod does not happen, with
ipv6 it does, that is all. not a fresh guest each time anymore, so kvm
tracing that is far, far simpler for me now. Unless you have any
specific reason for me to reinstall again. I would not be able to be
fast enough to press esc after second restart of reinstall as I am not
constantly watching this window, and I am blind, so even showing it
somewhere to be visible would not help. I will try doing a trace at this
time.

webczat, 2016-06-03:
It is large. Should I give a dat file or the report? here is a dat file
http://webczatnet.pl/webczat/public/trace.dat
No report. why? because I get this error when trying to generate it:
  cound not load plugin '/usr/lib/trace-cmd/plugins/plugin_kvm.so'
/usr/lib/trace-cmd/plugins/plugin_kvm.so: undefined symbol: ud_translate_att

lersek, 2016-06-03:
@stefanha thank you for the info!

lersek, 2016-06-03:
@webczat thank you for the dat file. I hope I can translate it to a textual report on my end, although it is really big. Downloading...

(Also it's really impressive you can do all this while being blind; I hope I haven't given you a hard time with all my requests.)

lersek, 2016-06-03:
BTW can you reproduce it if you keep IPv6 enabled, but disable the secure boot feature?

lersek, 2016-06-03:
(Also, such trace DATs compress really well... for the next time :))

webczat, 2016-06-03:
With everything except secureboot enabled, there is no bsod. at least it
seems so. I didn't know secureboot takes so much space!

lersek, 2016-06-03:
yeah the SB feature embeds a nice large subset of OpenSSL in the firmware binary.

So, I looked at the KVM trace. Translated to text form, it is 3.3G in size. It is unmanageable. I tried to employ various "tricks" (grepping, sorting, counting unique lines and looking at low-frequency lines, etc), but I'm not seeing anything bad in it. Of course, even if there is anything bad in it, I would likely miss it, at this size.

Now, the question about secure boot and ipv6 is relevant for the following reason. In the OVMF DSC files, we include the `NetworkPkg/IScsiDxe/IScsiDxe.inf` driver only if *both* SB *and* IPv6 support are built into the firmware. Otherwise (if either flag is missing), we build in `MdeModulePkg/Universal/Network/IScsiDxe/IScsiDxe.inf`. (See commit 36c6413f76e5f.)

Thus far you have reported that keeping SB enabled, and flipping just IPv6 back and forth controls the BSOD. Similarly, keeping IPv6 enabled, flipping SB back and forth, it also controls the BSOD.

I suspect that `NetworkPkg/IScsiDxe/IScsiDxe.inf` is somehow related to the crash. For example, it is able to install an ACPI table called `IBFT`, and the guest could definitely see that.

Or else... what happens if you restore the guaranteed-to-BSOD setup (both IPv6 and SB compiled in), but change the network backend? In your setup you have `-netdev br=vmnet` -- do you have some DHCP server on that network for example? What happens with `-netdev user`?

I'm trying to figure out if some special traffic on your network contributes to this BSOD. I guess that would be consistent with the fact that your laptop (and my laptop too, for that matter) are on a different physical subnet than your server. (Given that you employ bridged networking, the VM is affected by the traffic on the server's subnet.)

I've really never seen anything like this before, and debugging it from afar can only lead to semi-random ideas, sorry :(

webczat, 2016-06-03:
I may be able to test it in a while. I am actually bridging to an ipv6
tunnel and to an ipv4 network, so I run dhcpv6/dhcpv4/radvd.
About secureboot/ipv6 being flipped: so then why if everything is
enabled, the thing works properly in case it is a debug build?
About testing other net backend, this is your first bad guess. changing
it to -netdev user,id=net does not help, it bsods too.
Ahh, this time even removing network cable does not stop bsods, because
of one of the other devices like pcieport or virtio scsi, or both of
them. Removing scsi requires reinstall so not sure if I will even try
for now.

webczat, 2016-06-03:
I still think that even if the issue is related to what you specify, it
may be something about timing. you cannot reproduce it, I cannot on a
laptop, but can here, and only if debug is not enabled. debug always
makes everything a little slower. that is my guess, although I cannot
tell which thing is a problem here of course.

[12 more replies]

closed 2016-07-21 by lersek