tianocore/edk2 #97: RFE: OvmfPkg: Set MSR_IA32_FEATURE_CONTROL by following QEMU fw_cfg file

hzzhan9, 2016-06-22:
Hi,

A recent QEMU patch series [[1](https://lists.nongnu.org/archive/html/qemu-devel/2016-06/msg06269.html)] adds a new fw_cfg file "etc/msr_feature_control" to publish the value that the guest firmware is advised to write to MSR_IA32_FEATURE_CONTROL (MSR 0x3a). This should be supported by OVMF as well [[2](https://lists.nongnu.org/archive/html/qemu-devel/2016-06/msg05221.html)].

- What is it good for?
For some hardware features, e.g. Intel VMX and local MCE, OS usually expects certain bits in MSR_IA32_FEATURE_CONTROL are set by firmware. Using corresponding features (e.g access VMX MSRs) without setting those bits can result in #GP.
Instead of adding code to guest firmware everytime a new feature is added to QEMU, we can use a QEMU fw_cfg file to publish the advised value of MSR_IA32_FEATURE_CONTROL.

- Path name and internal format
Path: etc/msr_feature_control
Format: one unsigned 64-bit integer which is the value that is advised to set to MSR_IA32_FEATURE_CONTROL.

- Is MSR_IA32_FEATURE_CONTROL part of chipset state?
Yes, firmware should set MSR_IA32_FEATURE_CONTROL at every boot. Firmware is expected to set  MSR_IA32_FEATURE_CONTROL to the value in etc/msr_feature_control at S3 resume.

lersek, 2016-06-22:
Thanks, the description looks good. A few more questions:
* the uint64_t value is in little endian, right?
* from the SeaBIOS patch set, it seems to me that the MSR has to be written on all APs separately. If that's the case (can you please confirm?), that will significantly complicate the implementation, because then we'll have to modify UefiCpuPkg modules too (CpuDxe minimally), and it could (likely) require changes to the SMM stuff as well (CpuS3DataDxe e.g.).

hzzhan9, 2016-06-23:
- Yes, the value is in little endian.
- Yes, MSR_IA32_FEATURE_CONTROL must be written on BSP and all APs.

lersek, 2016-07-05:
Design discussion started at http://thread.gmane.org/gmane.comp.bios.edk2.devel/14049

lersek, 2016-07-05:
The S3 requirement needs clarification:
http://www.seabios.org/pipermail/seabios/2016-July/010761.html

lersek, 2016-07-06:
Apparently S3 is required.

I'm also attaching a minimal kernel module that can be used for reading any MSR on any CPU, for testing this feature. Just create any empty git repo with `git init`, then apply the patch with `git am`. The commit message contains the build and run instructions.

... Sigh, github is stupid, it doesn't allow me to attach `*.patch`. I have to zip it just to make github happy. Sorry.
[tester.zip](https://github.com/tianocore/edk2/files/349250/tester.zip)

lersek, 2016-07-07:
Posted series:
`[PATCH 0/4] OvmfPkg: program MSR_IA32_FEATURE_CONTROL from fw_cfg on all processors`
http://thread.gmane.org/gmane.comp.bios.edk2.devel/14214

lersek, 2016-07-13:
Posted series
`[PATCH v2 0/5] OvmfPkg:  program MSR_IA32_FEATURE_CONTROL from fw_cfg on all processors`
http://thread.gmane.org/gmane.comp.bios.edk2.devel/14471

lersek, 2016-07-13:
Posted series
`[PATCH v3 0/5] OvmfPkg: program MSR_IA32_FEATURE_CONTROL from fw_cfg on all processors`
http://thread.gmane.org/gmane.comp.bios.edk2.devel/14529

lersek, 2016-07-15:
(Commit range 51773d49de84..dbab994991c7.)

closed 2016-07-15 by lersek

lersek, 2016-07-21:
This (closed) item has been manually migrated to
https://tianocore.acgmultimedia.com/show_bug.cgi?id=86