tianocore/edk2 #98: DEBUG OVMF build fails on UDK2015 branch

bcran, 2016-06-27:
With gcc 6.1.1, the a DEBUG build of OVMF from the UDK2015 branch fails:

./OvmfPkg/build.sh -a X64 -t GCC49 -b DEBUG

/home/bcran/workspace/edk2/UefiCpuPkg/Library/MtrrLib/MtrrLib.c: In function ‘ProgramFixedMtrr’:
/home/bcran/workspace/edk2/UefiCpuPkg/Library/MtrrLib/MtrrLib.c:462:11: error: variable ‘TempQword’ set but not used [-Werror=unused-but-set-variable]
   UINT64  TempQword;
           ^~~~~~~~~
cc1: all warnings being treated as errors
make: *** [GNUmakefile:299: /home/bcran/workspace/edk2/Build/OvmfX64/DEBUG_GCC49/X64/UefiCpuPkg/Library/MtrrLib/MtrrLib/OUTPUT/MtrrLib.obj] Error 1

lersek, 2016-06-28:
Hi Bruce,

on the master branch, this issue has been fixed by commit 16e93125b506e30e53f63be4af0255a3afb0210b (part of a larger series).

I'm unsure what the UDK2015 cherry-picking policy is (considering especially that we don't (yet?) support OvmfPkg outside of the master branch) , but for now I'll move this report to UefiCpuPkg where it belongs.

lgao4, 2016-06-29:
We will sync it to UDK2015 branch.

closed 2022-11-29 by mdkinney