tianocore/edk2 #99: rework gcc detection to look forward instead of backward

olafhering, 2016-06-29:
The way OvmfPkg/build.sh is written it will detect everything not covered as GCC44. Since this file is rarely updated the whole build process will fail, as it does right now with gcc6.

I suggest to rework the detection logic to explicit handle existing, released gcc versions and handle everything unknown as GCC49, aka "the latest".

Doing so will help third party as they do not have to patch their copy of ovmf if the system compiler happens to change. As it stands, each released Xen version will fail to compile just because GCC44 is enforced.

Thanks!

lersek, 2016-06-29:
Considering my current workload, best I can say is "patches welcome". :) Also, I don't use `build.sh` for building OVMF (I prefer direct invocations of the `build` utility with elaborate command line options that I stash in various personal wrapper scripts), so I think any such changes should be primarily reviewed by @jljusten.

(BTW I wonder if BaseTools should get a GCC61 toolchain...)

Thanks!

jljusten, 2016-06-29:
Defaulting to GCC49 seems fine.

@lersek: Regarding GCC61, the new GCC release method would make it GCC6 instead. But, until there a reason for a new toolchain, let's not add one.

lersek, 2016-07-21:
This issue is now tracked by
https://tianocore.acgmultimedia.com/show_bug.cgi?id=62

closed 2016-07-21 by lersek