tianocore/edk2 #103: RFE: OvmfPkg: GOP driver for virtio-gpu-pci

lersek, 2016-07-05:
Specs (in progress) at:
* https://lists.oasis-open.org/archives/virtio-dev/201605/msg00002.html
* https://www.kraxel.org/virtio/

@kraxel, @agraf, you might want to watch this item. (No ETA promises, but it did get bumped a bit higher up on my list due to last week's kvm-arm discussion -- see also https://github.com/tianocore/edk2/issues/84) 

The GOP driver should be separate from QemuVideoDxe. It should be virtio-1.0 only, and should bind our edk2-internal VIRTIO_DEVICE_PROTOCOL only (enforcing the >=1 protocol revision). PCI details and VGA compatibility should be ignored (which includes reverting commit 94210dc95e9f7 -- with the new virtio-gpu-pci driver in place, QemuVideoDxe should just steer clear of the device, and both virtio-gpu-pci and virtio-gpu should be driven by the new driver.)

kraxel, 2016-07-05:
The GOP driver should be separate from QemuVideoDxe. It should be
virtio-1.0 only, and should bind our edk2-internal VIRTIO_DEVICE_PROTOCOL
only (enforcing the >=1 protocol revision). PCI details and VGA
compatibility should be ignored (which includes reverting commit 94210dc
<https://github.com/tianocore/edk2/commit/94210dc95e9f7c6ff4066a9b35a288e6f1c271bf>
-- with the new virtio-gpu-pci driver in place, QemuVideoDxe should just
steer clear of the device, and both virtio-gpu-pci and virtio-gpu should be
driven by the new driver.)

Well, it is also possible to have QemuVideoDxe handle virtio-vga and the
new driver virtio-gpu-pci (and virtio-gpu @ mmio).
You can figure which device is which by checking the pci device class,
virtio-vga has DISPLAY_VGA and virtio-gpu-pci has DISPLAY_OTHER.

lersek, 2016-07-05:
> You can figure which device is which by checking the pci device class, virtio-vga has DISPLAY_VGA and virtio-gpu-pci has DISPLAY_OTHER.

This is easy to do in QemuVideoDxe (which is already a PCI device driver).

In the new driver however I'd like to look at the VIRTIO_DEVICE_PROTOCOL interface only. It wouldn't be very hard to find the related PciIo protocol / handle, and check the PCI device class using that (i.e., read the config space through PciIo), but in a sense it would be a "layering violation". The PciIo interface of the device is already in use by Virtio10Dxe (the generic driver that provides the "modern" VIRTIO_DEVICE_PROTOCOL impl. on top of PciIo).

Admittedly, this would be a one-off check, not a permanent reference (the UEFI spec recognizes this difference; `EFI_OPEN_PROTOCOL_GET_PROTOCOL` vs. `EFI_OPEN_PROTOCOL_BY_DRIVER`.) Still I feel the separation would be cleaner without accessing PciIo in the new driver.

kraxel, 2016-07-05:
2016-07-05 20:09 GMT+02:00 Laszlo Ersek <notifications@github.com>:

> You can figure which device is which by checking the pci device class,
> virtio-vga has DISPLAY_VGA and virtio-gpu-pci has DISPLAY_OTHER.
>
> This is easy to do in QemuVideoDxe (which is already a PCI [...]
Possibly it is easier to do in the virtio-pci driver: Don't bind to
DISPLAY_VGA devices, so virtio-vga wouldn't show up as VIRTIO_DEVICE in the
first place.

lersek, 2016-07-21:
This issue is now tracked by
https://tianocore.acgmultimedia.com/show_bug.cgi?id=66

closed 2016-07-21 by lersek