tianocore/edk2 #106: Boot failed - 64bit OVMF doesn't find /efi/boot/BOOTX64.efi

michael-brade, 2016-07-10:
Hi,

I guess I need some help solving this one. Neither with KNOPPIX, nor with a 64bit Windows 8.1/10 ISO am I able to boot qemu in UEFI mode.

My system:
   Debian unstable, UEFI v2.60 (EDK II), qemu 2.6.0.
   installed Debian OVMF (64 bit).
   Downloaded latest KNOPPIX DVD as iso.

Starting qemu with
```
qemu-system-x86_64 -bios /usr/share/ovmf/OVMF.fd -enable-kvm -cpu host -smp 4 -m 2048 -cdrom KNOPPIX_V7.6.1DVD-2016-01-16-EN.iso  -vga qxl 
```
leads to:
```
Boot failed: EFI DVD/CDROM
Boot failed: EFI Floppy
Boot failed: EFI Floppy 1
```
I then get transferred into UEFI Interactive Shell v2.1, which shows me no FS0 or so in the Mapping table. If it does (no I idea what I did to get it), I can do
```
> FS0:
> ls efi/boot
```
but even though there is a `bootx64.efi` in the directory, it lists *only* bootia32.efi`?! No wonder it cannot boot, right?

Any ideas what is going wrong?

lersek, 2016-07-11:
Hi @michael-brade, for this kind of discussion I suggest that you please subscribe to the mailing list, and ask there: https://lists.01.org/mailman/listinfo/edk2-devel

(NOTE: please await your subscription request's confirmation before you post your actual question!)

We're happy to help, but this bug tracker is for bugs / feature requests, not general discussion.

In this case, you have two (or three) problems in your setup:

* The first (more generic) problem is that you are using the `-bios` option. *Never* use `-bios` with OVMF. Instead:

```
# Copy the varstore template into an actual varstore file for the VM,
# if the actual varstore file doesn't exist yet. This must be done separately
# for each VM.
if ! [ -e vm1_varstore.fd ]; then
  cp .../OVMF_VARS.fd vm1_varstore.fd
fi

# Start the VM. The firmware binary is shared by all VMs.
qemu-system-x86_64 \
    -drive if=pflash,format=raw,unit=0,file=.../OVMF_CODE.fd,readonly=on \
    -drive if=pflash,format=raw,unit=1,file=vm1_varstore.fd \
    ...
```

* Second, you can't boot a 32-bit operating system / boot loader on 64-bit UEFI firmware. If your OVMF binary comes from the Ia32X64 or the X64 build, then you can only boot 64-bit UEFI boot loaders with it (Linux and Windows alike). For Windows installers, look for `x64` (as opposed to `x86`) in the ISO file name.

* Third, what you see in the ISO9660 filesystem on the CD-ROM image is entirely separate from what you see in the UEFI El Torito boot image on the CD. Please refer to #93.

... Actually, your report seems to be a duplicate of issue #93. I'm getting the impression that Debian have mis-built their installer media (or installer preparation scripts), and now we're getting reports from users of Debian and derivatives.

So, I'm closing this. If you'd like to discuss it further, please let's take it to the mailing list. Thanks.

closed 2016-07-11 by lersek

michael-brade, 2016-07-19:
Hi Laszlo, thank you very much for your explanation, it is spot on and fixed my problem. Sorry for using the wrong channel, I'll use the mailinglist if I need more help. Cheers!