tianocore/edk2 #3874: i need help with using edk2, and installing it

dropier, 2023-01-10:
i was lead here by a youtube video that this is related to being able to run uefi operating systems on android phones, if it is, could someone give me a lead with a guide on how to use it to run windows 10 arm/any linux distro that is good for an android phone?
i searched around a bit and nothing about a guide here on this github repo. i would absolutely love it if someone helped me here, thank you.

MarcoLopez02, 2023-01-18:
https://github.com/tianocore/tianocore.github.io/wiki/EDK-II-Documentation

lersek, 2023-01-20:
@dropier I don't clearly understand your goal (and I don't know the statements that the video (?) may have made about edk2), but, as far as I can tell, the edk2 repository *in itself* will not help you "run uefi operating systems on android phones". Edk2 stands for "EFI Development Kit II"; it is a kit of tools, libraries, drivers and core firmware modules for implementing PI/UEFI platform firmware, platform drivers, UEFI drivers, and UEFI applications.

MohitSavner, 2023-01-30:
@dropier you can refer this [link ](https://www.youtube.com/watch?v=V2EAccnaSvo&list=PLz-YdBAdSGeh8MT3c7_sfolRzXM8cNXPk&ab_channel=AshrafAliS) for EDK2 installation and for more can follow the series.

wangzhankun, 2023-02-05:
Hello sir, I am using edk2 as the bios for loongson. When I try to boot on qemu, I find an assert. at `edk2-platforms/Platform/Loongson/loongArchQemuPkg/Library/NorFlashQemuLib.c(56): !(((INTN)(RETURN_STATUS)(FindNodeStatus))<0)`. I don't know how to report and how to get help. I have to report here and could you tell me where to report the bug and get help?

lersek, 2023-10-04:
I don't necessarily believe any topics raised here have been resolved, but this ticket is stale, and the github.com issue tracker isn't the right place for ... anything related to edk2, really. Please peruse one of the mailing lists (although I can't promise that you'll get any better feedback there).

closed 2023-10-04 by lersek