tianocore/edk2 #4055: I NEED HELP! I

fisheggM, 2023-02-18:
this is the log in the terminal when i execute "edksetup forcerebuild Vs2019"

 Workspace is not changed.
          PATH      = C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\Bin\Win32;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\IDE\\Extensions\Microsoft\IntelliCode\CLI;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX64\x64;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\IDE\VC\VCPackages;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\IDE\CommonExtensions\Microsoft\TestWindow;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\IDE\CommonExtensions\Microsoft\TeamFoundation\Team Explorer;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\MSBuild\Current\bin\Roslyn;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Team Tools\Performance Tools\x64;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Team Tools\Performance Tools;C:\Program Files (x86)\Microsoft Visual Studio\Shared\Common\VSPerfCollectionTools\vs2019\\x64;C:\Program Files (x86)\Microsoft Visual Studio\Shared\Common\VSPerfCollectionTools\vs2019\;C:\Program Files (x86)\Microsoft SDKs\Windows\v10.0A\bin\NETFX 4.8 Tools\x64\;C:\Program Files (x86)\HTML Help Workshop;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\Tools\devinit;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\\MSBuild\Current\Bin;C:\WINDOWS\Microsoft.NET\Framework64\v4.0.30319;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\IDE\;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\Tools\;C:\Program Files\Eclipse Foundation\jdk-8.0.302.8-hotspot\bin;C:\WINDOWS\System32\OpenSSH\;C:\WINDOWS\System32\Wbem;E:\mongoserver\bin;C:\Users\Administrator\AppData\Roaming\python3;E:\PythonMap\Python_v3.11.1\;E:\PythonMap\Python_v3.11.1\\Scripts\;C:\Users\Administrator\AppData\Roaming\python2\appdata\Lib\site-packages;C:\Users\Administrator\AppData\Roaming\python3\Scripts;C:\ProgramData\chocolatey\bin;C:\Program Files\NASM;C:\Program Files\qemu;C:\Users\a\Desktop\mingw64\bin;C:\Program Files\Microsoft VS Code\bin;C:\Program Files\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\NASM;C:\IASL;C:\Windows\System32;C:\WINDOWS\system32;SystemRoot%\system32;C:\WINDOWS;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\IncrediBuild;C:\Program Files\Git\cmd;E:\crgwin;C:\Users\a\scoop\shims;C:\Users\a\AppData\Local\Microsoft\WindowsApps;;C:\Program Files\JetBrains\CLion 2022.3.1\bin;;C:\Program Files\JetBrains\IntelliJ IDEA 2022.3.1\bin;;C:\Program Files\JetBrains\PyCharm 2022.3.1\bin;;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\VC\Tools\Llvm\x64\bin;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\Ninja;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\IDE\VC\Linux\bin\ConnectionManagerExe

     WORKSPACE      = C:\FishLoader
EDK_TOOLS_PATH      = C:\FishLoader\BaseTools
BASE_TOOLS_PATH     = C:\FishLoader\BaseTools
 EDK_TOOLS_BIN      = C:\FishLoader\BaseTools\Bin\Win32
     CONF_PATH      = C:\FishLoader\Conf

Using Pip Basetools
               PATH = C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\BinPipWrappers\WindowsLike;C:\FishLoader\BaseTools\Bin\Win32;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\IDE\\Extensions\Microsoft\IntelliCode\CLI;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX64\x64;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\IDE\VC\VCPackages;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\IDE\CommonExtensions\Microsoft\TestWindow;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\IDE\CommonExtensions\Microsoft\TeamFoundation\Team Explorer;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\MSBuild\Current\bin\Roslyn;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Team Tools\Performance Tools\x64;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Team Tools\Performance Tools;C:\Program Files (x86)\Microsoft Visual Studio\Shared\Common\VSPerfCollectionTools\vs2019\\x64;C:\Program Files (x86)\Microsoft Visual Studio\Shared\Common\VSPerfCollectionTools\vs2019\;C:\Program Files (x86)\Microsoft SDKs\Windows\v10.0A\bin\NETFX 4.8 Tools\x64\;C:\Program Files (x86)\HTML Help Workshop;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\Tools\devinit;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\\MSBuild\Current\Bin;C:\WINDOWS\Microsoft.NET\Framework64\v4.0.30319;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\IDE\;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\Tools\;C:\Program Files\Eclipse Foundation\jdk-8.0.302.8-hotspot\bin;C:\WINDOWS\System32\OpenSSH\;C:\WINDOWS\System32\Wbem;E:\mongoserver\bin;C:\Users\Administrator\AppData\Roaming\python3;E:\PythonMap\Python_v3.11.1\;E:\PythonMap\Python_v3.11.1\\Scripts\;C:\Users\Administrator\AppData\Roaming\python2\appdata\Lib\site-packages;C:\Users\Administrator\AppData\Roaming\python3\Scripts;C:\ProgramData\chocolatey\bin;C:\Program Files\NASM;C:\Program Files\qemu;C:\Users\a\Desktop\mingw64\bin;C:\Program Files\Microsoft VS Code\bin;C:\Program Files\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\NASM;C:\IASL;C:\Windows\System32;C:\WINDOWS\system32;SystemRoot%\system32;C:\WINDOWS;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\IncrediBuild;C:\Program Files\Git\cmd;E:\crgwin;C:\Users\a\scoop\shims;C:\Users\a\AppData\Local\Microsoft\WindowsApps;;C:\Program Files\JetBrains\CLion 2022.3.1\bin;;C:\Program Files\JetBrains\IntelliJ IDEA 2022.3.1\bin;;C:\Program Files\JetBrains\PyCharm 2022.3.1\bin;;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\VC\Tools\Llvm\x64\bin;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\Ninja;E:\Visual Studio 2022\Visual Studio IDE\2019\Community\Common7\IDE\VC\Linux\bin\ConnectionManagerExe
         PYTHONPATH =

Microsoft (R) Program Maintenance Utility Version 14.29.30148.0
Copyright (C) Microsoft Corporation.  All rights reserved.

execute command "nmake cleanall" in directory C:\FishLoader\BaseTools\Source\Python

Microsoft (R) Program Maintenance Utility Version 14.29.30148.0
Copyright (C) Microsoft Corporation.  All rights reserved.

Could Not Find C:\FishLoader\BaseTools\Source\Python\*.pyc

C:\FishLoader\BaseTools\Source\C\Makefiles\NmakeSubdirs.py:135: DeprecationWarning: currentThread() is deprecated, use current_thread() instead
  self.running.remove(threading.currentThread())
subprocess not exit successfully
Error while execute command 'nmake cleanall' in direcotry C:\FishLoader\BaseTools\Source\C

Microsoft (R) Program Maintenance Utility Version 14.29.30148.0
Copyright (C) Microsoft Corporation.  All rights reserved.

C:\FishLoader\BaseTools\Source\C\Makefiles\NmakeSubdirs.py:135: DeprecationWarning: currentThread() is deprecated, use current_thread() instead
  self.running.remove(threading.currentThread())
execute command "nmake cleanall" in directory C:\FishLoader\BaseTools\Source\C\Common

Microsoft (R) Program Maintenance Utility Version 14.29.30148.0
[... cut ...]
BasePeCoff.c(1477): warning C4311: 'type cast': pointer truncation from 'void *' to 'UINTN'
BasePeCoff.c(1477): warning C4311: 'type cast': pointer truncation from 'void *' to 'UINTN'
BasePeCoff.c(1477): warning C4312: 'type cast': conversion from 'UINTN' to 'void *' of greater size
NMAKE : fatal error U1077: '"E:\Visual Studio 2022\Visual Studio IDE\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX64\x64\cl.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: 'if' : return code '0x1'
Stop.
NMAKE : fatal error U1077: 'if' : return code '0x1'
Stop.

Javagedes, 2023-06-26:
It looks like you did not compile BaseTools before attempting to do an actual build.

gdong1, 2023-08-28:
Close this issue since it was opened for a long time and there is no update.

closed 2023-08-28 by gdong1