tianocore/edk2 #5111: Change devcontainer to Fedora 39

osteffenrh, 2023-12-06:


osteffenrh, 2023-12-06:
Depends on https://github.com/tianocore/containers/issues/84

lersek, 2024-02-26:
@osteffenrh https://github.com/tianocore/containers/pull/88 seems to have been merged :)

osteffenrh, 2024-02-26:
@lersek, true.
I would like to keep devcontainer and CI image the same. Last time I tried to switch CI to Fedora 39, I had problems with gcc13 and some CI job.

Let me check that again first.

helix-osdev, 2024-03-12:
> @lersek, true. I would like to keep devcontainer and CI image the same. Last time I tried to switch CI to Fedora 39, I had problems with gcc13 and some CI job.
> 
> Let me check that again first.

Sounds like a skill issue.

osteffenrh, 2024-03-12:
@helix-osdev, contributions are always welcome.

bcran, 2024-06-19:
I just saw this issue. I updated the devcontainer to Fedora 39 a few days ago - see https://github.com/tianocore/edk2/pull/5782

closed 2024-06-19 by bcran