tianocore/edk2 #6135: Bad variable interpolation in UefiPayloadPkg.dsc

matdibu, 2024-08-29:
https://github.com/tianocore/edk2/blob/bb248a95091ab542440053d9c289a97e80eb6630/UefiPayloadPkg/UefiPayloadPkg.dsc#L25

does that actually evaluate as intended? on my local machine I get this:
```
$ ls Build/
'UefiPayloadPkg$(BUILD_ARCH)'
```

ProfessorLongBeard, 2024-08-29:
> https://github.com/tianocore/edk2/blob/bb248a95091ab542440053d9c289a97e80eb6630/UefiPayloadPkg/UefiPayloadPkg.dsc#L25
> 
> does that actually evaluate as intended? on my local machine I get this:
> 
> ```
> $ ls Build/
> 'UefiPayloadPkg$(BUILD_ARCH)'
> ```
> 
> which in turn breaks the build

Seems like your local machine is garbage.

matdibu, 2024-08-30:
:))) I didn't expect that

but seriously now, could it be because of the shell using during the build?

it was from within a nix derivation, so the shell definitely isn't the same as a user-interactive bash session started on ubuntu

that being said, the build worked after I replaced that with `Build/UefiPayloadPkgX64`, and I successfully flashed and booted on real hardware

iangoegebuer, 2024-11-27:
The line should be the below

```
 OUTPUT_DIRECTORY                    = Build/UefiPayloadPkg$(ARCH) 
```

Not sure what's up with @ProfessorLongBeard s system. Likely some leftover environment vars.

No other dsc uses BUILD_ARCH

```
grep -r OUTPUT_DIRECTORY | grep ARCH
```

matdibu, 2024-11-27:
@dhaval-rivos could you chime in?

matdibu, 2024-11-27:
@iangoegebuer if I use `Build/UefiPayloadPkg$(ARCH)` then I'm back to this problem
```
build.py...
/build/source/payloads/external/edk2/workspace/MrChromebox/UefiPayloadPkg/UefiPayloadPkg.dsc(25): error 3001: No space is allowed in OUTPUT_DIRECTORY
     Build/UefiPayloadPkgIA32 X64
```

dhaval-rivos, 2024-11-27:
So in your case $(ARCH)  is resulting into UefiPayloadPkgIA32 X64 with a space and that is the problem? It is supposed to come from [ input build arg -a: parser.add_argument('-a', '--Arch'] and used in BuildUniversalPayload().
I am sure it works in X86 environment as @ChaselChiu and teams have been using it.

matdibu, 2024-11-27:
yeah, coreboot builds `UefiPayloadPkgX64` via `build -a IA32 -a X64`
is that how it's supposed to be used?