tianocore/edk2 #6398: The subhook upstream is gone

dvzrv, 2024-11-03:
Hi! :wave: 

I'm trying to adapt the Arch Linux package setup for this project.

Unfortunately it appears that the upstream project subhook is now gone, but is still being used as a submodule:

https://github.com/tianocore/edk2/blob/47ba459fc1c2cda532b5d7cf16c9fdcbb200784d/.gitmodules#L28C1-L28C43

It would be great if you could fix this. Many thanks! :pray:

os-d, 2024-11-03:
Thanks for the heads up. @mdkinney @spbrogan  @makubacki , it indeed looks like subhook has been removed from GitHub and I don’t see where else it has migrated to. All of Zeex’s (the subhook owner) repos are gone. Looks like we will need to carry our own copy or choose a different path. I believe subhook was brought in for GoogleTest and that is the only current consumer.

kuqin12, 2024-11-03:
Is there any real usage for this submodule? I understand it is supported through the header. But the intended usage seems like a hack for big files that are hard to test. I understand this might be useful for existing codebase, but I think for newly developed modules, this model should not be encouraged.

In addition, the repo itself is incomplete and brought in issues of expanding to arm64 environment. Should we just drop its usage in edk2?

mdkinney, 2024-11-03:
Yes.  They are used from the macros that support using google mocks for internal functions.  I will investigate what has happened and what alternatives may be available.  The use case is for host based unit testing which is enabled for IA32/X64.  Are there plans to expand host based unit testing to more architectures?

kuqin12, 2024-11-03:
I think expanding to more architectures will allow testing on the native environments (i.e. arm, risc-v). At this point, google test will only allow logic testing, which will be of limited usage if the routine is very low level like assembly functions.

mdkinney, 2024-11-03:
Agree.  Testing assembly function can be hard.  If assembly code is in a lib function and it only uses instructions that are allowed from a user mode application, then host based unit testing is possible.

There are a number of updates that would be required in the UnitTestFrameworkPkg and potentially the stuart tool CI usages to 
enable additional architectures for host based unit testing.

mdkinney, 2024-11-04:
I will add mirror to TianoCore and will update status here and on mailing list

mdkinney, 2024-11-04:
Mirror created: https://github.com/tianocore/edk2-subhook
PR opened to use mirror: https://github.com/tianocore/edk2/pull/6402

@leiflindholm and @ajfish please review PR for use of edk2 mirror of subhook to unblock dev and CI clones.

bssrikanth, 2024-11-05:
@mdkinney I see #6402 is now merged which will unblock CI runs on master branch. We have CIs which uses stable branch of edk2, they still fail :(

mdkinney, 2024-11-05:
There is a release planned for 11-22-24.  https://github.com/tianocore/tianocore.github.io/wiki/EDK-II-Release-Planning

We can cherry-pick this fix to the previous stable tag edk2-stable202408 if that is required.

bssrikanth, 2024-11-05:
> There is a release planned for 11-22-24. https://github.com/tianocore/tianocore.github.io/wiki/EDK-II-Release-Planning
> 
> We can cherry-pick this fix to the previous stable tag edk2-stable202408 if that is required.

That would be helpful :)

rnertney, 2024-11-05:
> There is a release planned for 11-22-24. https://github.com/tianocore/tianocore.github.io/wiki/EDK-II-Release-Planning
> 
> We can cherry-pick this fix to the previous stable tag edk2-stable202408 if that is required.

This would be a much appreciated cherry pick.

eric-unc, 2024-11-05:
Yeah, I would also appreciate it :)

mdkinney, 2024-11-05:
@lgao4 in order to complete this request, we need to follow the process outlined in RFC https://edk2.groups.io/g/rfc/message/474.

This would include
* Create new branch in edk2 repo with branch name `stable/202408`
* Apply PR https://github.com/tianocore/edk2/pull/6402 to stable/202408 and verify CI pass
* Add tag name `edk2-stable202408.01` to head of `stable/202408` branch

mdkinney, 2024-11-05:
@lgao4 I have completed first 2 steps

* Created branch https://github.com/tianocore/edk2/tree/stable/202408 from `edk2-stable202408` tag
* Open PR https://github.com/tianocore/edk2/pull/6407 with commit https://github.com/tianocore/edk2/pull/6402/commits/12aed1cb330368fe3027b09d7b9079d9fab441b7 against https://github.com/tianocore/edk2/tree/stable/202408

gdong1, 2024-11-06:
Is it possible to fix the same build failure in the old stable branch (e.g.: edk2-stable202311)? The projects using the statable branch also failed to build. It would be great if this fix could apply to old stable branches.

mdkinney, 2024-11-06:
Release published and edk2-stable202408.01 tag created:

https://github.com/tianocore/edk2/releases/tag/edk2-stable202408.01

mdkinney, 2024-11-06:
@gdong1 That would be up to 5 hot fix releases and the process being followed https://edk2.groups.io/g/rfc/message/474 only documents supporting the most recent stable tag release.

gdong1, 2024-11-06:
Got it. Thanks @mdkinney for this info.

dvzrv, 2024-11-10:
Many thanks for looking into this!

I've successfully updated to edk2-stable202408.01 and I guess this can be closed :)

closed 2024-11-10 by dvzrv