tianocore/edk2 #6876: CapsuleApp  add –NR (no-reset) flag to skip reset according to user reset (Bugzilla Bug 388)

tianocore-issues, 2024-12-19:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 388
Date: 2017-02-19T20:26:27+00:00
From: Wang Cloud &lt;<winggundum82>&gt;
To:   @lzeng14
CC:   @lzeng14, yonghong.zhu

Last updated: 2017-03-21T03:15:14+00:00

tianocore-issues, 2024-12-19:
## Comment 1473
Date: 2017-02-19 20:26:27 +0000
From: Wang Cloud &lt;<winggundum82>&gt;

- Industry Specification: ---
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): @lzeng14

user cannot distinguish whether the reset is triggered by bios code or application code , there is no difference for CAPSULE_FLAGS_PERSIST_ACROSS_RESET only capsule and CAPSULE_FLAGS_PERSIST_ACROSS_RESET + CAPSULE_FLAGS_INITIATE_RESET capsule in current design.To make the solution complete,please add –NR (no-reset) flag for CAPSULE_FLAGS_PERSIST_ACROSS_RESET only capsule.
For CAPSULE_FLAGS_PERSIST_ACROSS_RESET + CAPSULE_FLAGS_INITIATE_RESET capsule ,it should still reset after bios update complete

tianocore-issues, 2024-12-19:
## Comment 1652
Date: 2017-03-21 02:51:53 +0000
From: @lzeng14

Fixed at 7043a90eee1150db1d55cdde2a98700fa10838e5

tianocore-issues, 2024-12-19:
## Comment 1653
Date: 2017-03-21 03:15:14 +0000
From: Wang Cloud &lt;<winggundum82>&gt;

Verified in latest edk2 trunk

pierregondois, 2026-10-07:
Effectively fixed with https://github.com/tianocore/edk2/commit/7043a90eee1150db1d55cdde2a98700fa10838e5

closed 2026-10-07 by pierregondois