tianocore/edk2 #10022: AllocatePool use wrong size value (Bugzilla Bug 3139)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 3139
Date: 2020-12-25T01:57:42+00:00
From: @LiuZhiguang001
To:   @ardbiesheuvel
CC:   @lgao4, @pierregondois

Last updated: 2023-09-29T08:12:28+00:00

tianocore-issues, 2024-12-20:
## Comment 14665
Date: 2020-12-25 01:57:42 +0000
From: @LiuZhiguang001

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): @ardbiesheuvel

In EmbeddedPkg\Library\PrePiMemoryAllocationLib\MemoryAllocationLib.c
Function AllocatePool, line 187
Hob = (EFI_HOB_MEMORY_POOL *)CreateHob (EFI_HOB_TYPE_MEMORY_POOL, (UINT16)(sizeof (EFI_HOB_TYPE_MEMORY_POOL) + AllocationSize));

it shouldn&apos;t use &quot;sizeof (EFI_HOB_TYPE_MEMORY_POOL)&quot; but should use &quot;sizeof (EFI_HOB_MEMORY_POOL)&quot;

tianocore-issues, 2024-12-20:
## Comment 14725
Date: 2021-01-05 21:53:41 +0000
From: @lgao4

Ard: can you fix it?

tianocore-issues, 2024-12-20:
## Comment 21949
Date: 2023-09-29 08:12:28 +0000
From: @pierregondois

This should be fixed by:
https://edk2.groups.io/g/devel/message/69807 (merged at 181981715591)

pierregondois, 2025-06-11:
Closing the issue as this was merged a long time ago