tianocore/edk2 #10046: CpuDxe may allocate above 4GB memory for GDT (Bugzilla Bug 3233)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 3233
Date: 2021-02-23T20:43:58+00:00
From: @niruiyu
To:   @niruiyu
CC:   @lgao4, patrick.rudolph

Last updated: 2021-06-22T05:09:52+00:00

tianocore-issues, 2024-12-20:
## Comment 15110
Date: 2021-02-23 20:43:58 +0000
From: @niruiyu

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): @niruiyu

GDT needs to be allocated below 4GB in 64bit environment because AP needs it for entering to protected mode. CPU running in big real mode cannot access above 4GB GDT.

But CpuDxe driver contains below code:
  gdt = AllocateRuntimePool (sizeof (GdtTemplate) + 8);
  ..... 
  gdtPtr.Base = (UINT32)(UINTN)(VOID*) gdt;

The AllocateRuntimePool() may allocate memory above 4GB.
Thus, we cannot use AllocateRuntimePool (), instead, we should use AllocatePages() to make sure GDT is below 4GB space.
  gdt = 0xFFFFFFFF;
  Status = gBS-&gt;AllocatePages (
                  AllocateMaxAddress,
                  EfiRuntimeServicesData,
                  EFI_SIZE_TO_PAGES (sizeof (GdtTemplate) + 8),
                  &amp;gdt 
                  );

Thanks to Jeff@Byosoft for reviewing the whole code flow and providing the code change suggestion.

tianocore-issues, 2024-12-20:
## Comment 15111
Date: 2021-02-23 20:49:37 +0000
From: @niruiyu

The bug can be seen when there is not enough memory below 4GB.
The system hangs when AP is about to enter protected mode.

tianocore-issues, 2024-12-20:
## Comment 15113
Date: 2021-02-23 21:38:35 +0000
From: @lgao4

Ray: can you take it?

tianocore-issues, 2024-12-20:
## Comment 15126
Date: 2021-02-24 00:54:01 +0000
From: @niruiyu

I can take it if Jeff doesn&apos;t want to:)

tianocore-issues, 2024-12-20:
## Comment 16416
Date: 2021-06-22 05:09:52 +0000
From: Patrick Rudolph &lt;<patrick.rudolph>&gt;

Any update on this issue?

pierregondois, 2026-10-07:
Fixed by: https://github.com/tianocore/edk2/commit/313d86c956d4599054a9dcd524668f67797d317a

closed 2026-10-07 by pierregondois