tianocore/edk2 #10090: About the efficiency of Shell command (Bugzilla Bug 3388)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 3388
Date: 2021-05-11T07:03:37+00:00
From: ptabckimo
To:   unassigned &lt;<unassigned>&gt;
CC:   @lgao4

See also:     https://bugzilla.tianocore.org/show_bug.cgi?id=3151
Last updated: 2022-03-16T12:28:57+00:00

tianocore-issues, 2024-12-20:
## Comment 15988
Date: 2021-05-11 07:03:37 +0000
From: ptabckimo

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): unassigned &lt;<unassigned>&gt;

Created attachment 723
Shell Grantley

Hi,

There is a problem with the runtime efficiency of Shell.
We have two different Shells from Grantley and Whitley.
When we used Grantley Shell to run the command and it costs 5 secs.
When we used Whitley Shell to run the command and it costs 100 secs.
This issue happens on all platforms.
Please advise us on how to speed up the efficiency.
Thanks.

Test Step:
1. Prepare a UEFI bootable USB drive and copy Shell_Grantley.efi and Shell_Whitley.efi into the USB drive.
2. Plug the USB drive in the Whitley machine.
3. Power on and boot into Builtin shell.
4. Run EFI program Shell_Grantley.efi. 
5. Start the timer and run the command &quot;ls -r Shell*&quot;. 
6. Exit and return to Builtin shell
7. Redo step 3 to step 6 and only change to run Shell_Whitley.efi.

Attachment: [BZ-723-Shell_Grantley.efi](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-723-Shell_Grantley.efi)

tianocore-issues, 2024-12-20:
## Comment 15989
Date: 2021-05-11 07:04:47 +0000
From: ptabckimo

Created attachment 724
Shell Whitley

Attachment: [BZ-724-Shell_Whitley.7z](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-724-Shell_Whitley.7z)

tianocore-issues, 2024-12-20:
## Comment 16011
Date: 2021-05-11 21:41:14 +0000
From: @lgao4

Do you know where are these shell from? Which Edk release and Shell version.

tianocore-issues, 2024-12-20:
## Comment 16038
Date: 2021-05-12 07:36:15 +0000
From: ptabckimo

I have try UDK2010 and UDK2018.

The Shell from UDK2010 is fast than the Shell fromUDK2018.

UDK2010:
https://github.com/tianocore/tianocore.github.io/wiki/UDK2010

UDK2018:
https://github.com/tianocore/tianocore.github.io/wiki/UDK2010

We think it related to the size and architecture of the file system.
In our USB drive, there are 4635 files in 255 directories and the total usage size is 4.92GB.
The files scattered in different folders.
The file system architecture is like 1 to 6 layers,
For example, layer 1 is fs0: ; the layer 2 is EFI ; the layer 3 is python27; the layer 4 is email; the layer 5 is test; the layer 6 is data.
So the layer 6 path is fs0:\EFI\StdLib\lib\python.27\email\test\data.

And search fs0: from the file system.

tianocore-issues, 2024-12-20:
## Comment 16052
Date: 2021-05-13 08:53:50 +0000
From: @lersek

See also: bug 3151.

tianocore-issues, 2024-12-20:
## Comment 16053
Date: 2021-05-13 08:57:12 +0000
From: @lersek

To be more explicit, I think that, with bug 3151 fixed, the symptom reported in this ticket should disappear.

tianocore-issues, 2024-12-20:
## Comment 16085
Date: 2021-05-17 21:38:02 +0000
From: ptabckimo

Thank you, Laszlo.
We will try that.

tianocore-issues, 2024-12-20:
## Comment 18632
Date: 2022-03-16 12:28:57 +0000
From: ptabckimo

I will try it this week.

tianocore-issues, 2024-12-20:
## Comment 18620
Date: 2022-03-15 22:19:52 +0000
From: @lgao4

ptabckimo@yahoo.com.tw: have you verified the latest Shell?

pierregondois, 2025-07-30:
There is no news for more than 3 years. It is assumed the issue is not visible anymore...

closed 2025-07-30 by pierregondois