tianocore/edk2 #10094: EFI_MM_COMMUNICATE_HEADER has architecture dependent sized field (Bugzilla Bug 3398)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 3398
Date: 2021-05-17T23:19:36+00:00
From: @kuqin12
To:   @kuqin12
CC:   @lgao4

Last updated: 2022-01-14T15:21:08+00:00

tianocore-issues, 2024-12-20:
## Comment 16087
Date: 2021-05-17 23:19:36 +0000
From: @kuqin12

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): @kuqin12

The MessageLength field of EFI_MM_COMMUNICATE_HEADER (also defined as EFI_SMM_COMMUNICATE_HEADER) is currently defined as type UINTN per https://github.com/tianocore/edk2/blob/1fbf5e30ae8eb725f4e10984f7b0a208f78abbd0/MdePkg/Include/Protocol/MmCommunication.h#L29.

But this structure, as a generic definition, could be used for both PEI and DXE MM communication. Thus for a system that supports PEI MM launch, but operates PEI in 32bit mode and MM foundation in 64bit, the current EFI_MM_COMMUNICATE_HEADER definition will cause structure parse error due to UINTN used.

The suggested fix is to change this field defined as UINT64: https://github.com/kuqin12/mu_basecore/commit/b396dfe54b5f7556a12622cd94a5fa869086c4d3

tianocore-issues, 2024-12-20:
## Comment 16101
Date: 2021-05-19 04:21:36 +0000
From: @kuqin12

Note that this change could cause some modules to fail to build. So far identified:
https://github.com/tianocore/edk2/blob/15ee7b76891a78141e6e30ef3f8572e8d6b326d2/MdeModulePkg/Core/PiSmmCore/PiSmmIpl.c#L530

https://github.com/tianocore/edk2/blob/15ee7b76891a78141e6e30ef3f8572e8d6b326d2/MdeModulePkg/Application/SmiHandlerProfileInfo/SmiHandlerProfileInfo.c#L119

https://github.com/tianocore/edk2/blob/15ee7b76891a78141e6e30ef3f8572e8d6b326d2/MdeModulePkg/Application/SmiHandlerProfileInfo/SmiHandlerProfileInfo.c#L152

https://github.com/tianocore/edk2/blob/15ee7b76891a78141e6e30ef3f8572e8d6b326d2/MdeModulePkg/Application/MemoryProfileInfo/MemoryProfileInfo.c#L1193

https://github.com/tianocore/edk2/blob/15ee7b76891a78141e6e30ef3f8572e8d6b326d2/MdeModulePkg/Application/MemoryProfileInfo/MemoryProfileInfo.c#L1216

https://github.com/tianocore/edk2/blob/15ee7b76891a78141e6e30ef3f8572e8d6b326d2/MdeModulePkg/Application/MemoryProfileInfo/MemoryProfileInfo.c#L1233

https://github.com/tianocore/edk2/blob/15ee7b76891a78141e6e30ef3f8572e8d6b326d2/MdeModulePkg/Application/MemoryProfileInfo/MemoryProfileInfo.c#L1264

https://github.com/tianocore/edk2/blob/15ee7b76891a78141e6e30ef3f8572e8d6b326d2/MdeModulePkg/Application/MemoryProfileInfo/MemoryProfileInfo.c#L1323

Thus if anyone has a better idea in fixing the original issue (size mismatch in IA32 vs X64), please also propose.

tianocore-issues, 2024-12-20:
## Comment 16148
Date: 2021-05-25 21:50:23 +0000
From: @lgao4

Kun, this strucure is defined in PI spec. So, need to update spec first. This should belong to the code first.

tianocore-issues, 2024-12-20:
## Comment 16161
Date: 2021-05-25 22:37:39 +0000
From: @kuqin12

(In reply to gaoliming from comment #2)
&gt; Kun, this strucure is defined in PI spec. So, need to update spec first.
&gt; This should belong to the code first.

Liming, thanks for letting me know about the process. How do you think of the original issue (the UINTN in this structure will be parsed differently on 32bit PEI + 64bit MM foundation)?

Could you please let me know how to request PI change if the above issue is agreed/confirmed? Thanks!

tianocore-issues, 2024-12-20:
## Comment 16186
Date: 2021-05-26 21:23:15 +0000
From: @lgao4

Kun, I agree this is a issue in PI spec. Seemly, original design doesn&apos;t consider the usage model of 32bit and 64bit standalone MM module. 

Here is code first process: https://github.com/tianocore/tianocore.github.io/wiki/EDK-II-Code-First-Process

tianocore-issues, 2024-12-20:
## Comment 16192
Date: 2021-05-27 02:52:04 +0000
From: @kuqin12

(In reply to gaoliming from comment #4)
&gt; Kun, I agree this is a issue in PI spec. Seemly, original design doesn&apos;t
&gt; consider the usage model of 32bit and 64bit standalone MM module. 
&gt; 
&gt; Here is code first process:
&gt; https://github.com/tianocore/tianocore.github.io/wiki/EDK-II-Code-First-
&gt; Process

Thanks for the link. I will follow up with what I can do from here.

tianocore-issues, 2024-12-20:
## Comment 18141
Date: 2022-01-14 15:21:08 +0000
From: @kuqin12

v4 Patches sent here to request change of PI spec: https://edk2.groups.io/g/devel/message/85308

tianocore-issues, 2024-12-20:
## Comment 16223
Date: 2021-05-31 21:39:37 +0000
From: @lgao4

Kun takes it.

tianocore-issues, 2024-12-20:
## Comment 16255
Date: 2021-06-02 23:18:54 +0000
From: @kuqin12

The code first ticket is created here: https://bugzilla.tianocore.org/show_bug.cgi?id=3430

The branch for reference implementation is currently here: https://github.com/kuqin12/edk2/tree/BZ3398-MmCommunicate-Length

The PR that passes CI build check is here: https://github.com/tianocore/edk2/pull/1686

sachinami, 2025-01-28:
Hi @kuqin12 

As the required modifications to the specification has been done in PI 1.9. Would it be possible now to merge these changes to edk2 master.

kuqin12, 2025-01-31:
Hi @sachinami 

This is on the roadmap of upstreaming. I expected it to occur sometime in February.

sachinami, 2025-02-01:
Thank you @kuqin12

Looking forward to it!

sachinami, 2025-03-13:
Hi @kuqin12 

Are we ready to upstream the changes.

kuqin12, 2025-03-13:
Actually, just in time. Here is the draft PR: https://github.com/tianocore/edk2/pull/10854. I added the support for Stmm PEI IPL as well. Will publish the PR once I perform more testing some time this week.

sachinami, 2025-03-13:
Thank you @kuqin12. I will track the PR.

kuqin12, 2025-04-07:
Resolved with https://github.com/tianocore/edk2/pull/10854.