tianocore/edk2 #10100: build.py: post-prebuild environment capturing breaks with multi-line variables (Bugzilla Bug 3409)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 3409
Date: 2021-05-21T13:57:12+00:00
From: @lersek
To:   @BobCF
CC:   @lgao4, @lersek

Last updated: 2023-11-10T12:20:18+00:00

tianocore-issues, 2024-12-20:
## Comment 16110
Date: 2021-05-21 13:57:12 +0000
From: @lersek

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): @BobCF

In commit f0dc69e61bf2 (&quot;BaseTools: Add support to merge Prebuild and Postbuild into build Process&quot;, 2016-04-07), logic was added to &quot;build.py&quot; for saving the environment just after the execution of the prebuild script. The environment is dumped into a local file, it is read back as separate lines (as name/value pairs), and then &quot;os.environ&quot; is updated with the variable assignments that were read back.

The purpose of this procedure was explained in commit 134bbe88ac18 (&quot;BaseTools: add some comment for .PrebuildEnv file&apos;s usage&quot;, 2017-07-26).

The logic was later updated in commit 2f28dca16ff3 (&quot;BaseTools:use iterate list to replace the itertools&quot;, 2019-02-01). This is what we have today (as of master @ 15ee7b76891a).

The bug is that the environment saving/restoring logic cannot deal with variables whose values contain newline characters (i.e., the logic cannot deal with multi-line variables).

Such variables are not frequent, but there is a special case: in the bash shell, shell functions may be exported with the &quot;export -f&quot; command. The shell turns such function definitions into special variables called

  BASH_FUNC_name%%

When a subshell or explicit child shell is then started, the new bash process automatically imports the inherited shell functions from the above-named variables.

The bug is the following: after the end of the prebuild script, the environment dump contains a multi-line format for such variables. (These variables may come from the prebuild script, or more likely, from the outermost parent shell, where the user started the &quot;build&quot; command manually.) The restoration, with &quot;os.environ.update(dict(envs))&quot;, corrupts these variables.

Subsequently, when &quot;build&quot; invokes any shell script, for example any one of the POSIX wrappers, the shell attempts to auto-parse these corrupted variables, and issues complaints such as

/bin/sh: which: line 1: syntax error: unexpected end of file
/bin/sh: error importing function definition for `which&apos;

In this instance, the relevant section of the &quot;Conf/.cache/.PrebuildEnv&quot; file is:

BASH_FUNC_which%%=() {  ( alias;
 eval ${which_declare} ) | /usr/bin/which --tty-only --read-alias --read-functions --show-tilde --show-dot &quot;$@&quot;
}

For comparison, here&apos;s a hexdump of the &quot;build.py&quot; python process&apos;s environment, and a hexdump of the environment of the shell process that was started by &quot;build.py&quot;:

000009f0  4c 45 53 53 4f 50 45 4e  3d 7c 7c 2f 75 73 72 2f  |LESSOPEN=||/usr/|
00000a00  62 69 6e 2f 6c 65 73 73  70 69 70 65 2e 73 68 20  |bin/lesspipe.sh |
00000a10  25 73 00 42 41 53 48 5f  46 55 4e 43 5f 77 68 69  |%s.BASH_FUNC_whi|
00000a20  63 68 25 25 3d 28 29 20  7b 20 20 28 20 61 6c 69  |ch%%=() {  ( ali|
00000a30  61 73 3b 0a 20 65 76 61  6c 20 24 7b 77 68 69 63  |as;. eval ${whic|
00000a40  68 5f 64 65 63 6c 61 72  65 7d 20 29 20 7c 20 2f  |h_declare} ) | /|
00000a50  75 73 72 2f 62 69 6e 2f  77 68 69 63 68 20 2d 2d  |usr/bin/which --|
00000a60  74 74 79 2d 6f 6e 6c 79  20 2d 2d 72 65 61 64 2d  |tty-only --read-|
00000a70  61 6c 69 61 73 20 2d 2d  72 65 61 64 2d 66 75 6e  |alias --read-fun|
00000a80  63 74 69 6f 6e 73 20 2d  2d 73 68 6f 77 2d 74 69  |ctions --show-ti|
00000a90  6c 64 65 20 2d 2d 73 68  6f 77 2d 64 6f 74 20 22  |lde --show-dot &quot;|
00000aa0  24 40 22 0a 7d 00                                 |$@&quot;.}.|

000009f0  00 4c 45 53 53 4f 50 45  4e 3d 7c 7c 2f 75 73 72  |.LESSOPEN=||/usr|
00000a00  2f 62 69 6e 2f 6c 65 73  73 70 69 70 65 2e 73 68  |/bin/lesspipe.sh|
00000a10  20 25 73 00 42 41 53 48  5f 46 55 4e 43 5f 77 68  | %s.BASH_FUNC_wh|
00000a20  69 63 68 00 25 00 28 29  20 7b 20 20 28 20 61 6c  |ich.%.() {  ( al|
00000a30  69 61 73 3b 00 5f 3d 2f  75 73 72 2f 62 69 6e 2f  |ias;._=/usr/bin/|
00000a40  65 6e 76 00 4d 41 4b 45  5f 46 4c 41 47 53 3d 00  |env.MAKE_FLAGS=.|

Note that the first hexdump (in the parent process) contains BASH_FUNC_which%% correctly, but the child process has a malformed variable.

tianocore-issues, 2024-12-20:
## Comment 16144
Date: 2021-05-25 21:44:34 +0000
From: @lgao4

Bob: can you check this issue?

tianocore-issues, 2024-12-20:
## Comment 16111
Date: 2021-05-21 14:11:06 +0000
From: @lersek

Note that the symptom only affects DSC files that define a PREBUILD script.

tianocore-issues, 2024-12-20:
## Comment 20402
Date: 2023-01-03 04:29:36 +0000
From: @lersek

I still see, at commit bbd30066e137, in
&quot;BaseTools/Source/Python/build/build.py&quot;:

&gt;| f0dc69e61bf23 (Yonghong Zhu         2016-03-31 14:05:59 +0800 1195)             if os.path.exists(PrebuildEnvFile):
&gt;| f0dc69e61bf23 (Yonghong Zhu         2016-03-31 14:05:59 +0800 1196)                 f = open(PrebuildEnvFile)
&gt;| f0dc69e61bf23 (Yonghong Zhu         2016-03-31 14:05:59 +0800 1197)                 envs = f.readlines()
&gt;| f0dc69e61bf23 (Yonghong Zhu         2016-03-31 14:05:59 +0800 1198)                 f.close()
&gt;| 2f28dca16ff34 (Zhijux Fan           2018-12-06 09:09:36 +0800 1199)                 envs = [l.split(&quot;=&quot;, 1) for l in envs ]
&gt;| 2f28dca16ff34 (Zhijux Fan           2018-12-06 09:09:36 +0800 1200)                 envs = [[I.strip() for I in item] for item in envs if len(item) == 2]
&gt;| f0dc69e61bf23 (Yonghong Zhu         2016-03-31 14:05:59 +0800 1201)                 os.environ.update(dict(envs))

So I believe nobody has cared about this report.

tianocore-issues, 2024-12-20:
## Comment 22148
Date: 2023-11-10 12:20:18 +0000
From: @lersek

I&apos;m reopening this because this bug still breaks the

  OvmfPkg/AmdSev/AmdSevX64.dsc

platform build, which uses PREBUILD.

lersek, 2025-05-15:
This bug persists. Can BaseTools maintainers take a look, please? Thank you.