tianocore/edk2 #10105: EFI_MM_COMMUNICATE_HEADER has architecture dependent sized field (Bugzilla Bug 3430)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 3430
Date: 2021-06-02T21:41:04+00:00
From: @kuqin12
To:   @kuqin12
CC:   @lgao4

Last updated: 2022-01-14T15:21:30+00:00

tianocore-issues, 2024-12-20:
## Comment 16254
Date: 2021-06-02 21:41:04 +0000
From: @kuqin12

- Industry Specification: UEFI Platform Initialization Specification Specification
- Target OS: ---
- Bugzilla Assignee(s): @kuqin12

In PI Spec v1.7 Errata A, Vol.4, Sec 5.7 MM Communication Protocol, the MessageLength field of EFI_MM_COMMUNICATE_HEADER (also defined as EFI_SMM_COMMUNICATE_HEADER) is currently defined as type UINTN.

But this structure, as a generic definition, could be used for both PEI and DXE MM communication. Thus for a system that supports PEI MM launch, but operates PEI in 32bit mode and MM foundation in 64bit, the current EFI_MM_COMMUNICATE_HEADER definition will cause structure parse error due to UINTN used.

The suggested change is to make the MessageLength field defined with definitive size as below:
```
typedef struct {
EFI_GUID  HeaderGuid;
UINT64    MessageLength;
UINT8     Data[ANYSIZE_ARRAY];
} EFI_MM_COMMUNICATE_HEADER;
```

This change will impact the structure consumers including:
MdeModulePkg/Core/PiSmmCore
MdeModulePkg/Application/SmiHandlerProfileInfo
MdeModulePkg/Application/MemoryProfileInfo

Note: MdeModulePkg/Library/SmmLockBoxLib/SmmLockBoxPeiLib is also consuming this structure, but it handled this size discrepancy internally. (Without the spec change in this ticket, all applicable PEI MM communicate callers will need to engage the same routine)

The reference implementation is tracked by ticket: https://bugzilla.tianocore.org/show_bug.cgi?id=3398.

tianocore-issues, 2024-12-20:
## Comment 16307
Date: 2021-06-08 21:44:37 +0000
From: @lgao4

Kun is working on it.

tianocore-issues, 2024-12-20:
## Comment 16325
Date: 2021-06-09 21:45:48 +0000
From: @kuqin12

(In reply to gaoliming from comment #1)
&gt; Kun is working on it.

First round of review is sent here: https://edk2.groups.io/g/devel/message/76303.

I cannot seem to push edk2-staging repo, so this round of code patches are still based off the personal branch on github.

Please advise if any. Thanks.

tianocore-issues, 2024-12-20:
## Comment 16398
Date: 2021-06-18 05:07:16 +0000
From: @kuqin12

v2 patch sent here: https://edk2.groups.io/g/devel/message/76738

tianocore-issues, 2024-12-20:
## Comment 18142
Date: 2022-01-14 15:21:30 +0000
From: @kuqin12

v4 patch sent here to request change of PI spec: https://edk2.groups.io/g/devel/message/85308

sachinami, 2025-01-28:
Hi @kuqin12 

As the required modifications to the specification has been done in PI 1.9. Would it be possible now to merge these changes to edk2 master.

kuqin12, 2025-04-07:
Resolved with https://github.com/tianocore/edk2/pull/10854