tianocore/edk2 #10124: Update Openssl to the latest version 1.1.1k (Bugzilla Bug 3524)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 3524
Date: 2021-08-03T22:55:51+00:00
From: @lgao4
To:   @lgao4
CC:   @jyao1, ken.geer, @mdkinney

Last updated: 2022-11-10T23:06:10+00:00

tianocore-issues, 2024-12-20:
## Comment 16770
Date: 2021-08-03 22:55:51 +0000
From: @lgao4

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): @lgao4

The latest stable openssl version is 1.1.1k.

tianocore-issues, 2024-12-20:
## Comment 17057
Date: 2021-08-24 22:15:35 +0000
From: @lgao4

lIMING TAKES IT.

tianocore-issues, 2024-12-20:
## Comment 17184
Date: 2021-09-07 08:15:59 +0000
From: @jyao1

Do you want to consider openssl-3.0 (https://bugzilla.tianocore.org/show_bug.cgi?id=3466) ?

tianocore-issues, 2024-12-20:
## Comment 17773
Date: 2021-11-16 14:32:34 +0000
From: Ken Geer &lt;<ken.geer>&gt;

The Latest Openssl version is now 1.1.1L

tianocore-issues, 2024-12-20:
## Comment 19990
Date: 2022-11-10 23:06:10 +0000
From: @mdkinney

edk2 is at openssl tag:

tag: OpenSSL_1_1_1n

Can this issue be closed now?

trhpx, 2026-09-26:
> ## Comment 19990
> 
> Date: 2022-11-10 23:06:10 +0000 From: [@mdkinney](https://github.com/mdkinney)
> 
> edk2 is at openssl tag:
> 
> tag: OpenSSL_1_1_1n
> 
> Can this issue be closed now?

Agreed. This should be closed to get things cleaned up 0.1%