tianocore/edk2 #10126: ConvertAsn1TimeToEfiTime function, SetMem actual params out of order (Bugzilla Bug 3550)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 3550
Date: 2021-08-10T17:11:16+00:00
From: Chris Stewart &lt;<chris.stewart>&gt;
To:   Chris Stewart &lt;<chris.stewart>&gt;
CC:   chris.stewart, @lgao4

Last updated: 2021-08-16T21:49:37+00:00

tianocore-issues, 2024-12-20:
## Comment 16864
Date: 2021-08-10 17:11:16 +0000
From: Chris Stewart &lt;<chris.stewart>&gt;

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): Chris Stewart &lt;<chris.stewart>&gt;

Created attachment 793
Patch to correct the bug.

In function ConvertAsn1TimeToEfiTime, there is an invocation to SetMem that has the 2nd and 3rd parameters swapped. The code has no effect:

SetMem (EfiTime, 0, sizeof (EFI_TIME));

It should be:

SetMem (EfiTime, sizeof (EFI_TIME), 0);

Attachment: [BZ-793-CryptTs.c.patch](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-793-CryptTs.c.patch)

tianocore-issues, 2024-12-20:
## Comment 16930
Date: 2021-08-16 21:49:37 +0000
From: @lgao4

Chris, please send your patch to edk2 mail list (devel@edk2.groups.io) for code review.

pierregondois, 2026-10-07:
Fixed by: https://github.com/tianocore/edk2/commit/7b4a99be8a39c12d3a7fc4b8db9f0eab4ac688d5

closed 2026-10-07 by pierregondois