tianocore/edk2 #10187: Smbios64BitPrintEPSInfo hit a stack guard exception due to buffer overflow. (Bugzilla Bug 3805)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 3805
Date: 2022-01-14T22:33:51+00:00
From: @ajfish
To:   Giri Mudusuru &lt;<girim>&gt;
CC:   @lgao4, girim, zhichao.gao

Last updated: 2023-05-05T22:49:07+00:00

tianocore-issues, 2024-12-20:
## Comment 18163
Date: 2022-01-14 22:33:51 +0000
From: @ajfish

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): Giri Mudusuru &lt;<girim>&gt;

Smbios64BitPrintEPSInfo () is coded like: 
UINT8  Anchor[5];

MemToString (Anchor, SmbiosTable-&gt;AnchorString, 5);

But the definition of MemToString()
  Copy Length of Src buffer to Dest buffer,
  add a NULL termination to Dest buffer.

So Anchor needs to be +1 the size of the SMBIOS Anchor string `_SM3_`.

typedef struct {
  UINT8   AnchorString[5];
  UINT8   EntryPointStructureChecksum;
  UINT8   EntryPointLength;
  UINT8   MajorVersion;
  UINT8   MinorVersion;
  UINT8   DocRev;
  UINT8   EntryPointRevision;
  UINT8   Reserved;
  UINT32  TableMaximumSize;
  UINT64  TableAddress;
} SMBIOS_TABLE_3_0_ENTRY_POINT;

The older SMBIOS_TABLE_ENTRY_POINT had a 4 byte AnchorString, so this looks like a copy paste bug.

This code used a lot of magic values. It would be nice if that was not the case.

tianocore-issues, 2024-12-20:
## Comment 18275
Date: 2022-02-08 22:01:28 +0000
From: @lgao4

Zhichao: can you help check this issue?

tianocore-issues, 2024-12-20:
## Comment 21462
Date: 2023-05-02 11:15:52 +0000
From: Giri Mudusuru &lt;<girim>&gt;

Patches sent for review 

[PATCH 1/2] MdePkg/IndustryStandard: Add SMBIOS anchor string &amp; length defines.
https://edk2.groups.io/g/devel/message/103786

[PATCH 2/2] ShellPkg/UefiShellDebug1CommandsLib: Replace hardcoded SMBIOS strings.
https://edk2.groups.io/g/devel/message/103785

tianocore-issues, 2024-12-20:
## Comment 21490
Date: 2023-05-05 22:49:07 +0000
From: @lgao4

Giri is working on it.

pierregondois, 2026-10-07:
Fixed by:
- 0b37723186ec1525b6caf14b0309fb0ed04084d7
- 2d4c76f783ee7eaab65b64f2f8c04f78394be0d1

closed 2026-10-07 by pierregondois