tianocore/edk2 #10277: Update the Serial Port Console Redirection Table Revision to 3 (Bugzilla Bug 4084)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4084
Date: 2022-09-27T06:26:06+00:00
From: sunil.i.dodamani
To:   sunil.i.dodamani
CC:   

Last updated: 2022-09-27T06:34:51+00:00

tianocore-issues, 2024-12-20:
## Comment 19639
Date: 2022-09-27 06:34:51 +0000
From: sunil.i.dodamani

Changing it to Confirmed

tianocore-issues, 2024-12-20:
## Comment 19638
Date: 2022-09-27 06:26:06 +0000
From: sunil.i.dodamani

- Industry Specification: ---
- Release Observed: EDK II Master
- Releases to Fix: EDK II Master
- Target OS: ---
- Bugzilla Assignee(s): sunil.i.dodamani

As per the latest specification here : https://learn.microsoft.com/en-us/windows-hardware/drivers/serports/serial-port-console-redirection-table the revision has to be 3. It is currently 2. Need to update the the header file

pierregondois, 2025-07-30:
Latest revision is now 4. This was updated at:
https://github.com/tianocore/edk2/commit/605582551a30ecebd6f90a5f5cf84b7de3a803d7

A macro for revision 3 has not been added, but could be if needed.
Closing the issue for now.