tianocore/edk2 #10279: UEFI cryptography agile solution - post quantum support (Bugzilla Bug 4087)

tianocore-issues, 2024-12-20:
This issue was created automatically with [bugzilla2github](https://github.com/makubacki/bugzilla2github/tree/bz2gh_tianocore)
# Bugzilla Bug 4087
Date: 2022-09-30T03:37:08+00:00
From: @jyao1
To:   @jyao1
CC:   @lgao4, kpanos, @SaloniKasbekar, @terryplee, timothy_hoppen, @vincent-j-zimmer

Last updated: 2024-08-14T10:38:06+00:00

tianocore-issues, 2024-12-20:
## Comment 19678
Date: 2022-09-30 03:38:26 +0000
From: @jyao1

Created attachment 1030
CodeFirst - UEFI Post-Quantum Cryptography Support

Attachment: [BZ-1030-CodeFirst - UEFI Post-Quantum Cryptography Support.docx](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-1030-CodeFirst%20-%20UEFI%20Post-Quantum%20Cryptography%20Support.docx)

tianocore-issues, 2024-12-20:
## Comment 19685
Date: 2022-09-30 08:59:39 +0000
From: @jyao1

The PI PQC support is at https://bugzilla.tianocore.org/show_bug.cgi?id=4089

tianocore-issues, 2024-12-20:
## Comment 19757
Date: 2022-10-11 21:54:25 +0000
From: @lgao4

Jiewen is working on it.

tianocore-issues, 2024-12-20:
## Comment 19677
Date: 2022-09-30 03:37:08 +0000
From: @jyao1

- Industry Specification: UEFI Specification
- Target OS: ---
- Bugzilla Assignee(s): @jyao1

In September 2022, NSA announced CNSA 2.0 compliance transition plan in
https://www.nsa.gov/Press-Room/News-Highlights/Article/Article/3148990/nsa-releases-future-quantum-resistant-qr-algorithm-requirements-for-national-se/.

We need prepare the post quantum cryptography support in UEFI.

tianocore-issues, 2024-12-20:
## Comment 22155
Date: 2023-11-14 00:17:04 +0000
From: @jyao1

Created attachment 1429
CodeFirst - UEFI Post-Quantum Cryptography Support - v2

Upload v2, use the official name (ML-DSA) published by NIST https://csrc.nist.gov/projects/post-quantum-cryptography.

Attachment: [BZ-1429-CodeFirst - UEFI Post-Quantum Cryptography Support - v2.docx](https://github.com/tianocore/user-attachments/raw/refs/heads/main/tianocore/edk2/BZ-1429-CodeFirst%20-%20UEFI%20Post-Quantum%20Cryptography%20Support%20-%20v2.docx)

tianocore-issues, 2024-12-20:
## Comment 22161
Date: 2023-11-15 13:32:53 +0000
From: @terryplee

Thank you for updating the document.  For the various crypto choices, could information be added to show the pros/cons of the choices?

tianocore-issues, 2024-12-20:
## Comment 22358
Date: 2024-01-12 18:24:01 +0000
From: @SaloniKasbekar

USWG is planning a new UEFI Specification release. Is this change ready to be intercepted for the next release?

tianocore-issues, 2024-12-20:
## Comment 22604
Date: 2024-02-15 13:17:59 +0000
From: @vincent-j-zimmer

@saloni - per late last year discussion with UEFI security subteam https://members.uefi.org/wg/usst/mail/thread/9909, park item until final NIST algorithm list released this year.  Not sure if this NIST info will arrive in time to catch the next UEFI spec release you mention.  

@terry - for comparison, material like https://uefi.org/sites/default/files/resources/Post%20Quantum%20Webinar.pdf provides some overview of overhead of the different algos.  Is this the type of material you were looking for?

tianocore-issues, 2024-12-20:
## Comment 22787
Date: 2024-03-13 13:49:51 +0000
From: kpanos

This comment is from MITRE’s PQCC Standards workstream. We want to stress the importance of introducing these algos to UEFI, especially SLH-DSA which is well-trusted and its 10-20KB is not prohibitive for most UEFI use-cases.  

PQCC is a coalition of industry members collaborating on various topics of interest regarding the upcoming post-quantum algorithm transition. One of these topics is gaps or areas of improvement in cryptographic specifications where there is no work taking place to integrate quantum-resistant algorithms to them. 

UEFI is such a specification. There have been some academic papers that investigate quantum-resistant signatures for UEFI [1] [2]. Also, Vincent and Jiewen&apos;s presentation from 2021 [3] laid out the landscape for new post-quantum signatures. A PoC implementation brought stateful hash-based signatures in UEFI [4]. The UEFI 2.10 Spec [5] introduced more crypto agility [6] by removing mandates of specific digest for the image as discussed in an old issue [7]. Theoretically, the specification can include any signature today if it was defined externally. The IETF is working on bringing stateless hash-based signatures in PKCS#7/CMS [8] [9]. So, after the IETF has specified the identifiers for these signatures, they could be used in UEFI. 

The software signing topic is relatively more urgent because updating BIOS or firmware is not trivial. And because BIOS or firmware stays in the field for a long time, adding quantum-safe signatures is important. Introducing the new signatures does not mean removing the classical ones given that an upgrade is not possible. But UEFI / PKCS #7 / CMS  allows for multiple signatures to exist so that each verifier can verify the signature it supports. This approach already was tested in the migration of using SHA-1 to SHA-256 in Secure Boot.

[1] https://eprint.iacr.org/2021/041.pdf 
[2] https://ieeexplore.ieee.org/document/9116252
[3] https://uefi.org/events/impact-post-quantum-cryptography-uefi-bios
[4] https://github.com/jyao1/CryptoEx
[5] https://uefi.org/specs/UEFI/2.10/
[6] https://uefi.org/specs/UEFI/2.10/Frontmatter/Revision_History.html?highlight=agile
[7] https://bugzilla.tianocore.org/show_bug.cgi?id=3413
[8] https://www.ietf.org/rfc/rfc8708.html
[9] https://datatracker.ietf.org/doc/draft-ietf-lamps-cms-sphincs-plus/

tianocore-issues, 2024-12-20:
## Comment 22793
Date: 2024-03-14 10:24:17 +0000
From: @jyao1

I fully agree with you that SLH-DSA is good option, as it is selected by NIST.
I also agree the stateful concern on LMS/XMSS. No doubt on that.

I should ask question more explicitly: &quot;Why SLH-DSA is more trusted than ML-DSA?&quot;

tianocore-issues, 2024-12-20:
## Comment 22791
Date: 2024-03-14 06:16:11 +0000
From: @jyao1

I am curious on why &quot;... especially SLH-DSA which is well-trusted ...&quot;.

Recommendation from CNSA 2.0 [1] is LMS/XMSS or Dilithium (ML-DSA).

Would you please share your insight on that?

[1] https://media.defense.gov/2022/Sep/07/2003071834/-1/-1/0/CSA_CNSA_2.0_ALGORITHMS_.PDF

tianocore-issues, 2024-12-20:
## Comment 22792
Date: 2024-03-14 09:46:29 +0000
From: kpanos

&gt; I am curious on why &quot;... especially SLH-DSA which is well-trusted ...&quot;.

Hash-based signatures are well-trusted primitives. They are the safest post-quantum options. They have been around for 40 years and generally the community thinks they are conservatively secure. They are the safest options we have.

Stateful Hash-Base Signatures (LMX/XMSS) are great but they have the state concern. If you reuse a state (meaning a one-time private key) the whole structure falls apart and you can forge a signature. As such, Stateful HBS requires careful operations at the signer to make sure state is not reused. This is not trivial especially for distributed systems. 

On top of that, NIST SP 800-208 has some stringent requirements about how keys are used in LMS/XMSS. For example you can&apos;t export a LMS/XMSS private key out of an HSM for backup. This is a big operational challenge. Vendors are trying to convince NIST to relax these requirements, but that is how the situation is right now. 

In the next couple of months (in their last presentation they said April 10-12, 2024), NIST will have published the final Stateless HBS document, SLH-DSA. SLH-DSA does not have the state management issues and can be used as-is. The verification is fast. Signing is slower. And the size varies ~10-20KB. It is still a conservatively secure scheme. 

For sizable software or firmware, including a signature of 10KB will not make a difference. If the firmware is 2KB, then maybe adding 10KB signature to it is wasteful. But in most use-cases where firmware or software is 100KB or more, an SLH-DSA signature will not be a problem.

So, who do we need HBS for firmware. Generally, firmware is hard to upgrade in the field. So, if we are to deploy a quantum-safe signature scheme in firmware that will live for 15-20 years, it ought to be a well-trusted one which will last and is unlikely to ever be broken. SLH-DSA is a great candidate. Signers can sign their firmware today with ECDSA or RSA and SLH-DSA. Legacy verifiers will only verify ECDSA or RSA. Upgraded verifiers will verify SLH-DSA. Backwards compatibility does not break.

&gt; Recommendation from CNSA 2.0 [1] is LMS/XMSS or Dilithium (ML-DSA).

Indeed, NSA came out with LMS/XMSS for software/firmware signing. When asked, they explained that that was because LMS/XMSS were the only &quot;finalized&quot; HBS options at the time. Personally, I think they rushed it a bit. 

But they have been asked repeatedly and have indicated that CNSA may be updated in the future to include SLH-DSA after it is ratified by NIST (in a couple of months that is). Intuitively it makes sense. SLH-DSA is HBS without the state management problem and it makes sense for CNSA to include it.

tianocore-issues, 2024-12-20:
## Comment 22794
Date: 2024-03-14 10:36:12 +0000
From: kpanos

&gt; I should ask question more explicitly: &quot;Why SLH-DSA is more trusted than ML-DSA?&quot;

Ah, gotcha. ML-DSA is based on module-learning with errors and structured lattices. Unstructured lattices and module learning with errors are older primitives that have been analyzed. Module learning with errors (ML-KEM/ ML-DSA) adds some structure to the lattice and is a newer primitive. That means that the structure in the lattice makes it more efficient, but theoretically it could be revealing something to a cryptanalyzer we don&apos;t know. In other words, theoretically there may be some unknown issue in ML-DSA and ML-KEM. There have been some recent papers that claim some reduction in the security of lattices. These are sometimes a little theoretical and hard to prove. There are implications in memory constraints for these analyses which affect their practicality, so there is no known issue with ML-KEM and ML-DSA. But intuitively they are newer schemes and if we had to bet on an attack if there was one, it is more likely it would be on ML-DSA than SLH-DSA. 

Merkle trees have been around for 40 years. SLH-DSA is like XMSS Merkle Trees with a FORS tree at the bottom. There have been many many analyses of these structures in the literature. So, SLH-DSA is so mature that the community trusts there is no unknown issue in it that we are not aware of. 

OK, NIST is standardizing both of them, they is no known issue with them. But conservative users would deploy SLH-DSA every time if the size and the signing performance was not an issue. Live connections like in TLS cannot afford the SLH-DSA size and signing time, but firmware signing can, I am arguing.

tianocore-issues, 2024-12-20:
## Comment 22795
Date: 2024-03-14 10:56:55 +0000
From: @jyao1

&gt; But conservative users would deploy SLH-DSA every time if the size and the signing performance was not an issue. 
&gt; Live connections like in TLS cannot afford the SLH-DSA size and signing time, but firmware signing can, I am arguing.

Understood. Thanks for sharing your insight.
So far, we can add SLH-DSA to the crypto agile list as well, especially when CNSA updates the recommendation, or NIST publishes the official specification.
We can discuss the final solution in USST with BIOS vendor, OS vendor, and PCI Card vendor.

tianocore-issues, 2024-12-20:
## Comment 22796
Date: 2024-03-14 15:06:13 +0000
From: @vincent-j-zimmer

great insight on SLH-DSA

BTW

anyone on this thread have thoughts or insight of the actual dates around &quot;...when CNSA updates the recommendation, or NIST publishes the official specification&quot; will land?

tianocore-issues, 2024-12-20:
## Comment 22797
Date: 2024-03-14 15:22:45 +0000
From: kpanos

Not sure about CNSA 2.0 potential updates. 

NIST said  in a panel this week that they will publish their first PQ standards (ML-KEM, ML-DSA, SLH-DSA) in their 5th PQC conference April 10-12. Not sure if they will hold that date, but that is what they said.

tianocore-issues, 2024-12-20:
## Comment 22843
Date: 2024-04-01 16:39:18 +0000
From: @terryplee

(In reply to Vincent Zimmer from comment #7)
&gt; @terry - for comparison, material like
&gt; https://uefi.org/sites/default/files/resources/Post%20Quantum%20Webinar.pdf
&gt; provides some overview of overhead of the different algos.  Is this the type
&gt; of material you were looking for?

Sorry for not responding sooner.  Thanks for the link to the overview presentation.  Yes, this is the type of overview I was looking for.

tianocore-issues, 2024-12-20:
## Comment 23044
Date: 2024-05-20 03:33:09 +0000
From: @jyao1

FYI: CNSA 2.0 FAQ (April 2024) still does NOT list SLH-DSA as approved.

https://media.defense.gov/2022/Sep/07/2003071836/-1/-1/0/CSI_CNSA_2.0_FAQ_.PDF

============
Q: Can I use SLH-DSA (aka SPHINCS+) to sign software?
A: While SLH-DSA is hash-based, it is not part of CNSA and is not approved for any
use in NSS
============

tianocore-issues, 2024-12-20:
## Comment 23319
Date: 2024-08-14 10:38:06 +0000
From: @vincent-j-zimmer

finalized algorithms https://www.federalregister.gov/documents/2024/08/14/2024-17956/announcing-issuance-of-federal-information-processing-standards-fips-fips-203-module-lattice-based

csosto-pk, 2025-02-10:
Hi all, just following up on this. 

1. Has there been any progress with an algorithm over the other? I did not see anything the github repo.
1. If you are moving on with ML-DSA, have you considered if you will do PQ-hybrid? That would mean that updated verifiers would need to apply "AND logic" for the classical and ML-DSA embedded signatures which is different than what UEFI verifiers do today.
1. Have you given any thought on pre-digesting? a/ ML-DSA and SLH-DSA have a pure and a prehash version and identifiers, but the IETF is shying away from using the latter. This means that UEFI could use the prehash identifiers for these signatures, but it would deviate slightly from what others are doing. b/ Another option is to not include a digest algorithm and use pure ML-DSA or SLH-DSA which do their internal digesting anyway. The issue with it, especially with SLH-DSA, would be the the signer can't support it if the digest is calculated in a different place than the signature. c/ There is a third option which is to use Signed Attributes in PKCS7/CMS which basically allows for digesting. This is probably the most straightforward option.

jyao1, 2025-02-11:
1. I think we are waiting for OSV's feedback. e.g. Microsoft. @vincent-j-zimmer, do you have any info?

2. Yes, it is not difficult to add Hybrid from implementation perspective. And we still need OSV's feedback.
We have 2 options for Hybrid: A) Define hybrid algorithms. B) Not define hybrid algorithm, but let user combine 2 algorithms.
There are PROs and CONs either way.

3. Technically, it is OK to choose one of them (pre-hash or pure version) or support both. Again, we need OSV's feedback.

BTW: I think we have other dependency on implementation - the PQC related RFC is still in draft, such as:
* https://datatracker.ietf.org/doc/draft-ietf-lamps-dilithium-certificates/
* https://datatracker.ietf.org/doc/draft-ietf-lamps-cms-sphincs-plus/
* https://datatracker.ietf.org/doc/draft-ietf-lamps-cert-binding-for-multi-auth/
* https://datatracker.ietf.org/doc/draft-ietf-lamps-pq-composite-sigs/

jyao1, 2025-03-26:
For implementation perspective:
1) liboqs (https://github.com/open-quantum-safe/liboqs) seems a promising project and it supports openssl provider (https://github.com/open-quantum-safe/oqs-provider). But liboqs claimed `WE DO NOT CURRENTLY RECOMMEND RELYING ON THIS LIBRARY IN A PRODUCTION ENVIRONMENT OR TO PROTECT ANY SENSITIVE DATA.`.
2) OpenSSL claimed the PQC support in the future version (https://openssl-foundation.org/post/2024-09-17-post-quantum/).
3) MbedTLS claimed the PQC support in the future roadmap (https://mbed-tls.readthedocs.io/en/latest/project/roadmap/).

csosto-pk, 2025-03-26:
[OpenSSL 3.5 coming on April 8](https://openssl-library.org/post/2025-02-04-release-announcement-3.5/) will introduce ML-DSA (it is already in their early branch). So, I think you could continue using OpenSSL in EDK II. 

But not you will need to bring in new verification logic in EDKII which looks at both signatures (classical and post-quantum) and validates only if at least the post-quantum passes (PQ pure) or both pass (PQ hybrid). That at least will need to happen in upgraded verifiers that can parse these new signatures. Legacy ones can ignore the PQ signature and only validate the classical one that they understand.

jyao1, 2025-04-03:
I have filed https://github.com/tianocore/edk2/issues/10917 as feature request to upgrade to OpenSSL 3.5+.

csosto-pk, 2025-04-07:
FYI, the UEFI section of https://pqcc.org/artifact-signing-dual-post-quantum-traditional-hybrid-signatures-and-downgrades/ lays out some challenges with UEFI signing that would need to be addressed to use the new signatures along with classical ones and prevent downgrade concerns. Also it includes some changes necessary regarding PKCS#7 signedData and encryptedDigest attributes.

cshroffgithub, 2025-05-21:
Is there any update on UEFI signing aspects with PQ ? Recently NSA published updated timelines for certification that includes software/firmware signing at https://pqcc.org/artifact-signing-dual-post-quantum-traditional-hybrid-signatures-and-downgrades/

[image]